cocotb/cocotb #2935: Command Line Arguments for python

PiyushSaini09, 2022-04-05:
Hi-

I have been trying to get the command line arguments for some part of code inside python. Normally I can do that using `sys.argv`. But cocotb runs via `makefile`. I got confused on how to take the command line arguments for python file. 
Is it possible to do something like this here?

======================================================================================
Cocotb version --> 1.6.0
Python --> 3.9.0
Linux Distro --> Fedora
Simulator --> Cadence Xcelium

marlonjames, 2022-04-05:
VHPI and VPI provide a means to get argv, and that is put into `cocotb.argv`. What you get in the list will probably vary by simulator.
https://docs.cocotb.org/en/stable/library_reference.html#cocotb.argv

PiyushSaini09, 2022-04-06:
`cocotb.argv` holds the values that are given to the simulator. 
Is it possible to give some new user arguments on runtime?
For example, in case of a simple python file invoking command as `python <filename>.py <argv1>` will provide these arguments in `sys.argv`. Can the same arguments be reflected to my python TB if invoke `make <argv1>`? 
Or is there some other method?

What I'm trying to do is something like this:

  ```
      if (test1 == 1):
                print("statement1 was ran")
          else:
                print("statement1 was deselected")
   ```

The trigger of test1 will be controlled by the runtime argument.

eric-wieser, 2022-04-06:
`cocotb.plusargs` might also be of interest

PiyushSaini09, 2022-04-06:
Understood.
Was able to pass the arguments via `PLUSARGS` and call them in python using `cocotb.plusargs`

Thanks for the help :)

I'll close the issue.

krishnatejwani, 2022-04-06:
> `cocotb.plusargs` might also be of interest

Hi eric,
Can you please provide some examples of how this can be used inside cocotb-testbench?

cmarqu, 2022-04-11:
Here is an example: https://github.com/cocotb/cocotb/blob/master/tests/test_cases/test_plusargs/plusargs.py

PiyushSaini09, 2022-05-13:
Hi,
I played a bit with plusargs and now I'm stuck with one issue.

Makefile snippet:
PLUSARGS+=+a=3 
PLUSARGS+=+b=4
PLUSARGS+=+c=0

Python snippet:
print(cocotb.plusargs)
a=(cocotb.plusargs)
aRandom = a['doRandom']
sclVal = a['SCL']
oscClkVal = a['oscClk']
if (sclVal=='0' and oscClkVal=='0' and aRandom == '1'):
        oscClkFreq,sclFreq = generateRandomClk()    else:   
        oscClkFreq = float(oscClkVal)*10**6
        sclFreq = float(sclVal)*10**6
    print(oscClkFreq,sclFreq)

PiyushSaini09, 2022-05-13:
Hi,
I played a bit with plusargs and now I'm stuck with one issue.

Makefile snippet:
```
PLUSARGS+=+a=3
PLUSARGS+=+b=4
PLUSARGS+=+c=0
```

Python snippet:
```
print(cocotb.plusargs)
plusargsValues=(cocotb.plusargs)
cVal = plusargsValues['c']
aVal = plusargsValues['a']
bVal = plusargsValues['b']
if (aVal=='0' and bVal=='0' and cVal=='1'):
   #dosomething 
else:
     #dosomething
print(aVal,bVal)
```

What I am doing now is invoking the command `make TESTCASE=<testname> PLUSARGS+=+a=11 PLUSARGS+=+b=4`. This command however will produce an error because `c` was not defined anywhere in the command. The error will be thrown by the python file (stating no such variable as `c`). Is there a way to avoid this error or will I have to always give the value to the variable so that it can be seen by the python file?

marlonjames, 2022-05-13:
You need to use the `override` directive in your Makefile for the `PLUSARGS+=` lines:

https://www.gnu.org/software/make/manual/html_node/Override-Directive.html#Override-Directive

PiyushSaini09, 2022-05-16:
Thank you for the help. This method fulfills my requirement to some extent. 
But with this now the variable will always have a particular value whether I pass it from the command line or not.

Please rectify me if I'm wrong here: Is it that Makefile can see the values of the variables only at one place? If I've defined the values of the variables in command line then it won't see the values inside the Makefile?

marlonjames, 2022-05-16:
If you want to be able to override *only* the `a=3` default inside the list of `PLUSARGS`, I'm not enough of a make expert for that.
You could however use separate variables and then add them to `PLUSARGS` after they have been resolved to their final values.

So `make TESTCASE=<name> A=11 B=4`
along with:
```makefile
A?=3
B?=4
C?=0

PLUSARGS+=$(A)
PLUSARGS+=$(B)
PLUSARGS+=$(C)
```

closed 2022-05-16 by marlonjames