cocotb/cocotb #2938: Get test name as a string?

dalleggio, 2022-04-12:
Is there a cocotb method to get the test name as a string?

I am currently using:
test_name = sys._getframe().f_code.co_name
which returns the method's (test) name.
But i'm thinking there should be a way to get it from cocotb.

Thanks
Anthony

imphil, 2022-04-24:
Anthony, thanks for filing this issue.

I don't think we have a much easier way to get the test name right now. Can you elaborate a bit what you're using the test name for?

dalleggio, 2022-04-24:
In my use case, each test uses several files with base names the same as the test names, e.g., test1.conf, test1.dat, etc.
I open the files in the test bench class using the test name.

eric-wieser, 2022-04-24:
For reference, pytest exposes this via the `request` fixture:
```
def test_name1(request):
    testname = request.node.name
```
That likely can't inform a cocotb API all that much sadly

ktbarrett, 2022-07-19:
???
```python
@cocotb.test()
async def test_example(_):
    print(test_example.__name__)
```

Or if you need it to be more abstract.
```python
def do_stuff():
    cocotb.regression_manager._test.__name__
```

We really should have a "current test" API which returning the currently running test coroutine. That would the responsibility of the Regression Manager. We also should have a "current task" API which returns the currently running task. That would be the responsibility of the scheduler.