cocotb/cocotb #2942: couldn't pip install on ubuntu 20.04 using g++ 7.5.0

poleguy, 2022-04-13:
I couldn't pip install on ubuntu 20.04 using g++ 7.5.0
I think this is the default g++ for a clean ubuntu install. 
Using conda environment.
Python 3.10.4 (main, Mar 31 2022, 08:41:55) [GCC 7.5.0] on linux
g++ (Ubuntu 7.5.0-6ubuntu2) 7.5.0

pip install cocotb

Error output:

Collecting cocotb
  Using cached cocotb-1.6.2.tar.gz (209 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: cocotb
  Building wheel for cocotb (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for cocotb (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [103 lines of output]
...
      g++ -pthread -B .../cenv/lib build/temp.linux-x86_64-cpython-310/cocotb/libs/libgpilog/cocotb/share/lib/gpi_log/gpi_logging.o -Lbuild/lib.linux-x86_64-cpython-310/cocotb/libs -o build/lib.linux-x86_64-cpython-310/cocotb/libs/libgpilog.so -flto -Wl,-rpath,$ORIGIN
      lto1: fatal error: bytecode stream in file ‘build/temp.linux-x86_64-cpython-310/cocotb/libs/libgpilog/cocotb/share/lib/gpi_log/gpi_logging.o’ generated with LTO version 8.1 instead of the expected 6.2
      compilation terminated.
      lto-wrapper: fatal error: /usr/bin/g++ returned 1 exit status
      compilation terminated.
      /home/.../cenv/compiler_compat/ld: error: lto-wrapper failed
      collect2: error: ld returned 1 exit status
      error: command '/usr/bin/g++' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb, which is required to install pyproject.toml-based projects

This error message was not helpful... after googling it I came up empty I tried a later g++ on a hunch and it worked. The requirements for cocotb don't specify what version of g++ is required or that you need to switch to it.
Work around:
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-9 60
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-9 60 
sudo update-alternatives --set g++ /usr/bin/g++-9
g++ --version
g++ (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
pip install cocotb
works!
I hope this helps somebody.

Possible room for improvement: update dependency documentation, automatically find correct g++.

marlonjames, 2022-04-13:
Install prerequisites are documented here:
https://docs.cocotborg/en/stable/install.html#installation-of-prerequisites

poleguy, 2022-04-14:
Thanks. I stand corrected... the link had a typo... it does indeed say that gcc 4.8.1+ here (corrected typo in link)
https://docs.cocotb.org/en/stable/install.html#installation-of-prerequisites

But here it is less specific:
https://docs.cocotb.org/en/v1.2.0/quickstart.html#installing-cocotb

And either way, it doesn't seem to work with 7.5.0 which is greater than 4.8.1
So maybe update the requirement to reflect that 7.5.0 is a no-go? Or maybe this bug report will be googleable-enough, and we're done here.

imphil, 2022-04-16:
> But here it is less specific:
[docs.cocotb.org/en/v1.2.0/quickstart.html#installing-cocotb](https://docs.cocotb.org/en/v1.2.0/quickstart.html#installing-cocotb)

That's the documentation for cocotb 1.2, you're using 1.6. 

> And either way, it doesn'Yes, that t seem to work with 7.5.0 which is greater than 4.8.1

GCC 7.5 should work and be recent enough to compile cocotb. The error message "bytecode stream in file ‘build/temp.linux-x86_64-cpython-310/cocotb/libs/libgpilog/cocotb/share/lib/gpi_log/gpi_logging.o’ generated with LTO version 8.1 instead of the expected 6.2" looks on first sight like a version mismatch, i.e. somehow two different compiler versions are being used together.

Can you post steps to reproduce this issue, ideally from a "standard" Ubuntu 20.04 environment? How did you set up your conda environment, what commands did you run before you ran the "pip install"?

closed 2022-05-29 by github-actions[bot]