cocotb/cocotb #2945: why await Timer doesn't work and feedback no error

cx104906, 2022-04-20:
I'm using ubuntu-18.04;cocotb-1.6.2;python-3.6.9;verilator-4.221.
When I run the example,I found it stucked:

make -f Makefile results.xml
make[1]: 进入目录“/home/cx/桌面/cocotbtest”
verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module my_design --vpi --public-flat-rw --prefix Vtop -o Vtop -LDFLAGS "-Wl,-rpath,/home/cx/.local/lib/python3.6/site-packages/cocotb/libs -L/home/cx/.local/lib/python3.6/site-packages/cocotb/libs -lcocotbvpi_verilator" --timescale 1ns/1ps /home/cx/桌面/cocotbtest/my_design.sv /home/cx/.local/lib/python3.6/site-packages/cocotb/share/lib/verilator/verilator.cpp
MODULE=test_my_design TESTCASE= TOPLEVEL=my_design TOPLEVEL_LANG=verilog \
         sim_build/Vtop 
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:99   in gpi_print_registered_impl       VPI registered
     0.00ns INFO     Running on Verilator version 4.221 devel
     0.00ns INFO     Running tests with cocotb v1.6.2 from /home/cx/.local/lib/python3.6/site-packages/cocotb
     0.00ns INFO     Seeding Python random module with 1650475532
     0.00ns INFO     Found test test_my_design.my_first_test
     0.00ns INFO     running my_first_test (1/1)

### It doesn't stop and have not other information or error.
The example is:

test_my_design.py
```
# test_my_design.py (simple)

import cocotb
from cocotb.triggers import Timer
#from remote_pdb import RemotePdb; rpdb = RemotePdb("127.0.0.1", 4000)

@cocotb.test()
async def my_first_test(dut):
    """Try accessing the design."""

    for cycle in range(10):
        dut.clk.value = 0
        #print('1')
        #rpdb.set_trace()
        await Timer(1, units="ns")
        #print('2')
        dut.clk.value = 1
        #print('3')
        await Timer(1, units="ns")

    dut._log.info("my_signal_1 is %s", dut.my_signal_1.value)
    assert dut.my_signal_2.value[0] == 0, "my_signal_2[0] is not 0!"

```
my_design.sv

```
// This file is public domain, it can be freely copied without restrictions.
// SPDX-License-Identifier: CC0-1.0

module my_design(input logic clk);

  timeunit 1s;
  timeprecision 1ns;

  logic my_signal_1;
  logic my_signal_2;

  assign my_signal_1 = 1'bx;
  assign my_signal_2 = 0;

endmodule

and makefile

```
# Makefile

```
# defaults
SIM ?= verilator
TOPLEVEL_LANG ?= verilog

VERILOG_SOURCES += $(PWD)/my_design.sv
# use VHDL_SOURCES for VHDL files

# TOPLEVEL is the name of the toplevel module in your Verilog or VHDL file
TOPLEVEL = my_design

# MODULE is the basename of the Python test file
MODULE = test_my_design

# include cocotb's make rules to take care of the simulator setup
include $(shell cocotb-config --makefiles)/Makefile.sim
```

I find it stucked at await Timer(1, units="ns")
but i don't known how to solve this
Thanks

ktbarrett, 2022-04-20:
Only Verilator v4.106 work with cocotb at this time.

closed 2022-04-20 by ktbarrett

DiegoRosales, 2022-05-11:
What is special about that specific version?

ktbarrett, 2022-05-11:
There were features/fixes added in that version that made Verilator and cocotb work together correctly. Then the very next release the Verilator author made a change which broke cocotb support. See #2300.