cocotb/cocotb #2949: Python 3.11: Tests fail due to "Trace function changed, data is likely wrong"

imphil, 2022-04-24:
In Python 3.11 (pre-release) we get a test failure in CI:

```
ERROR: called callback function threw exception
Traceback (most recent call last):
  File "/home/runner/work/cocotb/cocotb/cocotb/scheduler.py", line 385, in _react
    self._is_reacting = False
    ^^^^^^^^^^^^^^^^^
  File "/home/runner/work/cocotb/cocotb/.nox/dev_test/lib/python3.11/site-packages/coverage/pytracer.py", line 216, in _trace
    self.data_stack.pop()
    ^^^^^^^^^^^^^^^^^^^^^
IndexError: pop from empty list
Exception ignored in atexit callback: <bound method Coverage._atexit of <coverage.control.Coverage object at 0x7f21f17b1b10>>
Traceback (most recent call last):
  File "/home/runner/work/cocotb/cocotb/.nox/dev_test/lib/python3.11/site-packages/coverage/control.py", line 607, in _atexit
    self.stop()
    ^^^^^^^^^^^
  File "/home/runner/work/cocotb/cocotb/.nox/dev_test/lib/python3.11/site-packages/coverage/control.py", line 599, in stop
    self._collector.stop()
    ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/runner/work/cocotb/cocotb/.nox/dev_test/lib/python3.11/site-packages/coverage/collector.py", line 353, in stop
    self.pause()
    ^^^^^^^^^^^^
  File "/home/runner/work/cocotb/cocotb/.nox/dev_test/lib/python3.11/site-packages/coverage/collector.py", line 364, in pause
    tracer.stop()
    ^^^^^^^^^^^^^
  File "/home/runner/work/cocotb/cocotb/.nox/dev_test/lib/python3.11/site-packages/coverage/pytracer.py", line 269, in stop
    self.warn(
    ^^^^^^^^^^
  File "/home/runner/work/cocotb/cocotb/.nox/dev_test/lib/python3.11/site-packages/coverage/control.py", line 388, in _warn
    warnings.warn(msg, category=CoverageWarning, stacklevel=2)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
coverage.exceptions.CoverageWarning: Trace function changed, data is likely wrong: None != <bound method PyTracer._trace of <PyTracer at 0x7f21f15c02d0: 1711 lines in 18 files>> (trace-changed)
```

A CI log: https://github.com/cocotb/cocotb/runs/6147105555?check_suite_focus=true#step:25:976.

See also the comment by @ktbarrett in https://github.com/cocotb/cocotb/issues/2947#issuecomment-1107850909_

ktbarrett, 2022-04-24:
This is due to a bug in Python 3.11. It has already been fixed upstream. Not sure when we will see it flow downstream. https://github.com/nedbat/coveragepy/issues/1270.

imphil, 2022-04-24:
Good find, thanks for looking into it @ktbarrett!

imphil, 2022-09-01:
This issue is now resolved upstream, and our CI passes with Python 3.11 rc1.

closed 2022-09-01 by imphil