cocotb/cocotb #2950: Python 3.11: ResourceWarning: unclosed file

imphil, 2022-04-24:
In Python 3.11 we see repeated warnings like the one below in our CI logs:

```
2022-04-24T11:42:40.7823855Z Exception ignored in: <_io.FileIO name=2 mode='wb' closefd=True>
2022-04-24T11:42:40.7824437Z ResourceWarning: unclosed file <_io.TextIOWrapper name=2 mode='w' encoding='UTF-8'>
```

The following diff makes the warning disappear. It's most likely that re-opening stderr (fd 2) causes the warning.

```diff
diff --git a/cocotb/__init__.py b/cocotb/__init__.py
index 2425ecbf..3836e7b3 100644
--- a/cocotb/__init__.py
+++ b/cocotb/__init__.py
@@ -72,8 +72,8 @@ def _setup_logging():
     # buffering. In that case, try to reopen them with line buffering
     # explicitly enabled. This ensures that prints such as stack traces always
     # appear. Continue silently if this fails.
-    _stdout_buffer_result = _reopen_stream_with_buffering("stdout")
-    _stderr_buffer_result = _reopen_stream_with_buffering("stderr")
+    _stdout_buffer_result = True # _reopen_stream_with_buffering("stdout")
+    _stderr_buffer_result = True # _reopen_stream_with_buffering("stderr")
 
     # Don't set the logging up until we've attempted to fix the standard IO,
     # otherwise it will end up connected to the unfixed IO.
```

I'm unclear at this point what a good solution would be.

ktbarrett, 2022-04-24:
Strange that this is only a problem in Python 3.11. It might be a bug in CPython. Does this affect other simulators? One solution is to register a function which closes stdout or stderr if it is successfully reopened using the `atexit` module. This seem like a hack though. Another final question is whether we should be doing this at all.

rachitnigam, 2022-07-07:
Sorry to comment on a closed issue but I ran into this issue as well. I have a test harness that runs cocotb-based designs through a tool. When running the design with the test harness, I get the error message:
```
sys:1: ResourceWarning: unclosed file <_io.TextIOWrapper name=2 mode='w' encoding='UTF-8'>
```
However, if I just invoke the same test command from the CLI, I cannot recreate the error. This might have something to do with the fact that the test harness is not a TTY and maybe the behavior changes depending on that.

ktbarrett, 2022-07-08:
@rachitnigam This should probably be reported on [python/cpython](https://github.com/python/cpython/) as I think it's a change in behavior in Python 3.11. IMO stderr and stdout are expected to be automatically closed without warning on interpreter cleanup even if those streams were reopened.

PR #2965 (2022-05-11): Don't reopen output buffers
It works with GHDL and Icarus, we will see how Riviera and Questa react.

Closes #2950.