cocotb/cocotb #2952: Logging/printing doesn't seem to work

abarajithan11, 2022-04-28:
Info:
* Cocotb: 1.162
* OS: Windows 11, 64-bit
* Simulator: Icarus Verilog, Version 11 (2021-02-04), 64 bit
* Python: 3.9.7 via Anaconda 

Whenever I try `print()` or `cocotb.log.info()` it does not print anything in the console. All tests pass. Following is an example code. I tried setting the log level as well. But I'm not sure if I'm doing it right.
Note: Please ignore stuff like `dff` having different input & output widths...etc. It is an example dff design I gradually made more complex to test different features of cocotb.

Cocotb Testbench
```py
import random
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import FallingEdge

u/cocotb.test()
def test_dff_simple(dut):

    ''' Clock Generation '''
    clock = Clock(dut.clk, 10, units="us")
    cocotb.start_soon(clock.start())

    ''' Assign random values to input, wait for a clock and verify output '''
    for i in range(10):
        val = random.randint(0, 255)
        dut.d.value = val
        
        cocotb.log.info("hello") # THIS DOESNT WORK
        
        yield FallingEdge(dut.clk)
        assert dut.q.value == val, f"Failed on the {i}th cycle. Got {dut.q.value}, expected {val}"
```

RTL Design

```sv
// dff.sv
`timescale 1us/1ns

module dff #(
  parameter WIDTH_IN=8,
  parameter WIDTH_OUT=8
)(
    input  logic clk, 
    input  logic [WIDTH_IN -1:0] d,
    output logic [WIDTH_OUT-1:0] q
);

always @(posedge clk)
    q <= d;
endmodule
```

PyTest Setup
```py
# test_dff.py
from cocotb_test.simulator import run
from glob import glob
import pytest

u/pytest.mark.parametrize(
    "parameters", [
        {"WIDTH_IN": "8", "WIDTH_OUT": "16"}, 
        {"WIDTH_IN": "16"}
        ])
def test_dff(parameters):
    run(
        verilog_sources=glob('dff/hdl/*'),
        toplevel="dff",            # top level HDL
        module="dff_cocotb",       # name of cocotb test module
        simulator="icarus",

        parameters=parameters,
        extra_env=parameters,
        sim_build="dff/sim_build/" + ",".join((f"{key}={value}" for key, value in parameters.items())),
    )
```

cmarqu, 2022-04-28:
Add the `-s` option to the pytest call.

closed 2022-04-28 by ktbarrett

abarajithan11, 2022-04-29:
Thank you, my bad