cocotb/cocotb #2969: Using function with_timeout delays simulation end

kenobi314159, 2022-05-12:
I'm running a test with multiple testcases and I needed a watchdog to guard if the DUT doesn't freeze during individual stages of each test case, so I added usage of function `with_timeout` as in `await with_timeout(stage.join(), CLK_PERIOD * TIMEOUT_VALUE, "ns")`. Each test case has different expected time of execution and I don't want to figure out the ideal `TIMEOUT_VALUE` for each one of them, so I set the `TIMEOUT_VALUE` to a benevolent `5000` cycles. From function side of view it all works correctly, but when all the testcases finish and monitors, drivers, clock runner and everything else is killed, the simulations still continues for several thousand clock cycles (with the clocks not running any more).

I believe it is waiting for all the Timer triggers created in the `with_timeout` function to trigger. At the very end of the simulation I get these messages:
```
#      -.--ns DEBUG    cocotb.gpi                         ..b/fli/FliCbHdl.cpp:85   in cleanup_callback                Removing Postponed DELETE timer 0x9c4
#      -.--ns DEBUG    cocotb.gpi                         ..b/fli/FliCbHdl.cpp:85   in cleanup_callback                Removing Postponed DELETE timer 0x4c4b40
#      -.--ns DEBUG    cocotb.gpi                         ..b/fli/FliCbHdl.cpp:85   in cleanup_callback                Removing Postponed DELETE timer 0x1c9c380
#      -.--ns DEBUG    cocotb.gpi                         ..b/fli/FliCbHdl.cpp:85   in cleanup_callback                Removing Postponed DELETE timer 0x1c9c380
#      -.--ns DEBUG    cocotb.gpi                         ..b/fli/FliCbHdl.cpp:85   in cleanup_callback                Removing Postponed DELETE timer 0x1c9c380
#      -.--ns DEBUG    cocotb.gpi                         ..b/fli/FliCbHdl.cpp:85   in cleanup_callback                Removing Postponed DELETE timer 0x1c9c380
```

Is there some way to kill or delete the Timers sooner? It would be ideal if it was done at the end of the `with_timeout` function.

CocoTB version: 1.6.2
Modelsim version: 10.7f

ktbarrett, 2022-05-12:
Could you provide a minimal example? I'm not sure how the `with_timeout` and your "stages" are working together. `with_timeout` only ends when the time or the awaited object finish. Are you saying that `with_timeout` is continuing to block even though the timeout or awaited object finished? Or are you saying something is being "leaked" and the entire test is misbehaving?

marlonjames, 2022-05-12:
Looks like it's related to #188, which affected Modelsim 10.3.
I don't know if that issue is still present, or if there is a smarter work-around when the timer callback is far in the future of the simulation.

ktbarrett, 2022-05-12:
I want to say I've seen the same issue as #188 with edge triggers on 10.7c. I'm not sure why this should be causing test behavior to change. In my case I get "Trigger fired but no one was watching" or something to that effect from the scheduler, and my test continues normally.

kenobi314159, 2022-05-13:
> Could you provide a minimal example? I'm not sure how the `with_timeout` and your "stages" are working together. `with_timeout` only ends when the time or the awaited object finish. Are you saying [...]

No. The Timer doesn't block the program. The `with_timeout` exits after the stage job finishes and the test continues normally until the end of the last test case. But after that instead of finishing with the turning off of the clock runner, the simulation continues until the Timer trigger is set. This is why it's not actually a functional error. Just a nuisance. The waveform looks quite ugly when 80% of it is just static lines with nothing happening in the design.

kenobi314159, 2022-05-13:
> Looks like it's related to #188, which affected Modelsim 10.3. I don't know if that issue is still present, or if there is a smarter work-around when the timer callback is far in the future of the simulation.

I'm not sure. I get both `Not removing PRIMED timer` and `Removing DELETE timer` messages in my transcript, but it might not be related. I'm not that familiar with how the trigger implementation works with CocoTB and Modelsim.

ktbarrett, 2023-06-22:
We are fairly certain this has been resolved in cocotb 1.8. Please reopen with more info if it hasn't.

closed 2023-06-22 by ktbarrett