cocotb/cocotb #2975: Icarus Verilog failure "sorry: Format 1 not implemented for getting real values."

imphil, 2022-05-17:
As of today I'm getting the following error on Icarus Verilog master (5d97405724d7755ecfbe2bfc25f6e6e81f2d029a):

```
2022-05-17T18:53:51.1617134Z /usr/local/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s sample_module -f sim_build/cmds.f -g2012   /home/runner/work/cocotb/cocotb/tests/test_cases/issue_120/../../../tests/designs/sample_module/sample_module.sv
2022-05-17T18:53:51.3888688Z MODULE=issue_120 TESTCASE= TOPLEVEL=sample_module TOPLEVEL_LANG=verilog \
2022-05-17T18:53:51.3889876Z          /usr/local/bin/vvp -M /home/runner/work/cocotb/cocotb/.nox/release_test_sim-sim-icarus-toplevel_lang-verilog-gpi_interface-vpi/lib/python3.8/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp 
2022-05-17T18:53:51.3935770Z      -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:109  in set_program_name_in_venv        Using Python virtual environment interpreter at /home/runner/work/cocotb/cocotb/.nox/release_test_sim-sim-icarus-toplevel_lang-verilog-gpi_interface-vpi/bin/python
2022-05-17T18:53:51.4119241Z      -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:99   in gpi_print_registered_impl       VPI registered
2022-05-17T18:53:51.4752447Z      0.00ns INFO     cocotb                             Running on Icarus Verilog version 12.0 (devel)
2022-05-17T18:53:51.4755874Z      0.00ns INFO     cocotb                             Running tests with cocotb v1.7.0.dev0 from /home/runner/work/cocotb/cocotb/.nox/release_test_sim-sim-icarus-toplevel_lang-verilog-gpi_interface-vpi/lib/python3.8/site-packages/cocotb
2022-05-17T18:53:51.4756472Z      0.00ns INFO     cocotb                             Seeding Python random module with 1652813631
2022-05-17T18:53:51.6018398Z      0.00ns INFO     cocotb.regression                  Found test issue_120.issue_120_scheduling
2022-05-17T18:53:51.6019952Z      0.00ns INFO     cocotb.regression                  running issue_120_scheduling (1/1)
2022-05-17T18:53:51.6030760Z sorry: Format 1 not implemented for getting real values.
2022-05-17T18:53:51.6031850Z vvp: vpi_priv.cc:882: void vpip_real_get_value(double, s_vpi_value*): Assertion `0' failed.
```

* Bad: [5d97405724d7755ecfbe2bfc25f6e6e81f2d029a](https://github.com/steveicarus/iverilog/tree/5d97405724d7755ecfbe2bfc25f6e6e81f2d029a)
* Good: [58ac6ed1f89695acc0c8bcef2e71b3e005c025c2](https://github.com/steveicarus/iverilog/tree/58ac6ed1f89695acc0c8bcef2e71b3e005c025c2)

Most likely the regression window is closer to the bad commit, probably no more than a day or two of commits.

FYI @steveicarus Maybe you see the issue right away. Otherwise I'll try to narrow the regression window in the coming days and file an upstream issue.

marlonjames, 2022-05-17:
I did a `git bisect`:
```
a1485906ca05a708f8920977a4a7f3565c96e2b1 is the first bad commit
commit a1485906ca05a708f8920977a4a7f3565c96e2b1
Author: Stephen Williams <steve@icarus.com>
Date:   Sun May 15 18:47:18 2022 -0700

    Add parameters to vcd dumps

    Writing parameters into VCD files makes the values available to waveform
    tools. This can be done easily enough by writing out a $dumpadd section
    at the beginning of the file that sets the parameter values. We don't need
    to track the values over change, because by definition they do not change.

    This changes the typical vcd output as well, so a few of the regression tests
    need to be adjusted to account for this.

    Also, while tracking this down, found and fixed the vvp/README.txt documention
    for the .param/x records.

 ivtest/gold/array_dump.vcd.gold |  5 ++-
 ivtest/gold/br_gh156.vcd.gold   | 25 +++++++++++++
 ivtest/gold/dump_memword.vcd    |  5 ++-
 ivtest/gold/pr2859628.vcd.gold  |  5 ++-
 ivtest/gold/vcd-dup.vcd.gold    |  5 ++-
 ivtest/ivltests/br_gh156.v      | 17 +++++++++
 ivtest/regress-vlg.list         |  1 +
 vpi/sys_vcd.c                   | 77 +++++++++++++++++++++++++++++++++++++----
 vvp/README.txt                  |  8 ++---
 9 files changed, 132 insertions(+), 16 deletions(-)
 create mode 100644 ivtest/gold/br_gh156.vcd.gold
 create mode 100644 ivtest/ivltests/br_gh156.v
```
`sample_module.sv` is used in many of our tests, and it has a parameter of type `real`: https://github.com/cocotb/cocotb/blob/be1c95994e8143cdad481a65eea3a2be5db57e3f/tests/designs/sample_module/sample_module.sv#L42

For some reason the VCD dumping attempts to get the value in the unsupported format `vpiBinStrVal`.

steveicarus, 2022-05-18:
> Can you help us out by following the guidelines and attach a small,
> complete working (failing) example program? I will look into it, since I
> was the one who implemented this commit. I have a pretty good idea of
> what's going on.

And to be clear, I need this to be reported as an issue on the Icarus
Verilog github repository.

imphil, 2022-08-04:
Yes, Icarus master is working with cocotb's master currently (https://github.com/cocotb/cocotb/runs/7604209049?check_suite_focus=true).

closed 2022-08-04 by imphil

steveicarus, 2022-10-11:
Can you help us out by following the guidelines and attach a small,
complete working (failing) example program? I will look into it, since I
was the one who implemented this commit. I have a pretty good idea of
what's going on.