cocotb/cocotb #2978: Problems running examples/simple_dff on master branch

damnsavage, 2022-05-18:
Im having trouble running the example simple_dff. I can run other tests e.g. adder without any issues.
I have a fresh install of cocotb on 2 different linux systems (CentOS 64b, Ubuntu 64b) and the same problem in both cases.
Switching simulators between icarus and xcelium makes no difference.
cocotb master branch be1c959.

The error is: 
`     0.00ns CRITICAL Failed to import module test_dff: No module named 'cocotb.runner'                                                           `

I don't see why this error is given because module cocotb/runner.py exists and it has no problem importing cocotb/clock.py from the same directory. This version of the test_dff.py doesn't use cocotb.runner and it passes ok

`git checkout 720b0e1 -- test_dff.py`

Similarly switching to the stable branch also works.
By the way thank you for creating this project. It's fantastic to get away from SV...

imphil, 2022-05-18:
Thanks for using cocotb!

How did you install cocotb? For the examples and tests on master, you also need to install cocotb from master (i.e. from source).  You can find information about that at https://docs.cocotb.org/en/stable/install_devel.html

imphil, 2022-05-18:
Addition: Alternatively, you can use the examples that work with the release version of cocotb, e.g. version 1.6.2, by using the sources from back then: https://github.com/cocotb/cocotb/tree/v1.6.2

damnsavage, 2022-05-19:
Hi thanks, I didn't install the development cocotb that explains it. I've switched to the stable branch which works fine.

closed 2022-05-19 by damnsavage