cocotb/cocotb #2983: Cocotb don't return error when when some sources files aren't specified

brunbruns, 2022-05-20:
Hi,
I'm using Cocotb 1.6.2 with Questa
I had to test a design made with vivado HLS. I first only specified my top in the makefile. and neither cocotb nor questa return an error. In fact my output signal was "UUUUUUUUUUUUUUU". I now changed my makefile but I find surprising that the test can run since my top create entity declared in other source file.

ktbarrett, 2022-06-16:
Take this up with Questa, there is nothing we can do.

closed 2022-06-16 by ktbarrett

brunbruns, 2022-06-16:
Hi , my bad, I post this just after starting my verification with cocotb, it wasn't clear what was achieve by simulator or cocotb. 
In fact questa return a warning which is often lost in the sheel:
"component not bound" , as a consequence the component is replaced by a black box

Thanks for your answer