cocotb/cocotb #2986: M1 Architecture Incompatibility

incordios, 2022-05-22:
**the cocotb version used:** cocotb-1.6.2
**the operating system and version (32/64 bit):** m1 mac
**the simulator and version (32/64 bit):** homebrew icarus-verilog 11.0
**the Python version, and where it's coming from (e.g. system, Anaconda, self-installed, ...):** conda Python 3.10.4
**(part of) the log file (cleaned of proprietary information):**

/opt/homebrew/bin/iverilog -o out/sim.vvp -D COCOTB_SIM=1 -s adder -f out/cmds.f -g2012   sync_adder.v
MODULE=tb TESTCASE= TOPLEVEL=adder TOPLEVEL_LANG=verilog \
         /opt/homebrew/bin/vvp -M /opt/homebrew/Caskroom/miniconda/base/envs/cocotb/lib/python3.10/site-packages/cocotb/libs -m libcocotbvpi_icarus   out/sim.vvp 
libcocotbvpi_icarus:`/opt/homebrew/Caskroom/miniconda/base/envs/cocotb/lib/python3.10/site-packages/cocotb/libs/libcocotbvpi_icarus.vpl' failed to open using dlopen() because:
    dlopen(/opt/homebrew/Caskroom/miniconda/base/envs/cocotb/lib/python3.10/site-packages/cocotb/libs/libcocotbvpi_icarus.vpl, 0x0009): tried: '/opt/homebrew/Caskroom/miniconda/base/envs/cocotb/lib/python3.10/site-packages/cocotb/libs/libcocotbvpi_icarus.vpl' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e')).
VCD info: dumpfile sim.vcd opened for output.
ERROR: results.xml was not written by the simulation!

How can I specify the architecture - if one even exists for arm64 - when installing cocotb? Currently I'm just running "pip install cocotb" within my conda env. Though, I see it [might be possible to compile cocotb from source](https://github.com/cocotb/cocotb/issues/2930#issuecomment-1084921266)? 

Possibly related to #2732

dkim1235, 2022-07-02:
the cocotb version used: cocotb-1.6.2
the operating system and version (32/64 bit): m1 mac Monterey 12.4
the simulator and version (32/64 bit): homebrew icarus-verilog 11.0
the Python version, and where it's coming from (e.g. system, Anaconda, self-installed, ...): conda Python 3.9.12

```
/Library/Developer/CommandLineTools/usr/bin/make -f Makefile results.xml
/opt/homebrew/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s my_design -f sim_build/cmds.f -g2012   /Users/paulkim/Embedded_Systems/EmbeddedJourney/2022.07.01_coctb_quickstart/my_design.sv
MODULE=test_my_design_extended TESTCASE= TOPLEVEL=my_design TOPLEVEL_LANG=verilog \
         /opt/homebrew/bin/vvp -M /opt/homebrew/anaconda3/lib/python3.9/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp 
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:99   in gpi_print_registered_impl       VPI registered
     0.00ns INFO     Running on Icarus Verilog version 11.0 (stable)
     0.00ns INFO     Running tests with cocotb v1.6.2 from /opt/homebrew/anaconda3/lib/python3.9/site-packages/cocotb
     0.00ns INFO     Seeding Python random module with 1656723049
     0.00ns INFO     Found test test_my_design_extended.my_second_test
     0.00ns INFO     running my_second_test (1/1)
VCD info: dumpfile my_design.vcd opened for output.
     6.00ns INFO     my_signal_1 is x
     6.00ns INFO     my_second_test passed
     6.00ns INFO     ************************************************************************************************
                     ** TEST                                    STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                     ************************************************************************************************
                     ** test_my_design_extended.my_second_test   PASS           6.00           0.00       7353.20  **
                     ************************************************************************************************
                     ** TESTS=1 PASS=1 FAIL=0 SKIP=0                            6.00           0.00       1779.05  **
                     ************************************************************************************************
```
                     
I wanted to report that my attempt at using icarus verilog and cocotb worked and passed. 
Of course this was an extremely simple testcase from the quickstart guide. 

What I realized is that when I first bought my macbook m1, 
I downloaded whatever version of anaconda that was available at the time (3.8).
I remember it being set to use rosetta to convert the x86 to armx64.
Anaconda didn't have arm64 support at the time for macos.
I believe when installing cocotb, it doesn't port through rosetta if you have the x86 version of python installed.
Therefore I had compatibility issues where cocotb x86 was using iverilog arm64.
I uninstalled all versions of anaconda.
Then I downloaded the LTS version of anaconda (4.13.0)and python (3.9).
After all of that it worked. 

Best of luck.

vruizescribano, 2023-05-14:
the cocotb version used: latest
the operating system and version (32/64 bit): m1 mac
the simulator and version (32/64 bit): homebrew ghdl
the Python version, and where it's coming from (e.g. system, Anaconda, self-installed, ...): conda Python 3.9.12

I have read in the Documentation and here in github that cocotb only supports x86_64 architecture, leaving the Apple M1/M2 laptops away. Is there any workaround to make cocotb work properly in ARM64? It does not make sense that vunit+ghdl works very smooth in ARM64, and cocotb does not. Although I write VHDL,  I will try icarus-verilog + cocotb (+ yosys)  

Here is the logfile with the error:

make SIM=ghdl rm -f results.xml
/Library/Developer/CommandLineTools/usr/bin/make -f Makefile results.xml
\
	/opt/homebrew/bin/ghdl -i    --workdir=sim_build --work=work /Users/victoradrianruizescribano/Documents/Victor/cocotb/examples/simple_diff/dff.vhdl && \
	/opt/homebrew/bin/ghdl -m    --workdir=sim_build -Psim_build --work=work dff
rm -f results.xml
MODULE=test_dff TESTCASE= TOPLEVEL=dff TOPLEVEL_LANG=vhdl \
	 /opt/homebrew/bin/ghdl -r   --workdir=sim_build -Psim_build --work=work dff --vpi=/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.so  
loading VPI module '/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.so'
dlopen(/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.so, 0x0001): tried: '/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.so' (no such file), '/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/opt/homebrew/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.so' (no such file), '/opt/homebrew/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64'))
/opt/homebrew/bin/ghdl:error: cannot load VPI module '/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.so'
make[1]: *** [results.xml] Error 1
make: *** [sim] Error 2
victoradrianruizescribano@Victors-Laptop simple_diff %

imphil, 2023-05-14:
> I have read in the Documentation and here in github that cocotb only supports x86_64 architecture, leaving the Apple M1/M2 laptops away. Is there any workaround to make cocotb work properly in ARM64?

cocotb only provides prebuilt binaries for x86 architectures. It should be possible, however, to compile cocotb from source for arm architectures (see also https://github.com/cocotb/cocotb/issues/2986#issuecomment-1134886121). pip should do so automatically if necessary. Since you're not getting a compiler error, it seems that something else is wrong.

The message "(have 'arm64', need 'x86_64')" seems odd -- is the libcocotbvpi_ghdl.so file compiled for arm64 (i.e., exactly as you want it), but the Python x86? You'll need to do a bit more digging yourself here, as I'm not a Mac user.

paul-demo, 2023-12-27:
I'm seeing the same limitation trying to install cocotb using poetry on Apple silicon. Weirdly, the pip3 installed version of cocotb works fine on Apple M2 silicon. But, trying to use poetry fails because you have only published x86_64 wheels to pypi, so it gets to a a point where it tries to use a precompiled .so and realizes it's running on the wrong platform:

```bash
ImportError: dlopen(/path/to/.venv/lib/python3.9/site-packages/cocotb/simulator.cpython-39-darwin.so, 0x0002):
tried: '/path/to/.venv/lib/python3.9/site-packages/cocotb/simulator.cpython-39-darwin.so'
(mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')),
```

I've tried this but to no avail (I'm a beginner user of poetry):
```bash
poetry config --local installer.no-binary cocotb
```

This creates poetry.toml with the following contents (and increases the install time since it compiles cocotb using clang rather than using a prebuilt whl file). For some reason it still fails when loading the .so file, though it does not when installed using pip3.
```
[installer]
no-binary = ["cocotb"]
```

It would be really nice if someone could add support for Cocotb prebuilts on Arm64 Apple silicon in PyPI. That way it could be installed like any other OS/architecture is installed.

imphil, 2023-12-29:
> But, trying to use poetry fails because you have only published x86_64 wheels to pypi, so it gets to a a point where it tries to use a precompiled .so and realizes it's running on the wrong platform:

That sounds like a poetry issue then. If wheels for a given architecture are not available, cocotb should be compiled from source. Pip does the right things here. Can you reach out to the poetry people to figure out what's going on?

> It would be really nice if someone could add support for Cocotb prebuilts on Arm64 Apple silicon in PyPI. That way it could be installed like any other OS/architecture is installed -- that is, by downloading .whl files from PyPI.

We already have a huge build and (more importantly) test matrix for cocotb, adding Arm Mac support needs a dedicated volunteer who's willing to maintain these builds and tests. If you're willing to do it, we will accept patches -- but please be prepared to maintain this effort going forward.

paul-demo, 2023-12-29:
I traced the issue to Rosetta being activated during some poetry commands and not during others. So for example it might use Rosetta for ‘poetry install’ but not for ‘poetry add’ or ‘pip install’, in my particular case.

Having all bash wrapper scripts restart themselves using ‘arch -arm64’ when they discover that Rosetta is active worked for me!

There was one other commenter (Tom) who had a similar issue but didn’t post his solution.

anyway, probably not a Cocotb issue and it works fine building from source once I worked around the Rosetta issue.

closed 2025-08-27 by ktbarrett