cocotb/cocotb #2992: Weird behavior when accessing logic array "incorrectly"

mcejp, 2022-05-29:
Using cocotb v1.7.0.dev0 along with Icarus Verilog provided by [oss-cad-suite nightly](https://github.com/YosysHQ/oss-cad-suite-build/releases) from a few days ago

Design:

```verilog
module my_design(input logic clk);

  timeunit 1s;
  timeprecision 1ns;

  //logic my_array[0:2097152];
  logic my_array[0:1048576];
  //logic my_array[0:524288];
  //logic my_array[0:262144];

  // necessary to prevent my_array from disappearing
  assign my_signal_3 = my_array[0];

endmodule
```

Testbench:

```python
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
async def my_first_test(dut):
    """Try accessing the design."""

    dut.my_array.value[0] = 123
```

(example can be also cloned from: https://github.com/mcejp/cocotb_repro)

This example runs for ~30 seconds on my PC (generally seems to be _O(N)_ with the length of the array), and doesn't raise any errors.

Changing it to `dut.my_array[0].value` makes it raise an `OverflowError` as expected (assigning value 123 to 1-bit logic)

I know that the testbench is "wrong", but this behavior can be an ugly trap for newbies like myself :)

ktbarrett, 2022-05-29:
Change your timeunit to something reasonable. All cocotb tests end after a 1 step Timer to ensure the next test starts in a normal time phase. Your steps are enormous vs your precision.

mcejp, 2022-05-29:
That header is copied verbatim from the [Quickstart Guide](https://docs.cocotb.org/en/stable/quickstart.html)

ktbarrett, 2022-05-29:
I don't see `timeunit 1s` in that documentation.

marlonjames, 2022-05-29:
`dut.my_array.value` is going to build a Python list as long as the array is, getting the value at each index along the way. It makes sense that it takes a while for large arrays.

jonpovey, 2022-06-02:
> I don't see `timeunit 1s` in that documentation.

It's referenced, here. Does look fishy.
https://github.com/cocotb/cocotb/blob/master/examples/doc_examples/quickstart/my_design.sv

cmarqu, 2022-06-05:
I will take a look once I'm back from vacation.

PR #3052 (2022-08-23): Set timeunit to 1ns, same as timeprecision
Closes #2992.
commit 080b8dda (2022-08-23): Set timeunit to 1ns, same as timeprecision

Closes #2992.
closed 2022-08-23 by cmarqu