cocotb/cocotb #2993: Cocotb overrides user coverage configuration

miserva, 2022-05-30:
I created [a configuration file for Coverage.py](https://coverage.readthedocs.io/en/6.4/config.html) and defined COVERAGE_RCFILE environment variable pointing to it. It worked except my file omits were being ignored. It turns out cocotb passes its own omit argument to coverage in the regression module, overriding the user configuration. I think cocotb should leave all coverage configuration to the config file if COVERAGE_RCFILE is defined.

ktbarrett, 2022-06-16:
`COCOTB_LIBRARY_COVERAGE` collects coverage on the cocotb library, and `COVERAGE` collects user coverage omitting the library coverage. If you really want the cocotb library coverage included in your coverage, simply enable the library coverage and merge the coverage reports.

miserva, 2022-06-16:
What I am trying to do is to omit other Python packages in addition to cocotb. In the .coveragerc file I have the local/lib folder omitted but the setting is not going through because of cocotb passing its own omit argument. As explained in [coverage.py docs](https://coverage.readthedocs.io/en/6.4.1/api_coverage.html): "Many of these arguments duplicate and override values that can be provided in a configuration file. Parameters that are missing here will use values from the config file."
I can't think of a way to get around that.

ktbarrett, 2022-06-26:
So how do you propose we best address this? Wouldn't specifying `COVERAGE_RCFILE` affect the configurations passed to both call to `coverage.coverage`? Should we go down to a single `coverage.coverage` call and have `COCOTB_LIBRARY_COVERAGE` and `COVERAGE` simply turn on arguments to a default call, but if `COVERAGE_RCFILE` is specified pass no arguments?

miserva, 2022-06-27:
I think I see what you mean. So there are actually two `coverage.coverage` calls, one in regression module and one in \_\_init__ for the library coverage. And if, for example, you had Python site-packages folder omitted in a configuration file, then `COCOTB_LIBRARY_COVERAGE` doesn't work. To prevent that, you could pass `omit=""` or even `config_file=False` to the library `coverage.coverage`. For the regression one, I was thinking that if `COVERAGE_RCFILE` is specified then `coverage.coverage` should be passed no arguments. Else call like it is called right now.

ktbarrett, 2022-06-27:
I think that should work fine. We are open to PRs and 1.7 is closing in soon.

PR #3014 (2022-07-01): Add bypass for user-defined coverage configuration
coverage.py allows user to configure code coverage via a config file, whose filepath is specified by `COVERAGE_RCFILE` env variable. However, arguments passed to coverage class take priority over settings in the config file.
This commit makes coverage class to be called without arguments if `COVERAGE_RCFILE` is defined so that cocotb doesn't override the user code coverage configuration.
cocotb library coverage is set to ignore the config file.

Closes #2993
commit 591e6e2c (2022-07-01): Add bypass for user-defined coverage configuration (#3014)

coverage.py allows user to configure code coverage via a config file
specified with COVERAGE_RCFILE env variable. However, arguments passed
to coverage class take priority over settings in the config file.
This commit makes coverage class to be called without arguments if
COVERAGE_RCFILE is defined so that cocotb doesn't override the user's
code coverage configuration. cocotb library coverage is set to ignore
the config file. Fixes #2993
closed 2022-07-01 by ktbarrett