cocotb/cocotb #2997: Unable to find the package with SV file

racoon57, 2022-06-09:
I am having issues with packages not being found that are defined in the `top_m.sv`

In `top_m.sv`, ab_pkg is defined as -

```
import ab_pkg::*;                                                           #line 23

module top_m #(
. . .
. . .
parameter ab_pkg::ab_format_e AbFormat_p        = xyz            ,          #line 28
. . .
```
When I try to run `Makefile`, I have following errors:
```
# ** Error: /home/path1/path2/path3/top_m.sv(23): (vlog-13006) Could not find the package (ab_pkg).  Design read will continue, but expect a cascade of errors after this failure.  Furthermore if you experience a vopt-7 error immediately before this error then please check the package names or the library search paths on the command line.
# ** Error: /home/path1/path2/path3/top_m.sv(28): (vlog-2164) Class or package 'ab_pkg' not found.
# ** Error: (vlog-13069) /home/path1/path2/path3/top_m.sv(28): near "::": syntax error, unexpected ::, expecting ')'.
```

I am new to cocotb, your inputs might help me to understand cocotb in-depth, please help me fix the issue. 

Additional details - 

- ubuntu 20.04.4
- cocotb version: 1.6.2
- Simulator: Questa
- Python: 3.8.10

themperek, 2022-06-09:
You have to compile `ab_pkg`?

racoon57, 2022-06-09:
I need to simulate `top_m.sv` which has `ab_pkg`.
So I need to compile `ab_pkg`

I tried sourcing ab_pkg in `Makefile` but it has same errors

```
VERILOG_SOURCES += /home/path1/path2/path3/ab_pkg.sv
VERILOG_SOURCES += /home/path1/path2/path3/top_m.sv
```

miserva, 2022-06-09:
Delete sim_build folder and then try again. Changes in VERILOG_SOURCES won't take effect otherwise.

closed 2022-06-13 by racoon57