cocotb/cocotb #3000: Use newer assertion rewriting interface

ktbarrett, 2022-06-13:
See [here](https://docs.pytest.org/en/latest/reference/reference.html?highlight=register_assert_rewrite#pytest-register-assert-rewrite) for the newer assertion rewriting interface. Introduced in pytest 3.0.4 (Nov 2016).

ktbarrett, 2022-06-13:
Yeah just tested, doesn't work.

ktbarrett, 2022-06-13:
>I think the purpose of register_assert_rewrite is to apply the rewriting to filename that don't match test_*.py

And that was the problem I was trying to solve. It's causing some heartburn. People are wondering why they aren't getting the assertion rewriting. Perhaps we should throw a warning if pytest is available and your test is prefixed with "test" if there is no way around the name limitation.

closed 2022-06-13 by ktbarrett

eric-wieser, 2022-06-13:
There's an easy way around the name limitation, we just have to provide a non-default configuration to pytest

ktbarrett, 2022-06-14:
@eric-wieser I'm not finding the documentation too useful. Is there any way to supply a string built from `MODULE` to the pytest config object without dumping it into an ini first?