cocotb/cocotb #3001: Support for SV structures in cocotb

TomO700, 2022-06-14:
Hi,

I am trying to allocate some data to sv structs used in my design using cocotb-test.
Is there a format to do this, or do they have to be just allocated as a single array?

Thank you

themperek, 2022-06-16:
@TomO700 I have sent you back here. I can only guess this is `cocotb` issue, not `cocotb-test`.

We would need an example of what you want to do.

TomO700, 2022-06-18:
Hey,

Basically my dut has a sv packed struct like what's below. 
```
typedef struct packed{
        logic [63:0]   data;
        logic   [7:0]    addr;
        logic               wen;
        logic               ren;
        logic               en;
    }mem_s;
```
Is it possible to assign data through cocotb to each element of the structure separately? As I noticed the only straight forward way was to consider the structure as  a single array and assign data, which actually defeats the convenience of using a struct.
Any suggestions?

ktbarrett, 2022-06-26:
@TomO700 Have you tried not marking the struct as `packed`? The point of `packed` is to bit-pack the struct and support assignment like a logic array.

That being said, to support packed array you need to ensure your simulator is returning a struct handle to cocotb during iteration and ensure cocotb is finding the struct version first during our discovery *or* add some way to support both forms of assignment which would require adding a new GPI type, modifying the discovery process, and updating all of the GPI implementations.

Please provide an example design and cocotb test that reproduces the issue and state your simulator and version, *like the template asked you to do*.

closed 2022-08-05 by github-actions[bot]

AtaraxiaZ, 2023-10-26:
> I would prefer Icarus, Verilator has trouble running cocotb. All you need is to list your
> 
> * simulator and version (e.g. Icarus v11.0-dev)
> * operating system and version (e.g. Ubuntu 20.04 32-bit)
> * Python version (e.g. 3.8.8)
> * cocotb version affected (e.g. "all", or v1.6.0)
> 
> And for the minimum reproducible failing example, it can be a very simple SV design with just the signals going in and out and an assign statement, but also include a Makefile and a Python test that tries to discover and drive the struct members. You should put it in [this directory](https://github.com/cocotb/cocotb/tree/master/tests/test_cases) and make it like the other tests in that directory, and push that up in a fork.
> 
> That will allow us to reproduce the failure on Icarus, and we can see what happens on other simulators, like Questa or Riviera.

 I encountered the same problem with verilator, but when I changed to icarus verilog, cocotb cound find the signal under the struct.