cocotb/cocotb #3012: running cocotb inside a virtualenv

Gatherer, 2022-06-26:
I currently do have trouble to run cocotb inside a virtualenv.
I can not find any example to get it running. Is it even possible?

I think one of the problems is.
`cocotb-config --makefiles`
It always points to the system installation. Even I call the venv Makefile.sim
`@. $(VENVACTIVATE) && include $(VENVDIR)/lib/python3.8/site-packages/cocotb/share/makefiles/Makefile.sim`
The cocotb-config is used multiple times in other e.g. Makefile.verilator. So it would need quite a lot of changes to get around this.

* the cocotb version used: 1.6.2
* the operating system and version (32/64 bit): 64bit
* the simulator and version (32/64 bit): Verilator 4.106 
* the Python version, and where it's coming from (e.g. system, Anaconda, self-installed, ...): system 3.8

ktbarrett, 2022-06-30:
> the operating system and version (32/64 bit): 64bit

64 bit what? Windows? Red Hat Enterprise Linux? Solaris? 

Assuming you are running a POSIX system can you run and post the results of `command -v cocotb-config`?

ktbarrett, 2022-06-30:
By system installation do you mean the installation in `~/.local/` that `cocotb-config` is saying it's installed to, or do you mean the one in `/usr/`? Sounds like this is a problem with how your environment is set up. Do you have something setup to prepend `~/.local/` to your `PATH` in front of the virtualenv activation?

If Mint is setup like Ubuntu, the way they suggest for you to add `.local/` to the PATH is in the `.bashrc` file, which is ***incorrect***. All `PATH` manipulation should be done in `.bash_profile`. I was seeing a similar issue when invoking a bash shell instead of a virtualenv. Do you have `SHELL=bash`? `make` will load bashrc on every command...

Gatherer, 2022-07-01:
cocotb was installed by  `pip3 install cocotb`

> By system installation do you mean the installation in .....

By system installation I mean the standard installation which comes with Mint per default.

as example my Makefile for pytest together with cocotb. The base Makefile do always look the same for multiple project on work and privat, for e.g. Sphinx, pytest and pytest together with cocotb. It always works fine.

```
VENVDIR = venv
VENVACTIVATE = $(VENVDIR)/bin/activate

.PHONY: venv clean pytest_verilator pytest_icarus pytest_collect wave

venv: $(VENVACTIVATE)

$(VENVACTIVATE): requirements.txt
	test -d $(VENVDIR) || virtualenv --always-copy -p python3 $(VENVDIR); \
	. $(VENVACTIVATE); \
	pip install wheel; \
	pip install -r requirements.txt; \
	touch $(VENVACTIVATE)

clean:
	rm -rf $(VENVDIR)
	rm -rf sim_build
	rm -rf .pytest_cache
	rm -rf __pycache__
	rm -rf hpna_tb/__pycache__
	rm -rf test_files/__pycache__
	rm -rf log_files

pytest_verilator: venv
	@. $(VENVACTIVATE) && SIM=verilator pytest -o log_cli=True --log-cli-level=INFO pytest_setup.py

pytest_icarus: venv
	@. $(VENVACTIVATE) && SIM=icarus pytest -o log_cli=True pytest_setup.py

pytest_collect: venv
	@. $(VENVACTIVATE) && pytest --collect-only pytest_setup.py

wave:
	gtkwave sim_build/dump.fst wave.gtkw
```

Will test it on my MAC too on the weekend. The other questions and checks I can do when I am back home from work.

ktbarrett, 2022-07-01:
> No currently I do use echo "$SHELL" -> /usr/bin/zsh

Check your `.zshrc`.

>If cocotb is installed where ever but cocotb-config will always give the same results.

That's because when you call `cocotb-config` it's finding the `cocotb-config` you have installed in `~/.local/` first, because it comes first in `$PATH`. This causes `cocotb-config` to output the makefiles in the cocotb install in `~/.local/` rather than your virtualenv. The reason that `~/.local/` is showing up first in your path _in front of your virtualenv_ is likely because you are prepending it in your shell's rcfile which is run every time a shell is exec'd.

Gatherer, 2022-07-01:
mhhhh .. ok .. I guess I miss something and still have to figure out why this happens.
I do get the same problem on my mac. 

> Check your .zshrc.

Is nearly empty. No PATH updates here and in  `.oh-my-zsh/oh-my-zsh.sh`

When I remove cocotb `pip3 uninstall cocotb` or I remove `~/.local/` part from my PATH I do get `make: cocotb-config: Command not found`.
When I install cocotb `pip3 install cocotb` it will be installed to 

```
Processing /home/thomasg/.cache/pip/wheels/9c/de/1a/c8b7b7f3643d67e1f7f2e530ef51ae5993b64d8530ef706820/cocotb-1.6.2-cp38-cp38-linux_x86_64.whl
Installing collected packages: cocotb
  WARNING: The script cocotb-config is installed in '/home/thomasg/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed cocotb-1.6.2

```
So I have to add it to PATH again. 

Somehow I will figure it out. Maybe I try to avoid all the Makefiles and the `cocotb-config` stuff. 
Kind of confusing. All other virtual environments do work fine. Even pytest, cocotb-pytest together with cocotb.
Will figure it out.

closed 2022-08-09 by github-actions[bot]