cocotb/cocotb #3016: What does GPI stand for?

mole99, 2022-06-30:
I just want to make sure if GPI stands for "Generic Procedural Interface", similar to Verilog Procedural Interface (VPI) and VHDL Procedural Interface (VHPI), or if it stands for something else.

I searched the documentation for the meaning but could not find anything. Under ` cocotb/cocotb/share/include/gpi.h` it mentions that it defines a "Generic Language Interface", but that would be a different abbreviation.

Thanks!

miserva, 2022-06-30:
It stands for "Generalized Procedural Interface" according to the Wiki: https://github.com/cocotb/cocotb/wiki/cocotb-Internals#overview

mole99, 2022-06-30:
Yes, but it is also good to have a common wording, especially for my thesis :D

If it is okay for you I will create a PR to unify the namings to "Generic Procedural Interface".

cmarqu, 2022-06-30:
Yes, a PR would be welcome!

mole99, 2022-06-30:
@cmarqu Perfect, please see #3017