cocotb/cocotb #3018: MacOS : dlopen (        gpivpi.vpl, 0x0009): symbol not found in flat namespace '_PyErr_Print'.

TomO700, 2022-06-30:
I have installed cocotb and I am trying to run a test and get the following error. Posting after many failed attempts to fix. This comes with examples as well as my own test bench

```
/opt/homebrew/bin/vvp -M /opt/homebrew/lib/python3.9/site-packages/cocotb/share/makefiles/build/libs/arm64 -m gpivpi sim_build/sim.vvp   
gpivpi:`/opt/homebrew/lib/python3.9/site-packages/cocotb/share/makefiles/build/libs/arm64/gpivpi.vpl' failed to open using dlopen() because:
    dlopen(/opt/homebrew/lib/python3.9/site-packages/cocotb/share/makefiles/build/libs/arm64/gpivpi.vpl, 0x0009): symbol not found in flat namespace '_PyErr_Print'.

```

marlonjames, 2022-06-30:
You appear to be running cocotb 1.3.x or earlier. Have you tried a more recent version of cocotb?

TomO700, 2022-06-30:
Hey thanks. Yes I am, have a requirement to run an older version. Is there a fix for this?

marlonjames, 2022-06-30:
Can you update your post with the information asked for in the issue template?
* the cocotb version used,
* the operating system and version (32/64 bit),
* the simulator and version (32/64 bit),
* the Python version, and where it's coming from (e.g. system, Anaconda, self-installed, ...),

TomO700, 2022-06-30:
Done

marlonjames, 2022-06-30:
There's a discrepancy here:
> `/opt/homebrew/lib/python3.9/site-packages/cocotb/share/makefiles/build/libs/arm64`

> `/opt/anaconda3/bin/python   Python 3.7.4`

TomO700, 2022-06-30:
Oh ya. Are you suggesting to move the default to /opt/homebrew/lib/python3.9?

TomO700, 2022-06-30:
My path has both 
`/opt/anaconda3/bin:/opt/anaconda3/condabin:/Library/Frameworks/Python.framework/Versions/3.7/bin`

marlonjames, 2022-06-30:
I'm more wondering if the cocotb libraries are building and linking to the correct `libpython`. This appears to be run from the homebrew python 3.9 environment.

If you are using anaconda3, you should try activating a conda environment, installing cocotb 1.2 to that environment, then running a test from within the conda shell in that environment.

You could also check with `ldd libcocotb.so` from the `build/libs/arm64` directory when you run a test to check which `libpython` it is linked to.

TomO700, 2022-06-30:
using the macOS equivalent for `ldd`

```
$ otool -L libcocotb.so
libcocotb.so:
	/opt/homebrew/lib/python3.9/site-packages/cocotb/share/makefiles/build/libs/arm64/libcocotb.so (compatibility version 0.0.0, current version 0.0.0)
	/opt/homebrew/lib/python3.9/site-packages/cocotb/share/makefiles/build/libs/arm64/libgpilog.so (compatibility version 0.0.0, current version 0.0.0)
	/opt/homebrew/lib/python3.9/site-packages/cocotb/share/makefiles/build/libs/arm64/libcocotbutils.so (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.100.3)
```

TomO700, 2022-07-01:
So I changed python in the path to match the homebrew one, still the original error is there

marlonjames, 2022-07-01:
Could you do a clean and rebuild and post the `make` output of building the libraries?

TomO700, 2022-07-01:
Still the same :(

marlonjames, 2022-07-01:
1. Did you try this?
> If you are using anaconda3, you should try activating a conda environment, installing cocotb 1.2 to that environment, then running a test from within the conda shell in that environment.

2. Can you post the **output** of `make` when it builds the libraries?

closed 2022-08-09 by github-actions[bot]