cocotb/cocotb #3020: fails to access generic value with String type

gwbres, 2022-07-03:
Hello guys,

Please note I use GHDL (with --08 attribute) almost exclusively (I rarely do anything in verilog).

I can't seem to access a generic value if it bears the `string` type in my coco testbench, other values (`boolean` and `integer` namely, there seems to be another bug with `std_logic` type so I switched to `boolean`).

Here is my VHDL declaration:

```vhdl

entity my_entity is
generic (
     G_DATA_IN_WIDTH: positive := 16;
     G_OPMODE : string := "trunc"
     [...]
);
[...]
```

GHDL + cocotb makefile work fine, `G_OPMODE` is resolved properly:

```makefile
EXTRA_ARGS += --std=08 
SIM_ARGS += -gG_DATA_IN_WIDTH=10 # works fine
SIM_ARGS += -gG_OPMODE="trunc" # works fine
SIM_ARGS += -gG_OPMODE="opt2" # works fine
```

Access in python fails:

```python
d_in_width = dut.G_DATA_IN_WIDTH.value.integer # works fine
print(dut.G_OPMODE) 
# attributeError dut has no attribute G_OPMODE against ghdl 1.0.0
# panic! against ghdl >= 2.0.0
```
I got used to combining both the latest `cocotb` and `ghdl` (unstable?) releases. Mainly because my debian proposes `ghdl` too old, and I need proper `--std=O8` support. Currently I was using

```shell
ghdl --version   --> 3.0.0-dev
cocotb-config --version  --> 1.6.2
```
I downgraded ghdl to a stable release, and reinstalled both just to make sure.
Now using ghdl 2.0.0 stable and cocotb 1.6.2, i'm still having this panic:

```shell
:vpi_get: unknown property  # G_DATA_IN access ?
ghdl:error: NULL access dereferenced # print(G_OPMODE) ?
in process .XXXX(rtl).P4

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0 (2.0.0.r0.g8185ed878) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
Command line:
ghdl -r --std=08 --workdir=sim_build -Psim_build --work=work XXXXX --vpi=/XXXX/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_ghdl.so --vcd=testbench.vcd -gG_DATA_IN_WIDTH=10 -gG_OPMODE=trunc
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-backtraces.adb:237 index check failed
Call stack traceback locations:
0x56552d493540 [..]
******************************************************************
```
I'm not sure this is a 100% cocotb issue and not a mix GHDL/cocotb issue. 
GHDL has no problem assining the string value, in `sim` steps, with things like "-gG_OPMODE=feature1", which proves GHDL handles the type correctly.   
Having cocotb access other generic values fine, proves the __getattr__ somewhat works, but there seems to be an issue when type is set to "string"

ktbarrett, 2022-07-03:
Please provide a minimum reproducible failing example. GHDL uses VPI, which does not contain enough information to map all VHDL features, so I'm not surprised if something isn't being discovered correctly.

gwbres, 2022-07-03:
hello @ktbarrett,

Here are three short files that produce the error on my side

`Makefile`:
```makefile
TOPLEVEL_LANG = vhdl
export COCOTB_REDUCED_LOG_FMT
SIM = ghdl
EXTRA_ARGS += --std=08

SIM_ARGS += -gG_DATA_IN_WIDTH=10
SIM_ARGS += -gG_OPMODE=opt1
MODULE = testbench 

VHDL_SOURCES = test.vhd
TOPLEVEL = demo
include $(shell cocotb-config --makefiles)/Makefile.sim
```

One module in `vhdl`: `test.vhd`
```vhdl
library ieee;
use     ieee.std_logic_1164.all;
entity demo is
generic (
    G_OPMODE : string := "opt1";
    G_DATA_IN_WIDTH: positive := 16
);
port (
    clk: in std_logic;
    pointless: in std_logic;
    super_pointless: out std_logic
);
end entity demo;

architecture rtl of demo is

    signal reg_r: std_logic := '0';
begin
    reg_r <= reg_r xor pointless;
    super_pointless <= reg_r;
end architecture;
```

And a test script `testbench.py`

```python
import random
import logging

import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge
from cocotb.result import TestSuccess

@cocotb.test()
def test_opmode_access (dut):
    # opmodes
    d_in_width = dut.G_DATA_IN_WIDTH.value.integer
    opmode = dut.G_OPMODE
    
    clk = Clock(dut.clk, 10, 'ns')
    cocotb.fork(clk.start())
    raise TestSuccess
```

ktbarrett, 2022-11-29:
This is a known issue. Closing as dupe of #2584.

closed 2022-11-29 by ktbarrett