cocotb/cocotb #3034: cocotb.simulator link does not exist

H0sseinR0stami, 2022-07-24:
the following link is broken:
https://docs.cocotb.org/en/v1.6.2/_modules/cocotb/simulator.html

cmarqu, 2022-07-24:
The link is autogenerated, apparently from the imports it sees in the Python code, but that module is implemented in C++: https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/simulator/simulatormodule.cpp

ktbarrett, 2022-07-24:
As stated, the documentation is not broken, this is a Sphinx issue.

closed 2022-07-24 by ktbarrett

H0sseinR0stami, 2022-07-31:
Hi. I have found it in an other repository in the link below:
https://github.com/mciepluc/cocotb-test/blob/master/cocotb_test/simulator.py

marlonjames, 2022-08-01:
@H0sseinR0stami That is not the correct module.
The `cocotb.simulator` module is a Python extension module written in C++, and the source code was linked by @cmarqu above.