cocotb/cocotb #3041: incorrect GPI_EXTRA on windows

xwzheng1020, 2022-07-29:
cocotb version: latest
os: win10 64bit
simulator: modelsim 2019.2
python: 3.9.12

When I try to use multi language (Verilog and VHDL) to run simulation with modelsim, it will fail when loading `C:/ProgramData/Anaconda3/lib/site-packages/cocotb/libs/cocotbvpi_modelsim.dll` .

It seems to get a weird GPI_EXTRA value. I found in the makefile GPI_EXTRA is equal to `C:/ProgramData/Anaconda3/lib/site-packages/cocotb/libs/cocotbfli_modelsim.dll:cocotbfli_entry_point`. But when getenv( https://github.com/cocotb/cocotb/blob/bdcffda4cbec9bd9034cc787c60a9fadfb6bb904/cocotb/share/lib/gpi/GpiCommon.cpp#L183), it becomes to `C;C:\ProgramData\Anaconda3\Library\ProgramData\Anaconda3\lib\site-packages\cocotb\libs\cocotbfli_modelsim.dll;cocotbfli_entry_point`

This will cause load failure.

marlonjames, 2022-07-29:
See #3013

ktbarrett, 2022-07-29:
@marlonjames I guess this means it isn't just an issue with Riviera.

closed 2022-08-01 by xwzheng1020

Alfre-D-Wen, 2025-04-05:
@xwzheng1020 Hi, can you run simulation with modelsim correctly now?