cocotb/cocotb #3046: Queata GUI exits after simulation done

kele14x, 2022-08-08:
Hi,

I'm using cocotb with Questa sim. `GUI=1 make` works and I can see vim windows launch, it will be automatically closed after simulation done (`$finish` is called). Leave no change to investigating the waveform. 

Is there any way to pause the simulation or do not quit GUI at end of simulation?

themperek, 2022-08-08:
It should stop:

https://github.com/cocotb/cocotb/blob/bdcffda4cbec9bd9034cc787c60a9fadfb6bb904/cocotb/share/makefiles/simulators/Makefile.questa#L65-L67

kele14x, 2022-08-09:
Hi @themperek,

Thanks for your reply. I know where the problem is. 

I've run `make` before, and a **runsim.do** is generated with content `-onfinish exit`. Then I run `GUI=1 make`, the CMD argument `-gui` is append to tool, but the **runsim.do** file is not regenerated.

closed 2022-08-09 by kele14x