cocotb/cocotb #3051: Concurrent await for RisingEdge and FallingEdge breaks ModelSim

darsor, 2022-08-22:
Cocotb version 1.6.2
Ubuntu 18.04, 64-bit
ModelSim SE-64 10.6
Python 3.9.13 venv

I'm seeing an issue where concurrently awaiting for RisingEdge and FallingEdge on the same clock causes either or both to immediately return (instead of waiting for the proper edge), and sometimes causing ModelSim itself to segfault.

Here's a minimal example using the simple_dff example module (any module with clk input will work):

```
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge, FallingEdge

async def wait_falling_edge(clk):
    cocotb.log.info("Waiting for falling edge")
    await FallingEdge(clk)
    cocotb.log.info("At falling edge")

@cocotb.test()
async def dff_clock_test(dut):
    """Demonstrate waiting for rising and falling edges concurrently"""

    clock = Clock(dut.clk, 10, units="us")  # Create a 10us period clock on port clk
    cocotb.start_soon(clock.start())  # Start the clock
    
    # wait for a single falling edge
    cocotb.start_soon(wait_falling_edge(dut.clk))
    
    # at the same time, wait for sequential rising edges
    cocotb.log.info("Waiting for rising edges")
    for _ in range(5):
        await RisingEdge(dut.clk)
        cocotb.log.info("At rising edge")
```
The expected behavior is to print "At rising edge" at 0, 10, 20, 30, and 40 us, and to print "At falling edge" at 5 us. The actual result is:

```
#      0.00ns INFO     running dff_clock_test (1/1)
#      0.00ns INFO     Waiting for rising edges
#      0.00ns INFO     Waiting for falling edge
#      0.00ns INFO     At rising edge
#      0.00ns INFO     At rising edge
#      0.00ns INFO     At rising edge
#      0.00ns INFO     At rising edge
#      0.00ns INFO     At rising edge
#      0.00ns ERROR    Failing test at simulator request before test run completion: Simulator shutdown prematurely
```
accompanied with a VSIM stacktrace.
```
** Fatal: (SIGSEGV) Bad pointer access. Closing vsimk.
** Fatal: vsimk is exiting with code 211.
```

Note that I don't always see the segfault. Sometimes the test passes, but no simulation time has passed. The same thing happens if I using ClockCycles with rising=False instead of FallingEdge.

If I change the FallingEdge to a RisingEdge, everything works as expected.

ktbarrett, 2022-08-23:
Does this affect the FLI or the VPI or the VHPI?

darsor, 2022-11-08:
I believe this is using the FLI, but I don't know how to check.

imphil, 2022-11-10:
@darsor Are you using the the Verilog (then you're using VPI) or VHDL (then it's FLI) variant of simple_dff? Can you attach the full console log that you get when running "make", plus the actual command you're running, to this issue? That'll contain all relevant information.

darsor, 2022-11-10:
This is using the Verilog variant. The Makefile is essentially the standard cocotb Makefile example.
```
$ make
make -f Makefile results.xml
make[1]: Entering directory '/.../minimal_example'
mkdir -p sim_build
# Make sure all libs in SOURCES_VHDL_* are mentioned in VHDL_LIB_ORDER and vice versa
set -o pipefail; MODULE=test_dff TESTCASE= TOPLEVEL="work.dff" \
GPI_EXTRA= TOPLEVEL_LANG=verilog \
 /opt/modelsim/modeltech/linux_x86_64/vsim -c -64  -do sim_build/runsim.do  2>&1 | tee sim_build/sim.log
Reading pref.tcl

# 10.6

# do sim_build/runsim.do
# Model Technology ModelSim SE-64 vmap 10.6 Lib Mapping Utility 2016.12 Dec 13 2016
# vmap -c 
# Copying /opt/modelsim/modeltech/linux_x86_64/../modelsim.ini to modelsim.ini
# Model Technology ModelSim SE-64 vmap 10.6 Lib Mapping Utility 2016.12 Dec 13 2016
# vmap work sim_build/work 
# Modifying modelsim.ini
# Model Technology ModelSim SE-64 vlog 10.6 Compiler 2016.12 Dec 13 2016
# Start time: 11:50:08 on Nov 10,2022
# vlog -work work "+define+COCOTB_SIM" -sv -timescale 1ns/1ps -mfcu "+acc" /.../minimal_example/dff.sv 
# -- Compiling module dff
# 
# Top level modules:
#       dff
# End time: 11:50:08 on Nov 10,2022, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0
# vsim -onfinish exit -pli "/.../venv/lib/python3.9/site-packages/cocotb/libs/libcocotbvpi_modelsim.so" sim_build/work.dff 
# Start time: 11:50:09 on Nov 10,2022
# ** Note: (vsim-3812) Design is being optimized...
# ** Note: (vsim-3865) Due to PLI being present, full design access is being specified.
# //  ModelSim SE-64 10.6 Dec 13 2016Linux 4.15.0-193-generic
# //
# //  Copyright 1991-2016 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  ModelSim SE-64 and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
# Loading sv_std.std
# Loading work.dff(fast)
# Loading /.../venv/lib/python3.9/site-packages/cocotb/libs/libcocotbvpi_modelsim.so
#      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:110  in set_program_name_in_venv        Using Python virtual environment interpreter at /.../venv/bin/python
#      -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:99   in gpi_print_registered_impl       VPI registered
# Attempting stack trace sig 11
# Signal caught: signo [11]
# vsim_stacktrace.vstf written
# Current time Thu Nov 10 11:50:10 2022
# ModelSim SE Stack Trace
# Program = vsim
# Id = "10.6"
# Version = "2016.12"
# Date = "Dec 13 2016"
# Platform = linux_x86_64
# Signature = 38cedbd3f1f3b31c20fab0022d84937f
# 0    0x00000000006f90f5: '<unknown (@0x6f90f5)>'
# 1    0x00000000004a0476: '<unknown (@0x4a0476)>'
# 2    0x0000000000595e37: '<unknown (@0x595e37)>'
# 3    0x000000000059769e: '<unknown (@0x59769e)>'
# 4    0x00000000005de8eb: '<unknown (@0x5de8eb)>'
# 5    0x000000000054f37e: '<unknown (@0x54f37e)>'
# 6    0x000000000069393d: '<unknown (@0x69393d)>'
# 7    0x000000000049d423: '<unknown (@0x49d423)>'
# 8    0x00000000005df3e5: '<unknown (@0x5df3e5)>'
# 9    0x00000000009d2333: '<unknown (@0x9d2333)>'
# 10   0x00000000009d2cd1: '<unknown (@0x9d2cd1)>'
# 11   0x00000000009d441e: '<unknown (@0x9d441e)>'
# 12   0x00000000010d335d: '<unknown (@0x10d335d)>'
# 13   0x00000000010d77b6: '<unknown (@0x10d77b6)>'
# 14   0x00000000010d8ea1: '<unknown (@0x10d8ea1)>'
# 15   0x00000000010d9206: '<unknown (@0x10d9206)>'
# 16   0x0000000000dbc650: '<unknown (@0xdbc650)>'
# 17   0x000000000117910f: '<unknown (@0x117910f)>'
# 18   0x00000000011cc0a7: '<unknown (@0x11cc0a7)>'
# 19   0x000000000118efa7: '<unknown (@0x118efa7)>'
# 20   0x000000000118f289: '<unknown (@0x118f289)>'
# 21   0x0000000000ff2536: '<unknown (@0xff2536)>'
# 22   0x00000000009b2506: '<unknown (@0x9b2506)>'
# End of Stack Trace

** Fatal: (SIGSEGV) Bad pointer access. Closing vsimk.
** Fatal: vsimk is exiting with code 211.
Exit codes are defined in the "Error and Warning Messages"
appendix of the ModelSim User's Manual.
sys:1: ResourceWarning: unclosed file <_io.TextIOWrapper name=2 mode='w' encoding='UTF-8'>
#      0.00ns INFO     Running on ModelSim SE-64 version 10.6 2016.12
#      0.00ns INFO     Running tests with cocotb v1.6.2 from /.../venv/lib/python3.9/site-packages/cocotb
#      0.00ns INFO     Seeding Python random module with 1668106210
#      0.00ns INFO     Found test test_dff.dff_clock_test
#      0.00ns INFO     running dff_clock_test (1/1)
#      0.00ns INFO     Waiting for rising edges
#      0.00ns INFO     Waiting for falling edge
#      0.00ns INFO     At rising edge
#      0.00ns INFO     At rising edge
#      0.00ns INFO     At rising edge
#      0.00ns INFO     At rising edge
#      0.00ns INFO     At rising edge
#      0.00ns ERROR    Failing test at simulator request before test run completion: Simulator shutdown prematurely
# End time: 11:50:10 on Nov 10,2022, Elapsed time: 0:00:01
# Errors: 0, Warnings: 0
/.../venv/lib/python3.9/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:157: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 211
make[1]: Leaving directory '/.../minimal_example'
/.../venv/lib/python3.9/site-packages/cocotb/share/makefiles/Makefile.inc:39: recipe for target 'sim' failed
make: *** [sim] Error 2
```

ktbarrett, 2026-03-29:
`test_edge_triggers.test_both_edge_triggers` tests this and Questa does not seem to fail. This is likely caused by the very old version of Questa.