cocotb/cocotb #3061: How to iterate over input/output ports of a module ?

navaneeth-cirel, 2022-08-29:
Hi @imphil, 

There is a use case for me where I need to traverse the hierarchy and check values for all the signals(nets, regs, ports) at a particular time, it will be good if I can identify if a given signal is an internal net, input port or output port. Currently I am able to distinguish between `GPI_NET` or `GPI_REGISTER` by checking the `handle._type` attribute.

Any pointers on how to do this is appreciated.

Navaneeth

ktbarrett, 2022-08-29:
We never found a good way to do this without restructuring most of the C++ code. https://github.com/cocotb/cocotb/pull/1128

ktbarrett, 2022-08-29:
I think this is a dupe of #695. You can subscribe to that issue by clicking the "subscribe" button on the taskbar on the right.

closed 2022-08-29 by ktbarrett