cocotb/cocotb #3069: Cocotb 1.7.rc

Botnic, 2022-09-05:
I updated my Cocotb 1.6.2 to 1.7.rc with the following command:
```python3 -m pip install cocotb==1.7.rc2```

Now if I call  ```cocotb-config --makefiles``` (with or without any parameters), I get the following error message:

```Traceback (most recent call last):
  File "/home/reto.meier/.pyenv/versions/3.10.1/bin/cocotb-config", line 5, in <module>
    from cocotb.config import main
  File "/home/reto.meier/.pyenv/versions/3.10.1/lib/python3.10/site-packages/cocotb/__init__.py", line 43, in <module>
    import cocotb.handle
  File "/home/reto.meier/.pyenv/versions/3.10.1/lib/python3.10/site-packages/cocotb/handle.py", line 39, in <module>
    from cocotb import simulator
ImportError: cannot import name 'simulator' from partially initialized module 'cocotb' (most likely due to a circular import) (/home/reto.meier/.pyenv/versions/3.10.1/lib/python3.10/site-packages/cocotb/__init__.py)```

How can I fix this? Do I need to do some cleanup after installing?
Thank you for your help
Kind regards

Reto

OS: CentOS 8 / 64bit
Simulator 32bit Modelsim.

<!--
In case you are reporting a problem with running cocotb,
please remember to add relevant information about your environment, such as
* the cocotb version used,
* the operating system and version (32/64 bit),
* the simulator and version (32/64 bit),
* the Python version, and where it's coming from (e.g. system, Anaconda, self-installed, ...),
* (part of) the log file (cleaned of proprietary information).
If you are unsure about any of the above items, open the issue anyway and we will figure it out together.
Thanks for helping improve cocotb!
-->

imphil, 2022-09-05:
No cleanup should be necessary before/after installing a new version of cocotb, but things can certainly go wrong.

* Did you install a previous version of cocotb into the same pyenv? If so, try to remove it first.
* Try to create a clean venv to install cocotb into. `python3 -m venv; . venv/bin/activate; pip3 install cocotb --pre` should get you there.

Botnic, 2022-09-14:
I was not able to make it work until now.
There is one thing that I noticed:

When I install cocotb==1.6.2, I see the wheels work:

```
$ pip install --no-cache-dir --force-reinstall cocotb==1.6.2
Collecting cocotb==1.6.2
  Downloading cocotb-1.6.2.tar.gz (209 kB)
     |████████████████████████████████| 209 kB 5.3 MB/s            
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: cocotb
  Building wheel for cocotb (pyproject.toml) ... done
  Created wheel for cocotb: filename=cocotb-1.6.2-cp310-cp310-linux_i686.whl size=1851958 sha256=e013b2c862a5224a34d0033b64e339a9e364664295d59398a9f4b7bea719f237
  Stored in directory: /tmp/pip-ephem-wheel-cache-apyurxgh/wheels/30/59/d4/4e20410fcbe385403c90549a32d96703501e5a69041ab0bf46
Successfully built cocotb
Installing collected packages: cocotb
  Attempting uninstall: cocotb
    Found existing installation: cocotb 1.6.2
    Uninstalling cocotb-1.6.2:
      Successfully uninstalled cocotb-1.6.2
Successfully installed cocotb-1.6.2
WARNING: You are using pip version 21.3.1; however, version 22.2.2 is available.
You should consider upgrading via the '/home/user/.pyenv/versions/3.10.1/bin/python3.10 -m pip install --upgrade pip' command.
```

When installing cocotb==1.7.0 I never see the Building wheels:

```
Collecting cocotb==1.7.0
  Downloading cocotb-1.7.0-cp310-cp310-manylinux_2_5_i686.manylinux1_i686.manylinux_2_17_i686.manylinux2014_i686.whl (3.6 MB)
     |████████████████████████████████| 3.6 MB 5.2 MB/s            
Installing collected packages: cocotb
  Attempting uninstall: cocotb
    Found existing installation: cocotb 1.6.2
    Uninstalling cocotb-1.6.2:
      Successfully uninstalled cocotb-1.6.2
Successfully installed cocotb-1.7.0
WARNING: You are using pip version 21.3.1; however, version 22.2.2 is available.
You should consider upgrading via the '/home/user/.pyenv/versions/3.10.1/bin/python3.10 -m pip install --upgrade pip' command.

```

Could this issue be linked to the fact, that I run python with 32 bit? (Modelsim 32bit version)

PS: Updating Pip has no influence.

Botnic, 2022-09-14:
I see that the difference is, that once it downloades the tar file and once a (precompiled?) .whl file.

Botnic, 2022-09-14:
I managed to force pip to download the tar and compile it (does not compile yet)
Looking at the messages, I assume there is still an issue with 64bit vs. 32bit

```

CFLAGS="-I/usr/include/c++/8/i686-redhat-linux" pip install --no-binary :all: --no-cache-dir --force-reinstall cocotb==1.7.0
Collecting cocotb==1.7.0
  Downloading cocotb-1.7.0.tar.gz (196 kB)
     |████████████████████████████████| 196 kB 5.0 MB/s            
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: cocotb
  Building wheel for cocotb (pyproject.toml) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/reto.meier/.pyenv/versions/3.10.1/bin/python3.10 /home/reto.meier/.pyenv/versions/3.10.1/lib/python3.10/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /tmp/tmp1gddvrow
       cwd: /tmp/pip-install-6fxlr010/cocotb_1ccf056b72474e0fae879a56f2f94520
  Complete output (98 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-cpython-310
  creating build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/ANSI.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/__init__.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/_deprecation.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/_py_compat.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/_sim_versions.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/_version.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/binary.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/clock.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/config.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/decorators.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/handle.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/ipython_support.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/log.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/memdebug.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/outcomes.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/queue.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/regression.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/result.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/runner.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/scheduler.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/triggers.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/utils.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/wavedrom.py -> build/lib.linux-x86_64-cpython-310/cocotb
  copying cocotb/xunit_reporter.py -> build/lib.linux-x86_64-cpython-310/cocotb
  creating build/lib.linux-x86_64-cpython-310/pygpi
  copying pygpi/__init__.py -> build/lib.linux-x86_64-cpython-310/pygpi
  copying pygpi/entry.py -> build/lib.linux-x86_64-cpython-310/pygpi
  creating build/lib.linux-x86_64-cpython-310/cocotb/_vendor
  copying cocotb/_vendor/__init__.py -> build/lib.linux-x86_64-cpython-310/cocotb/_vendor
  copying cocotb/_vendor/distutils_version.py -> build/lib.linux-x86_64-cpython-310/cocotb/_vendor
  creating build/lib.linux-x86_64-cpython-310/cocotb/types
  copying cocotb/types/__init__.py -> build/lib.linux-x86_64-cpython-310/cocotb/types
  copying cocotb/types/array.py -> build/lib.linux-x86_64-cpython-310/cocotb/types
  copying cocotb/types/logic.py -> build/lib.linux-x86_64-cpython-310/cocotb/types
  copying cocotb/types/logic_array.py -> build/lib.linux-x86_64-cpython-310/cocotb/types
  copying cocotb/types/range.py -> build/lib.linux-x86_64-cpython-310/cocotb/types
  creating build/lib.linux-x86_64-cpython-310/cocotb/_vendor/find_libpython
  copying cocotb/_vendor/find_libpython/__init__.py -> build/lib.linux-x86_64-cpython-310/cocotb/_vendor/find_libpython
  copying cocotb/_vendor/find_libpython/__main__.py -> build/lib.linux-x86_64-cpython-310/cocotb/_vendor/find_libpython
  copying cocotb/_vendor/find_libpython/_version.py -> build/lib.linux-x86_64-cpython-310/cocotb/_vendor/find_libpython
  creating build/lib.linux-x86_64-cpython-310/cocotb/share
  creating build/lib.linux-x86_64-cpython-310/cocotb/share/makefiles
  copying cocotb/share/makefiles/Makefile.deprecations -> build/lib.linux-x86_64-cpython-310/cocotb/share/makefiles
  copying cocotb/share/makefiles/Makefile.inc -> build/lib.linux-x86_64-cpython-310/cocotb/share/makefiles
  copying cocotb/share/makefiles/Makefile.sim -> build/lib.linux-x86_64-cpython-310/cocotb/share/makefiles
  creating build/lib.linux-x86_64-cpython-310/cocotb/share/makefiles/simulators
[... 26 lines cut ...]
  copying cocotb/share/def/modelsim.def -> build/lib.linux-x86_64-cpython-310/cocotb/share/def
  creating build/lib.linux-x86_64-cpython-310/cocotb/share/lib
  creating build/lib.linux-x86_64-cpython-310/cocotb/share/lib/verilator
  copying cocotb/share/lib/verilator/verilator.cpp -> build/lib.linux-x86_64-cpython-310/cocotb/share/lib/verilator
  running build_ext
  building 'cocotb/libs/libgpilog' extension
  creating build/temp.linux-x86_64-cpython-310
  creating build/temp.linux-x86_64-cpython-310/cocotb
  creating build/temp.linux-x86_64-cpython-310/cocotb/libs
[... cut ...] e/reto.meier/.pyenv/versions/3.10.1/lib -o build/lib.linux-x86_64-cpython-310/cocotb/libs/libgpilog.so -flto -static-libstdc++ -Wl,-rpath,$ORIGIN
  /bin/ld: cannot find -lstdc++
  collect2: error: ld returned 1 exit status
  error: command '/usr/lib64/ccache/g++' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb, which is required to install pyproject.toml-based projects
```

Botnic, 2022-09-14:
Solution:
I had to install the packets  libstdc++-static.i686 gcc-toolset-11-libstdc++-devel.i686 (no idea why they were not used before)
Now I end up with the following message:

```
  In file included from cocotb/share/lib/vpi/VpiCbHdl.cpp:34:
  cocotb/share/lib/vpi/VpiImpl.h:42:10: fatal error: _vendor/vpi/sv_vpi_user.h: No such file or directory
   #include "_vendor/vpi/sv_vpi_user.h"
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  compilation terminated.
  error: command '/usr/lib64/ccache/gcc' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for cocotb

```

This is a known bug https://github.com/cocotb/cocotb/issues/3071
Using the lastest master now installes it :-) :+1:

ktbarrett, 2022-09-14:
> I had to install the packets libstdc++-static.i686 gcc-toolset-11-libstdc++-devel.i686 (no idea why they were not used before)

I ran into the same problem. This is due to #3002. We just reverted that commit.

imphil, 2022-09-15:
> We just reverted that commit.

"We", meaning you personally, Kaleb, I assume? The cocotb project didn't revert the commit. https://github.com/cocotb/cocotb/issues/3080 tracks improving the documentation.

ktbarrett, 2022-09-15:
Yes me personally.

themperek, 2022-10-05:
>ImportError: cannot import name 'simulator' from partially initialized module 'cocotb' (most likely due to a circular import) 

This can be solved with (in `site-packages/cocotb`) :

```cmd
mv simulator.cpython-310-i386-linux-gnu.so simulator.cpython-310-x86_64-linux-gnu.so
```

OS: CentOS 8 / 64bit
Python: 3.10 / 32bit

Botnic, 2022-10-05:
Thanks @themperek 
Works!

imphil, 2022-10-05:
But what's the problem, and why does it work? I'd like to really understand what's going on here, this area of code is too fragile for experiments :-)

themperek, 2022-10-05:
It looks like enumeration/naming for loading/(compiling?) `.so` is base on system architecture and not the python binary.

It looks like it is based no `sysconfig.get_config_var('EXT_SUFFIX')`  see: https://peps.python.org/pep-3149/
In this case it is `.cpython-310-x86_64-linux-gnu.so`.

```
Thus, an initial implementation of [PEP 384](https://peps.python.org/pep-0384), when Python is configured with the default set of flags, would search for the following file names when extension module foo is imported (in this order):

foo.cpython-XYm.so
foo.abi3.so
foo.so
```

themperek, 2022-10-05:
Some "interesting" "stuff": https://github.com/pypa/setuptools/pull/2723

themperek, 2022-10-06:
So there is a way to fix this:

While compiling 32bit python add `CC="gcc -m32"`:

So the command is : `env PYTHON_CONFIGURE_OPTS="--enable-shared" CC="gcc -m32" CFLAGS="-m32" LDFLAGS="-m32" pyenv install 3.x,y`

I have update the wiki.

@imphil I think we can close this.

imphil, 2022-10-06:
I would be interested in seeing how Debian compiles their 32b python packages on a 64b multiarch system (it they provide python that way).

themperek, 2022-10-06:
>I would be interested in seeing how Debian compiles their 32b python packages on a 64b multiarch system (it they provide python that way).

Ubuntu 20.04 works with `CC="gcc -m32"`

themperek, 2025-01-01:
It looks like it can be close. Please re-open in other case.

closed 2025-01-01 by themperek