cocotb/cocotb #3071: cocotb 1.7.0 build fails (_vendor/vpi/sv_vpi_user.h: No such file or directory)

stefan-muc, 2022-09-07:
I tried to build cocotb manually, but I fail:

``` 
building 'cocotb/libs/libcocotbvpi_icarus' extension
creating build/temp.linux-x86_64-cpython-310/cocotb/libs/libcocotbvpi_icarus
creating build/temp.linux-x86_64-cpython-310/cocotb/libs/libcocotbvpi_icarus/cocotb
creating build/temp.linux-x86_64-cpython-310/cocotb/libs/libcocotbvpi_icarus/cocotb/share
creating build/temp.linux-x86_64-cpython-310/cocotb/libs/libcocotbvpi_icarus/cocotb/share/lib
creating build/temp.linux-x86_64-cpython-310/cocotb/libs/libcocotbvpi_icarus/cocotb/share/lib/vpi
x86_64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DCOCOTBVPI_EXPORTS= -DVPI_CHECKING=1 -DICARUS= -D__STDC_FORMAT_MACROS= -Icocotb/share/include -Icocotb -I/tmp/build-env-frrfwotg/include -I/usr/include/python3.10 -c cocotb/share/lib/vpi/VpiCbHdl.cpp -o build/temp.linux-x86_64-cpython-310/cocotb/libs/libcocotbvpi_icarus/cocotb/share/lib/vpi/VpiCbHdl.o -std=c++11 -fvisibility=hidden -fvisibility-inlines-hidden -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wno-missing-field-initializers -Wnon-virtual-dtor -Woverloaded-virtual -flto
In file included from cocotb/share/lib/vpi/VpiCbHdl.cpp:34:
cocotb/share/lib/vpi/VpiImpl.h:42:10: fatal error: _vendor/vpi/sv_vpi_user.h: No such file or directory
   42 | #include "_vendor/vpi/sv_vpi_user.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1

ERROR Backend subprocess exited when trying to invoke build_wheel
```

I'm on cocotb `v1.7.0` tag in an environment that can build cocotb `v1.6.0` successfully.

OS: Ubuntu 22.04 LTS
Dependencies:
```
$ sudo apt-get install make gcc g++ python3 python3-dev python3-pip
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
g++ is already the newest version (4:11.2.0-1ubuntu1).
gcc is already the newest version (4:11.2.0-1ubuntu1).
make is already the newest version (4.3-4.1build1).
python3 is already the newest version (3.10.4-0ubuntu2).
python3-dev is already the newest version (3.10.4-0ubuntu2).
python3-pip is already the newest version (22.0.2+dfsg-1).
```

marlonjames, 2022-09-07:
Confirmed locally.
The source distribution on PyPI `cocotb-1.7.0.tar.gz` does not contain the vendored header files, only the Python files:
```
$ tree _vendor
_vendor
├── __init__.py
├── distutils_version.py
└── find_libpython
    ├── __init__.py
    ├── __main__.py
    └── _version.py

1 directory, 5 files
```
Looks like we need to update `MANIFEST.in` for the source distribution.

We should also probably add an sdist-based install+test to the Release pipeline.

imphil, 2022-09-07:
That's odd, the dev tests should use a sdist. More things to check :)

imphil, 2022-09-08:
Let's keep this open  until we have a backport for 1.7.1

imphil, 2022-09-08:
Looks like we lost not only environment variable isolation when switching from tox to nox, but also the "always install from sdist" step (which is one of the most helpful features of tox) :-( 

I'll add a CI Check, and then fast-track a fix for this for a 1.7.1 release to happen shortly.

imphil, 2022-09-08:
@stefan-muc One more question: Did you explicitly force an installation from sdist, or did pip choose to do that? (I'd expect Ubuntu 22.04 to be supported by the wheels we have)

stefan-muc, 2022-09-09:
@imphil 
I get the sources, change to the sources directory and do a `python3 -m  build`
So until then, no `pip` involved.

I later use `pip` to install built cocotb wheel on a build server without internet connection.

jrwagz, 2022-09-11:
For what it's worth I ran into this same issue when just trying to install cocotb from pip.  I haven't used cocotb before, but was just trying to learn more about it, and run into this issue when just trying to follow the standard installation instructions.

```
$ pip install cocotb
Defaulting to user installation because normal site-packages is not writeable
Collecting cocotb
  Using cached cocotb-1.7.0.tar.gz (196 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: cocotb
  Building wheel for cocotb (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for cocotb (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [193 lines of output]
```
...
```
clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers -arch arm64 -arch x86_64 -Werror=implicit-function-declaration -DCOCOTBVPI_EXPORTS= -DVPI_CHECKING=1 -DICARUS= -D__STDC_FORMAT_MACROS= -Icocotb/share/include -Icocotb -I/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/include/python3.8 -c cocotb/share/lib/vpi/VpiCbHdl.cpp -o build/temp.macosx-10.14-arm64-3.8/cocotb/libs/libcocotbvpi_icarus/cocotb/share/lib/vpi/VpiCbHdl.o -std=c++11 -fvisibility=hidden -fvisibility-inlines-hidden -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wno-missing-field-initializers -Wnon-virtual-dtor -Woverloaded-virtual -flto
      In file included from cocotb/share/lib/vpi/VpiCbHdl.cpp:34:
      cocotb/share/lib/vpi/VpiImpl.h:42:10: fatal error: '_vendor/vpi/sv_vpi_user.h' file not found
      #include "_vendor/vpi/sv_vpi_user.h"
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb, which is required to install pyproject.toml-based projects
```

imphil, 2022-09-11:
@jrwagz As a workaround, until 1.7.1 is released, you can follow https://docs.cocotb.org/en/latest/install_devel.html#install-devel to get a working installation.

Our of curiosity: which operating system/Python version/etc. are you using that the binaries didn't work for you?

jrwagz, 2022-09-11:
> @jrwagz As a workaround, until 1.7.1 is released, you can follow https://docs.cocotb.org/en/latest/install_devel.html#install-devel to get a working installation.
> 
> Our of curiosity: which operating system/Python version/etc. are you using that the binaries didn't work for you?

I'm on a M2 MacBook Air, so it's probably because I'm not on x86 but instead on ARM why `pip install cocotb` needed to build from source.

imphil, 2022-09-11:
OK, that's expected then. Thanks! (And expect a 1.7.1 release to come out shortly, fixing the "pip install cocotb" path for you.)

imphil, 2022-09-17:
cocotb 1.7.1 has just been released and should fix this issue on PyPi. Please reopen if that's not the case for you.

closed 2022-09-17 by imphil

jrwagz, 2022-09-17:
Confirming that `pip install cocotb` now works as expected on Apple Silicon, as of release 1.7.1, thanks!

imphil, 2022-09-17:
Thanks for the super quick confirmation and the bug report, of course!

stefan-muc, 2022-09-22:
Confirming that cloning the code and building it manually (as initially reported) works as of release 1.7.1
Thanks from my side too!

imphil, 2022-09-22:
For reference: A CI check to prevent such errors is coming in #3090.

PR #3072 (2022-09-08): Include vendored header files in source distribution
Fixes #3071