cocotb/cocotb #3073: modelsim cannot capture secondary generate loop

xwzheng1020, 2022-09-08:
cocotb version: 0.2.2
os: win10
simulator: modelsim se 2019.2

when i try to use generate statement with two-layer for loop, I cannot capture the secondary loop's block object in cocotb. The simulation will be successful when I do not try to access the instance in the secondary loop. 

test files: 

```verilog
# test_gen.v
`timescale 1ns / 1ps

module test_gen#(
    parameter X_SIZE = 4,
    parameter Y_SIZE = 4
)
(
    input   wire [X_SIZE*Y_SIZE-1:0] data_in,
    output  wire [X_SIZE*Y_SIZE-1:0] data_out, 
    input   wire clk,
    input   wire rst
);

genvar i,j;
generate
    for(i=0; i<X_SIZE; i=i+1) begin: blk
        for (j=0; j<Y_SIZE; j=j+1) begin: blk2
            bit_nor bit_nor_inst(clk, rst, data_in[i*X_SIZE+j], data_out[i*X_SIZE+j]);
        end
    end
endgenerate

endmodule

module bit_nor(
    input wire clk,
    input wire rst,
    input wire data_in,
    output reg data_out
);
    always @(posedge clk) begin
        data_out <= ~data_in;
    end
endmodule
```

```python
# test_gen.py

import cocotb
from cocotb.triggers import Timer
from cocotb.clock import Clock

@cocotb.test()
async def my_first_test(dut):
    await cocotb.start(Clock(dut.clk, 10.0, 'ns').start())

    for i in range(10):
        dut.data_in.value = i
        await Timer(9.9, units="ns")
        for b in dut.blk:
            dut._log.info(f"data_out = {b.blk2[0].bit_nor_inst.data_out}")
```

```makefile
# Makefile
SIM ?= modelsim
TOPLEVEL_LANG ?= verilog

VERILOG_SOURCES += test_gen.v
TOPLEVEL = test_gen
MODULE = test_gen
include $(shell cocotb-config --makefiles)/Makefile.sim
```

the failure log looks like:
```log
     0.00ns INFO     cocotb                             Running on ModelSim SE-64 version 2019.2 2019.04
     0.00ns INFO     cocotb                             Running tests with cocotb v1.7.0.dev0 from C:\ProgramData\Anaconda3\lib\site-packages\cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1662624140
     0.00ns INFO     cocotb.regression                  Found test test_gen.my_first_test
     0.00ns INFO     cocotb.regression                  running my_first_test (1/1)
                                                          Try accessing the design.
     9.90ns INFO     cocotb.regression                  my_first_test failed
                                                        Traceback (most recent call last):
                                                          File "E:\xx\projs\test_gen\test_gen.py", line 17, in my_first_test
                                                            dut._log.info(f"data_out = {b.blk2[0].bit_nor_inst.data_out}")
                                                          File "C:\ProgramData\Anaconda3\lib\site-packages\cocotb\handle.py", line 370, in __getattr__
                                                            raise AttributeError(f"{self._name} contains no object named {name}")
                                                        AttributeError: blk[3] contains no object named blk2
```

xwzheng1020, 2022-09-08:
If the simulator is changed to `icarus`, everything is ok.

xwzheng1020, 2022-09-08:
If the simulator is changed to icarus, everything is ok.

If only use one for-loop, modelsim can also capture the block's instance.

marlonjames, 2022-09-08:
Try setting `COCOTB_LOG_LEVEL=DEBUG`. That will print out more information about VPI accessing the design.

imphil, 2022-09-08:
"cocotb version: 0.2.2": really? (Did you run cocotb --version? Run cocotb-config --version instead) [Note to self and @themperek: we really need to remove the "cocotb" entry point from cocotb-test. I got confused by it in the past.]

xwzheng1020, 2022-09-09:
revise:
cocotb version: 1.7.0.dev0

yes, 0.2.2 is the output of `cocotb.exe -v`, not that of `cocotb-config.exe -v `

xwzheng1020, 2022-09-09:
the new failure log is: 

```
     0.00ns INFO     cocotb                             Running on ModelSim SE-64 version 2019.2 2019.04
     0.00ns INFO     cocotb                             Running tests with cocotb v1.7.0.dev0 from C:\ProgramData\Anaconda3\lib\site-packages\cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1662685234
     0.00ns DEBUG    gpi                                Looking for root handle 'test_gen' over 1 implementations
     0.00ns DEBUG    gpi                                Got a Root handle (test_gen) back from VPI
     0.00ns DEBUG    gpi                                Checking test_gen exists
     0.00ns DEBUG    cocotb.test_gen                    Created
     0.00ns DEBUG    cocotb.regression                  Python Path: ,C:\ProgramData\Anaconda3\python39.zip,C:\ProgramData\Anaconda3\DLLs,C:\ProgramData\Anaconda3\lib,C:\modeltech64_2019.2\win64,C:\Users\DELL\AppData\Roaming\Python\Python39\site-packages,C:\ProgramData\Anaconda3,C:\ProgramData\Anaconda3\lib\site-packages,C:\ProgramData\Anaconda3\lib\site-packages\cocotb-1.7.0.dev0-py3.9-win-amd64.egg,C:\ProgramData\Anaconda3\lib\site-packages\win32,C:\ProgramData\Anaconda3\lib\site-packages\win32\lib,C:\ProgramData\Anaconda3\lib\site-packages\Pythonwin
     0.00ns DEBUG    cocotb.regression                  PWD: E:\xx\projs\test_gen
     0.00ns INFO     cocotb.regression                  Found test test_gen.my_first_test
     0.00ns INFO     cocotb.regression                  running my_first_test (1/1)
                                                          Try accessing the design.
     0.00ns DEBUG    gpi                                Searching for clk
     0.00ns DEBUG    gpi                                VPI: clk initialized with 1 elements
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
     0.00ns DEBUG    gpi                                Checking test_gen.clk exists
     0.00ns DEBUG    cocotb.clk                         Created
     0.00ns DEBUG    gpi                                clk has 1 elements
     0.00ns DEBUG    gpi                                Searching for data_in
     0.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [15:0] and length >16<
     0.00ns DEBUG    gpi                                VPI: data_in initialized with 16 elements
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiNet(36)
     0.00ns DEBUG    gpi                                Checking test_gen.data_in exists
     0.00ns DEBUG    cocotb.data_in                     Created
     0.00ns DEBUG    gpi                                data_in has 16 elements
     9.90ns DEBUG    gpi                                Searching for blk
     9.90ns DEBUG    gpi                                Found pseudo-region test_gen.blk (hdl_name=test_gen but name=blk)
     9.90ns DEBUG    gpi                                VPI: Created GPI object from type vpiModule(32)
     9.90ns DEBUG    gpi                                Checking test_gen.blk exists
     9.90ns DEBUG    cocotb.blk                         Created
     9.90ns DEBUG    cocotb.blk                         Discovering all on blk
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=36 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=114 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=48 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=116 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=29 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=25 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=47 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=526 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=618 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=683 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=100 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=34 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=129 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=41 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=103 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=113 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=99 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=31 skipped for GPI_GENARRAY type
[... cut ...]
                                                        **************************************************************************************
                                                        ** TESTS=1 PASS=0 FAIL=1 SKIP=0                  9.90           0.30         33.23  **
                                                        **************************************************************************************
                                                        
```

@marlonjames

xwzheng1020, 2022-09-09:
after changing to icarus, part of the log:
```
     0.00ns INFO     cocotb                             Running on Icarus Verilog version 12.0 (devel)
     0.00ns INFO     cocotb                             Running tests with cocotb v1.7.0.dev0 from C:\ProgramData\Anaconda3\lib\site-packages\cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1662685954
     0.00ns DEBUG    gpi                                Looking for root handle 'test_gen' over 1 implementations
     0.00ns DEBUG    gpi                                Got a Root handle (test_gen) back from VPI
     0.00ns DEBUG    gpi                                Checking test_gen exists
     0.00ns DEBUG    cocotb.test_gen                    Created
...
     9.90ns DEBUG    gpi                                Searching for blk
     9.90ns DEBUG    gpi                                Found pseudo-region test_gen.blk (hdl_name=test_gen but name=blk)
     9.90ns DEBUG    gpi                                VPI: Created GPI object from type vpiModule(32)
     9.90ns DEBUG    gpi                                Checking test_gen.blk exists
     9.90ns DEBUG    cocotb.blk                         Created
     9.90ns DEBUG    cocotb.blk                         Discovering all on blk
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=36 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=114 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=48 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=116 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=29 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=25 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=47 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=526 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=618 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=683 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=100 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=34 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=129 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=41 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=103 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=113 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=99 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=31 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=61 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=105 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                vpi_iterator vpiOneToManyT=44 skipped for GPI_GENARRAY type
     9.90ns DEBUG    gpi                                Created iterator working from 'test_gen' with type vpiModule(32)
     9.90ns DEBUG    gpi                                vpi_scan found 'test_gen.blk[0]'
     9.90ns DEBUG    gpi                                VPI: Created GPI object from type vpiGenScope(134)
     9.90ns DEBUG    gpi                                Create a native handle
     9.90ns DEBUG    gpi                                Checking test_gen.blk[0] exists
     9.90ns DEBUG    cocotb.blk[0]                      Created
     9.90ns DEBUG    gpi                                vpi_scan found 'test_gen.blk[1]'
     9.90ns DEBUG    gpi                                VPI: Created GPI object from type vpiGenScope(134)
     9.90ns DEBUG    gpi                                Create a native handle
     9.90ns DEBUG    gpi                                Checking test_gen.blk[1] exists
     9.90ns DEBUG    cocotb.blk[1]                      Created
     9.90ns DEBUG    gpi                                vpi_scan found 'test_gen.blk[2]'
     9.90ns DEBUG    gpi                                VPI: Created GPI object from type vpiGenScope(134)
     9.90ns DEBUG    gpi                                Create a native handle
     9.90ns DEBUG    gpi                                Checking test_gen.blk[2] exists
     9.90ns DEBUG    cocotb.blk[2]                      Created
     9.90ns DEBUG    gpi                                vpi_scan found 'test_gen.blk[3]'
     9.90ns DEBUG    gpi                                VPI: Created GPI object from type vpiGenScope(134)
     9.90ns DEBUG    gpi                                Create a native handle
[... cut ...]
     9.90ns DEBUG    gpi                                VPI: Created GPI object from type vpiReg(48)
     9.90ns DEBUG    gpi                                Checking test_gen.blk[3].blk2[0].bit_nor_inst.data_out exists
     9.90ns DEBUG    cocotb.data_out                    Created
     9.90ns INFO     cocotb.test_gen                    data_out = x
    19.80ns DEBUG    cocotb.blk                         Yielding 0 of type <class 'cocotb.handle.HierarchyObject'> (test_gen.blk[0])
    19.80ns INFO     cocotb.test_gen                    data_out = 0
...
```

marlonjames, 2022-09-09:
Hmm. For VPI, we are only doing `vpi_handle_by_name` and that is not working for generate handles.
For Icarus we do a fallback check when `vpi_handle_by_name` fails, but not for other simulators.

xwzheng1020, 2022-09-13:
So I remove the [#ifdef](https://github.com/cocotb/cocotb/blob/784af383205d4b24f97f3209aed65b3db124599f/cocotb/share/lib/vpi/VpiImpl.cpp#L302) statement, then the test above passed both for modelsim and icarus.
Except maybe slow down, all thing seem to be ok.

imphil, 2022-09-13:
Let's keep this open until we have a fix for it.

imphil, 2022-09-13:
Would you mind doing a pull request @xwzheng1020?

xwzheng1020, 2022-09-14:
@imphil actually, i don't have the whole picture about what happened internally. According to #2143 , it was assumed that only icarus did not support the query of pseudo regions (vpiGenScopeArray), so the existing pseudo regions were inferred from the end objects. And, this solution is only applied to icarus. But now it seems that vpiGenScopeArray in modelsim only supports one level of generate statement (this is just my guess, not strictly confirmed). For the case of multi-level generation, we still need to infer like icarus. 
So I remove the `#ifdef ICARUS` in VPI.