cocotb/cocotb #3078: Mixed language simulation doesn't work with Aldec Riviera

abyszuk, 2022-09-14:
There's a problem when trying to run mixed language simulation under Aldec Riviera.
The problem can be reproduced even with the example in the repo.
Apart from the already know with VHDL->Verilog access mentioned in #2921 , it seems that Verilog->VHDL also doesn't work.

Here's a last part of simulator output when trying to run it with
`make SIM=riviera COCOTB_DEBUG_LEVEL=trace`:
```
# COUT:      0.00ns INFO     cocotb                             Running on Riviera-PRO version 2021.04.107.8106                                                                                                                                                                          
# COUT:      0.00ns INFO     cocotb                             Running tests with cocotb v1.7.0 from /home/adrian/.local/lib/python3.6/site-packages/cocotb                                                                                                                             
# COUT:      0.00ns INFO     cocotb                             Seeding Python random module with 1663159507                                                                                                                                                                             
# COUT:      0.00ns INFO     cocotb.regression                  pytest not found, install it to enable better AssertionError messages                                                                                                                                                    
# COUT:      0.00ns INFO     cocotb.regression                  Found test test_mixed_language.mixed_language_accessing_test                                                                                                                                                             
# COUT:      0.00ns INFO     cocotb.regression                  Found test test_mixed_language.mixed_language_functional_test                                                                                                                                                            
# COUT:      0.00ns INFO     cocotb.regression                  running mixed_language_accessing_test (1/2)                                                                                                                                                                              
# COUT:                                                           Try accessing handles and setting values in a mixed language environment.                                                                                                                                              
# COUT:    100.00ns INFO     cocotb.endian_swapper_mixed        Got: 'i_swapper_sv'                                                                                                                                                                                                      
# COUT:    100.00ns WARNING  gpi                                VPI: Simulator does not know this type (1029) via VPI                                                                                                                                                                    
# COUT: free(): invalid pointer                                                                                                                                                                                                                                                          
bash: line 1:  8315 Segmentation fault      (core dumped) GPI_EXTRA=/home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libcocotbvhpi_aldec.so:cocotbvhpi_entry_point TOPLEVEL_LANG=verilog MODULE=test_mixed_language TESTCASE= TOPLEVEL=endian_swapper_mixed /opt/aldec/riviera
pro/bin/vsimsa -do sim_build/runsim.tcl
```

marlonjames, 2022-09-14:
Try running again with `COCOTB_LOG_LEVEL=DEBUG` if you want the additional logging (not `COCOTB_DEBUG_LEVEL`).
`TRACE` is really only useful for tracing execution from simulator, through C++ callbacks, to Python, and back again.

abyszuk, 2022-09-14:
:facepalm: sorry, obvious typo. Here's a more detailed output (trimmed to IMHO interesting section)
```
# COUT:    100.00ns DEBUG    gpi                                Create a native handle
# COUT:    100.00ns DEBUG    gpi                                Checking endian_swapper_mixed.i_swapper_sv.csr_waitrequest exists
# COUT:    100.00ns DEBUG    gpi                                Found duplicate endian_swapper_mixed.i_swapper_sv.csr_waitrequest
# COUT:    100.00ns DEBUG    gpi                                vpi_scan found 'endian_swapper_mixed.i_swapper_sv.flush_pipe'
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# COUT:    100.00ns DEBUG    gpi                                VPI: flush_pipe initialized with 1 elements
# COUT:    100.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiReg(48)
# COUT:    100.00ns DEBUG    gpi                                Create a native handle
(cut...)
# COUT:    100.00ns DEBUG    gpi                                Create a native handle
# COUT:    100.00ns DEBUG    gpi                                Checking endian_swapper_mixed.i_swapper_sv.stream_out_valid exists
# COUT:    100.00ns DEBUG    gpi                                Found duplicate endian_swapper_mixed.i_swapper_sv.stream_out_valid
# COUT:    100.00ns DEBUG    gpi                                vpi_scan found 'endian_swapper_mixed.i_swapper_sv.stream_out_startofpacket'
# COUT:    100.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiPort(44)
# COUT:    100.00ns DEBUG    gpi                                Create a native handle
# COUT:    100.00ns DEBUG    gpi                                Checking endian_swapper_mixed.i_swapper_sv.stream_out_startofpacket exists
# COUT:    100.00ns DEBUG    gpi                                Found duplicate endian_swapper_mixed.i_swapper_sv.stream_out_startofpacket
# COUT:    100.00ns DEBUG    gpi                                vpi_scan found 'endian_swapper_mixed.i_swapper_sv.stream_out_endofpacket'
# COUT:    100.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiPort(44)
# COUT:    100.00ns DEBUG    gpi                                Create a native handle
# COUT:    100.00ns DEBUG    gpi                                Checking endian_swapper_mixed.i_swapper_sv.stream_out_endofpacket exists
# COUT:    100.00ns DEBUG    gpi                                Found duplicate endian_swapper_mixed.i_swapper_sv.stream_out_endofpacket
# COUT:    100.00ns DEBUG    gpi                                vpi_scan found 'endian_swapper_mixed.i_swapper_sv.stream_out_ready'
# COUT:    100.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiPort(44)
# COUT:    100.00ns DEBUG    gpi                                Create a native handle
# COUT:    100.00ns DEBUG    gpi                                Checking endian_swapper_mixed.i_swapper_sv.stream_out_ready exists
# COUT:    100.00ns DEBUG    gpi                                Found duplicate endian_swapper_mixed.i_swapper_sv.stream_out_ready
[... cut ...]
# COUT:    100.00ns DEBUG    gpi                                Checking if i_swapper_vhdl is native through implementation VHPI
# COUT:    100.00ns DEBUG    gpi                                VHPI: Creating :endian_swapper_mixed:i_swapper_vhdl of type 2 (vhpiCompInstStmtK)
bash: line 1:  5992 Segmentation fault      (core dumped) GPI_EXTRA=/home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libcocotbvhpi_aldec.so:cocotbvhpi_entry_point TOPLEVEL_LANG=verilog MODULE=test_mixed_language TESTCASE= TOPLEVEL=endian_swapper_mixed /opt/aldec/rivierapro/bin/vsimsa -do sim_build/runsim.tcl
```

marlonjames, 2022-09-14:
This line:
> `# COUT:    100.00ns DEBUG    gpi                                Searching for i_swapper_vhdl`

Is searching by name. I'm interested in the earlier section of the log that might show where this name was discovered.

marlonjames, 2022-09-14:
My guess is the segfault is occurring in `VhpiObjHdl::initialise()`.

https://github.com/cocotb/cocotb/blob/1ce9390bf7ea270e8ff8c91d436b99236e09ff7b/cocotb/share/lib/vhpi/VhpiCbHdl.cpp#L239-L257

abyszuk, 2022-09-14:
I've updated my post with full simulation log

marlonjames, 2022-09-14:
I found that it is being accessed directly by name by the Python test `mixed_language_accessing_test`.

I don't have a local setup for debugging Riviera+GPI right now. Would you be able to step through that function in gdb and see where it segfaults? Or check the core dump for a backtrace?

abyszuk, 2022-09-14:
Sure, I'll do that tomorrow!

abyszuk, 2022-09-15:
There are actually two coredumps: `core.Live simulation.6455` and `core.Main simulator .6483`, but only the second one is connected to cocotb.
Main simulator backtrace (I suspect only thread 1 is interesting):
```
(gdb) thread apply all bt                                                                                                                                                                                                                                                                
                                                                                                                                                                                                                                                                                         
Thread 8 (Thread 0x7fa326183700 (LWP 6492)):                                                                                                                                                                                                                                             
#0  0x00007fa32b8e99e4 in read () from /lib64/libpthread.so.0                                                                                                                                                                                                                            
#1  0x00007fa32f69f156 in ?? () from /opt/aldec/rivierapro/bin/Linux64/librmi_pipe_connection.so                                                                                                                                                                                         
#2  0x00007fa32f6a4a55 in ?? () from /opt/aldec/rivierapro/bin/Linux64/librmi_pipe_connection.so                                                                                                                                                                                         
#3  0x00007fa32c8e75be in ?? () from /opt/aldec/rivierapro/bin/Linux64/libaldec_boost_thread.so.1.60.0                                                                                                                                                                                   
#4  0x00007fa32b8e017a in start_thread () from /lib64/libpthread.so.0                                                                                                                                                                                                                    
#5  0x00007fa32b60fdc3 in clone () from /lib64/libc.so.6                                                                                                                                                                                                                                 
                                                          
Thread 7 (Thread 0x7fa325982700 (LWP 6494)):
#0  0x00007fa32b5dbd98 in nanosleep () from /lib64/libc.so.6
#1  0x00007fa32b607b28 in usleep () from /lib64/libc.so.6
#2  0x00007fa32f568688 in ?? () from /opt/aldec/rivierapro/bin/Linux64/libriviera_simulator.so
#3  0x00007fa32f5e8b51 in ?? () from /opt/aldec/rivierapro/bin/Linux64/libriviera_simulator.so
#4  0x00007fa32f5a1edc in ?? () from /opt/aldec/rivierapro/bin/Linux64/libriviera_simulator.so
#5  0x00007fa32f4e466f in RmiSupport::RmiMessageProcessor::Impl::processInvokeMethodMessage(RmiSupport::MessageReader&) () from /opt/aldec/rivierapro/bin/Linux64/librmi_support.so
#6  0x00007fa32f4e4af3 in RmiSupport::RmiMessageProcessor::processMessages() () from /opt/aldec/rivierapro/bin/Linux64/librmi_support.so
#7  0x00007fa32f4e4ecf in ?? () from /opt/aldec/rivierapro/bin/Linux64/librmi_support.so
#8  0x00007fa32f6a857d in ?? () from /opt/aldec/rivierapro/bin/Linux64/librmi_pipe_connection.so
#9  0x00007fa32f6a424f in ?? () from /opt/aldec/rivierapro/bin/Linux64/librmi_pipe_connection.so
#10 0x00007fa32c8e75be in ?? () from /opt/aldec/rivierapro/bin/Linux64/libaldec_boost_thread.so.1.60.0
#11 0x00007fa32b8e017a in start_thread () from /lib64/libpthread.so.0
#12 0x00007fa32b60fdc3 in clone () from /lib64/libc.so.6

Thread 6 (Thread 0x7fa30f28a700 (LWP 6507)):
#0  0x00007fa32b8ea5a2 in waitpid () from /lib64/libpthread.so.0
#1  0x0000000006fdab5c in NASDB::cpp::ProcessMonitor::run() ()
#2  0x00007fa32c8e75be in ?? () from /opt/aldec/rivierapro/bin/Linux64/libaldec_boost_thread.so.1.60.0
#3  0x00007fa32b8e017a in start_thread () from /lib64/libpthread.so.0
#4  0x00007fa32b60fdc3 in clone () from /lib64/libc.so.6

Thread 5 (Thread 0x7fa30fa8b700 (LWP 6505)):
#0  0x00007fa32b60729f in select () from /lib64/libc.so.6
#1  0x0000000006fddbca in NASDB::cpp::sys::miliSleep(unsigned int) ()
#2  0x0000000006f9001a in NASDB::KEventWriter::Impl::clientStorageLockThread() ()
#3  0x00007fa32c8e75be in ?? () from /opt/aldec/rivierapro/bin/Linux64/libaldec_boost_thread.so.1.60.0
#4  0x00007fa32b8e017a in start_thread () from /lib64/libpthread.so.0
#5  0x00007fa32b60fdc3 in clone () from /lib64/libc.so.6

Thread 4 (Thread 0x7fa326608980 (LWP 6483)):
#0  0x00007fa32b8e63fc in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x0000000001d70ebb in ?? ()
#2  0x0000000001d179cc in ?? ()
#3  0x00007fa32b536493 in __libc_start_main () from /lib64/libc.so.6
#4  0x0000000001d687a2 in _start ()

Thread 3 (Thread 0x7fa324eab700 (LWP 6498)):
#0  0x00007fa32b8e63fc in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
[... cut ...]
#5  0x00007fa31cce2519 in std::string::assign(std::string const&) () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libgpi.so
#6  0x000000000000005e in ?? ()
#7  0x00007fa3143e06c0 in ?? ()
--Type <RET> for more, q to quit, c to continue without paging--c
#8  0x00007fa3143e0680 in ?? ()
#9  0x00007fa31fffc5f0 in ?? ()
#10 0x00007fa31ccdc92b in GpiObjHdl::initialise(std::string&, std::string&) () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libgpi.so
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

abyszuk, 2022-09-15:
I've been debugging this with `gdb` and it looks like it doesn't even reach `VhpiObjHdl::initialise()`.
Setting breakpoint to `GpiObjHdl::initalise(std::string&, std::string&)` works as intended and gives reasonable backtrace:
```

Thread 5 "Main simulator " hit Breakpoint 1, 0x00007f62b8a2a904 in GpiObjHdl::initialise(std::string&, std::string&) () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libgpi.so
(gdb) bt
#0  0x00007f62b8a2a904 in GpiObjHdl::initialise(std::string&, std::string&) () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libgpi.so
#1  0x00007f62b8c43f0a in VpiObjHdl::initialise(std::string&, std::string&) () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libcocotbvpi_aldec.so
#2  0x00007f62b8c4de7a in VpiImpl::create_gpi_obj_from_handle(unsigned int*, std::string&, std::string&) () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libcocotbvpi_aldec.so
#3  0x00007f62b8c435a7 in VpiIterator::next_handle(std::string&, GpiObjHdl**, void**) () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libcocotbvpi_aldec.so
#4  0x00007f62b8a2a0ae in gpi_next () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libgpi.so
#5  0x00007f629e2dd798 in next((anonymous namespace)::gpi_hdl_Object<GpiIterator*>*) [clone .2688.2685] () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/simulator.cpython-36m-x86_64-linux-gnu.so
#6  0x00007f62a5c4a366 in _PyEval_EvalFrameDefault () from /lib64/libpython3.6m.so.1.0
#7  0x00007f62a5c25e28 in fast_function () from /lib64/libpython3.6m.so.1.0
#8  0x00007f62a5c49397 in call_function () from /lib64/libpython3.6m.so.1.0
#9  0x00007f62a5c49fd8 in _PyEval_EvalFrameDefault () from /lib64/libpython3.6m.so.1.0
#10 0x00007f62a5bce3c4 in gen_send_ex () from /lib64/libpython3.6m.so.1.0
#11 0x00007f62a5c4926d in call_function () from /lib64/libpython3.6m.so.1.0
#12 0x00007f62a5c49fd8 in _PyEval_EvalFrameDefault () from /lib64/libpython3.6m.so.1.0
#13 0x00007f62a5c25e28 in fast_function () from /lib64/libpython3.6m.so.1.0
#14 0x00007f62a5c49397 in call_function () from /lib64/libpython3.6m.so.1.0
#15 0x00007f62a5c49fd8 in _PyEval_EvalFrameDefault () from /lib64/libpython3.6m.so.1.0
#16 0x00007f62a5c25e28 in fast_function () from /lib64/libpython3.6m.so.1.0
#17 0x00007f62a5c49397 in call_function () from /lib64/libpython3.6m.so.1.0
#18 0x00007f62a5c49fd8 in _PyEval_EvalFrameDefault () from /lib64/libpython3.6m.so.1.0
#19 0x00007f62a5ba5ba4 in _PyEval_EvalCodeWithName () from /lib64/libpython3.6m.so.1.0
#20 0x00007f62a5c26010 in fast_function () from /lib64/libpython3.6m.so.1.0
#21 0x00007f62a5c49397 in call_function () from /lib64/libpython3.6m.so.1.0
#22 0x00007f62a5c4ac18 in _PyEval_EvalFrameDefault () from /lib64/libpython3.6m.so.1.0
#23 0x00007f62a5c25e28 in fast_function () from /lib64/libpython3.6m.so.1.0
#24 0x00007f62a5c49397 in call_function () from /lib64/libpython3.6m.so.1.0
#25 0x00007f62a5c49fd8 in _PyEval_EvalFrameDefault () from /lib64/libpython3.6m.so.1.0
#26 0x00007f62a5ba70b2 in _PyFunction_FastCallDict () from /lib64/libpython3.6m.so.1.0
#27 0x00007f62a5ba7e8e in _PyObject_FastCallDict () from /lib64/libpython3.6m.so.1.0
#28 0x00007f62a5bb9da0 in method_call () from /lib64/libpython3.6m.so.1.0
#29 0x00007f62a5baec0b in PyObject_Call () from /lib64/libpython3.6m.so.1.0
#30 0x00007f629e2dccbe in handle_gpi_callback(void*) [clone .2837] () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/simulator.cpython-36m-x86_64-linux-gnu.so
#31 0x00007f62b8a2a99c in GpiCbHdl::run_callback() () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libgpi.so
#32 0x00007f62b8c47450 in handle_vpi_callback () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libcocotbvpi_aldec.so
#33 0x00007f62c40adad1 in ?? () from /opt/aldec/rivierapro/bin/Linux64/libaldecpli.so
#34 0x00007f62c40b925f in ?? () from /opt/aldec/rivierapro/bin/Linux64/libaldecpli.so
#35 0x0000000002bd9948 in ?? ()
#36 0x00000000029634c7 in ?? ()
#37 0x000000000284dc88 in ?? ()
#38 0x0000000002602020 in ?? ()
#39 0x0000000003009398 in ?? ()
#40 0x00007f62c142617a in start_thread () from /lib64/libpthread.so.0
#41 0x00007f62c1155dc3 in clone () from /lib64/libc.so.6
(...cut...)

Thread 5 "Main simulator " hit Breakpoint 1, 0x00007f62b8a2a904 in GpiObjHdl::initialise(std::string&, std::string&) () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libgpi.so
(gdb) c
Continuing.

Thread 5 "Main simulator " received signal SIGABRT, Aborted.
0x00007f62c109037f in raise () from /lib64/libc.so.6
(gdb) bt
#0  0x00007f62c109037f in raise () from /lib64/libc.so.6
#1  0x00007f62c107adb5 in abort () from /lib64/libc.so.6
#2  0x00007f62c10d34e7 in __libc_message () from /lib64/libc.so.6
#3  0x00007f62c10da5ec in malloc_printerr () from /lib64/libc.so.6
#4  0x00007f62c10dbeec in _int_free () from /lib64/libc.so.6
#5  0x00007f62b8a30519 in std::string::assign(std::string const&) () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libgpi.so
#6  0x000000000000005e in ?? ()
#7  0x00007f62a8852d60 in ?? ()
#8  0x00007f62a8852d20 in ?? ()
#9  0x00007f62b3ffc5f0 in ?? ()
#10 0x00007f62b8a2a92b in GpiObjHdl::initialise(std::string&, std::string&) () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libgpi.so
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb) 

```

marlonjames, 2022-09-15:
If you step through `VhpiObjHdl::initialise()` does it segfault inside the call to `GpiObjHdl::initialise()` or is that from some later point in time?

marlonjames, 2022-09-15:
So I just checked the cocotb extended regression tests and found that in our setup this test passes using the same version of Riviera-PRO:

```
run -all
# COUT:      0.00ns INFO     cocotb                             Running on Riviera-PRO version 2021.04.107.8106
# COUT:      0.00ns INFO     cocotb                             Running tests with cocotb v1.8.0.dev0 from /home/az_devops_agent/agent/_work/3/s/cocotb
# COUT:      0.00ns INFO     cocotb                             Seeding Python random module with 1662891843
# COUT:      0.00ns INFO     cocotb.regression                  Found test test_mixed_language.mixed_language_accessing_test
# COUT:      0.00ns INFO     cocotb.regression                  Found test test_mixed_language.mixed_language_functional_test
# COUT:      0.00ns INFO     cocotb.regression                  running mixed_language_accessing_test (1/2)
# COUT:                                                           Try accessing handles and setting values in a mixed language environment.
# COUT:    100.00ns INFO     cocotb.endian_swapper_mixed        Got: 'i_swapper_sv'
# COUT:    100.00ns WARNING  gpi                                VPI: Simulator does not know this type (1029) via VPI
# COUT:    100.00ns INFO     cocotb.endian_swapper_mixed        Got: 'i_swapper_vhdl'
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# COUT:    300.00ns INFO     cocotb.regression                  mixed_language_accessing_test passed
# COUT:    300.00ns INFO     cocotb.regression                  running mixed_language_functional_test (2/2)
# COUT:                                                           Try concurrent simulation of VHDL and Verilog and check the output.
# COUT:    400.00ns INFO     cocotb.endian_swapper_mixed        Got: 'i_swapper_sv'
# COUT:    400.00ns INFO     cocotb.endian_swapper_mixed        Got: 'i_swapper_vhdl'
# COUT:   2690.00ns INFO     cocotb.regression                  mixed_language_functional_test passed
# COUT:   2690.00ns INFO     cocotb.regression                  ************************************************************************************************************
# COUT:                                                         ** TEST                                                STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
# COUT:                                                         ************************************************************************************************************
# COUT:                                                         ** test_mixed_language.mixed_language_accessing_test    PASS         300.00           4.37         68.59  **
# COUT:                                                         ** test_mixed_language.mixed_language_functional_test   PASS        2390.00           1.88       1270.02  **
# COUT:                                                         ************************************************************************************************************
# COUT:                                                         ** TESTS=2 PASS=2 FAIL=0 SKIP=0                                     2690.00           6.82        394.57  **
# COUT:                                                         ************************************************************************************************************
# COUT:                                                         
# RUNTIME: Info: RUNTIME_0068 $finish called.
# KERNEL: Time: 2690002 ps,  Iteration: 0,  TOP instance.
# KERNEL: stopped at time: 2690002 ps
# VSIM: Simulation has finished. There are no more test vectors to simulate.
endsim
# VSIM: Simulation has finished.
```

abyszuk, 2022-09-15:
Yes, it segfaults inside `GpiObjHdl::initialise()`. By stepping through the code looks it fails on first assignment:
```
(gdb) nexti
0x00007f032c88a926 in GpiObjHdl::initialise(std::string&, std::string&) () from /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libgpi.so
1: x/i $pc
=> 0x7f032c88a926 <_ZN9GpiObjHdl10initialiseERSsS0_+38>:        callq  0x7f032c886ae0 <_ZNSs6assignERKSs@plt>
(gdb) nexti

Thread 5 "Main simulator " received signal SIGABRT, Aborted.
0x00007f0334ef037f in raise () from /lib64/libc.so.6
1: x/i $pc
=> 0x7f0334ef037f <raise+271>:  mov    0x108(%rsp),%rc
```

abyszuk, 2022-09-15:
> So I just checked the cocotb extended regression tests and found that in our setup this test passes using the same version of Riviera-PRO:

OK, so what does this mean? `libstdc++` issue? I run Riviera inside docker container based on CentOs 8.

marlonjames, 2022-09-15:
The error is in libc `__intf_free()`.

> `# COUT: free(): invalid pointer`

Here is snippet of the source:
```c
static void
_int_free (mstate av, mchunkptr p, int have_lock)
{
  INTERNAL_SIZE_T size;        /* its size */
  mfastbinptr *fb;             /* associated fastbin */
  mchunkptr nextchunk;         /* next contiguous chunk */
  INTERNAL_SIZE_T nextsize;    /* its size */
  int nextinuse;               /* true if nextchunk is used */
  INTERNAL_SIZE_T prevsize;    /* size of previous contiguous chunk */
  mchunkptr bck;               /* misc temp for linking */
  mchunkptr fwd;               /* misc temp for linking */

  size = chunksize (p);

  /* Little security check which won't hurt performance: the
     allocator never wrapps around at the end of the address space.
     Therefore we can exclude some size values which might appear
     here by accident or by "design" from some intruder.  */
  if (__builtin_expect ((uintptr_t) p > (uintptr_t) -size, 0)
      || __builtin_expect (misaligned_chunk (p), 0))
    malloc_printerr ("free(): invalid pointer");

  ...
}
```

marlonjames, 2022-09-15:
> I run Riviera inside docker container based on CentOs 8.

What version of cocotb are you using and how did you install it in the container?

abyszuk, 2022-09-15:
> > I run Riviera inside docker container based on CentOs 8.
> 
> What version of cocotb are you using and how did you install it in the container?

1.7.0, installed via `pip3 install cocotb`

marlonjames, 2022-09-15:
I wonder if installing from source would make any difference, rather than using a wheel.

You could try https://docs.cocotb.org/en/latest/install_devel.html#install-devel if you have the development tools available in your container.

abyszuk, 2022-09-16:
Yes, it does make a difference, thanks!
Installing by `pip3 install git+https://github.com/cocotb/cocotb@v1.7.0 --user` makes it pass the test just fine.
So, is it something about the way a wheel is built? gcc/libstdc++ version?

imphil, 2022-09-16:
@abyszuk can you check which wheels were picked up for your system? Do a "pip install --verbose cocotb" and check its output, you should see something like (depending on your system state):

```
❯ pip3 install --verbose cocotb
Using pip 22.2.1 from /home/philipp/test-venv3/lib64/python3.10/site-packages/pip (python 3.10)
Collecting cocotb
  Using cached cocotb-1.7.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (3.7 MB)
```

It's the name of the whl file I'm interested in.

abyszuk, 2022-09-16:
I've just checked that this issue probably isn't related to #3080 
I've installed `yum install libstdc++-static` and reinstalled cocotb from wheel, but it didn't change anything.
Curious thing is that it I didn't get any linker errors even without that library installed.

abyszuk, 2022-09-16:
@imphil 
```
  Using version 1.7.0 (newest of versions: 1.2.0, 1.3.0, 1.3.1, 1.3.2, 1.4.0, 1.5.0, 1.5.1, 1.5.2, 1.6.0, 1.6.1, 1.6.2, 1.7.0)
  Looking up "https://files.pythonhosted.org/packages/a9/a7/ab79269dd8cb05427038a770d2957b470dbbcda4fad9c26b8440a2578e3b/cocotb-1.7.0-cp36-cp36m-manylinux_2_5_x86_64.manylinux1_x86_64.whl" in the cache
  Current age based on date: 262307
  Freshness lifetime from max-age: 365000000
  The response is "fresh", returning cached response
  365000000 > 262307
  Using cached https://files.pythonhosted.org/packages/a9/a7/ab79269dd8cb05427038a770d2957b470dbbcda4fad9c26b8440a2578e3b/cocotb-1.7.0-cp36-cp36m-manylinux_2_5_x86_64.manylinux1_x86_64.whl
  Downloading from URL https://files.pythonhosted.org/packages/a9/a7/ab79269dd8cb05427038a770d2957b470dbbcda4fad9c26b8440a2578e3b/cocotb-1.7.0-cp36-cp36m-manylinux_2_5_x86_64.manylinux1_x86_64.whl#sha256=e6a0bbf1311a9e3e95be71c1f19d56ab950fafe23d6d80f593cfbd8c272c1901 (from https://pypi.org/simple/cocotb/) (requires-python:>=3.6)
Installing collected packages: cocotb
```

imphil, 2022-09-16:
@abyszuk https://github.com/cocotb/cocotb/issues/3080 is only relevant when building cocotb from source. The prebuilt wheels statically link (required parts of) libstdc++ at build time, hence at runtime (i.e. when installing cocotb from a wheel), you don't need any libstdc++.

I have a suspicion that we have an incompatibility between statically linked symbols from libstdc++ within cocotb, and conflicting symbols from another libstdc++ version.

imphil, 2022-09-16:
You're using cocotb-1.7.0-cp36-cp36m-manylinux_2_5_x86_64.manylinux1_x86_64.whl, which is built on RHEL5. 

As an experiment, can you use a newer Python version than 3.6 (CentOS should have them available as SCLs) and try again? Using a newer Python version will give you a different wheel, built on a more recent distribution, also with a more recent libstdc++ linked in.

abyszuk, 2022-09-16:
I've uninstalled cocotb, changed Python to 3.9 and installed cocotb again
```
pip3 uninstall cocotb
dnf module install python39
alternatives --config python3
pip3 install cocotb --user --verbose
```
Wheel install:
```
Collecting cocotb
  Created temporary directory: /tmp/pip-unpack-otqwhxbk
  Looking up "https://files.pythonhosted.org/packages/77/90/5a819a2eaa92fbdd53e5031b86ef49d610cb8283cea2efa5e22864ce7b14/cocotb-1.7.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl" in the cache
  No cache entry available
  Starting new HTTPS connection (1): files.pythonhosted.org:443
  https://files.pythonhosted.org:443 "GET /packages/77/90/5a819a2eaa92fbdd53e5031b86ef49d610cb8283cea2efa5e22864ce7b14/cocotb-1.7.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl HTTP/1.1" 200 3695683
  Downloading cocotb-1.7.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (3.7 MB)
     |████████████████████████████████| 3.7 MB 6.4 MB/s eta 0:00:01  Ignoring unknown cache-control directive: immutable
  Updating cache with response from "https://files.pythonhosted.org/packages/77/90/5a819a2eaa92fbdd53e5031b86ef49d610cb8283cea2efa5e22864ce7b14/cocotb-1.7.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl"
  Caching due to etag
     |████████████████████████████████| 3.7 MB 6.4 MB/s
  Added cocotb from https://files.pythonhosted.org/packages/77/90/5a819a2eaa92fbdd53e5031b86ef49d610cb8283cea2efa5e22864ce7b14/cocotb-1.7.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl#sha256=6eded67546534bb51dd7b2f95ea0e2f7ac90579a7fad72a170fc796d05ac775a to build tracker '/tmp/pip-req-tracker-1rsv85er'
  Removed cocotb from https://files.pythonhosted.org/packages/77/90/5a819a2eaa92fbdd53e5031b86ef49d610cb8283cea2efa5e22864ce7b14/cocotb-1.7.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl#sha256=6eded67546534bb51dd7b2f95ea0e2f7ac90579a7fad72a170fc796d05ac775a from build tracker '/tmp/pip-req-tracker-1rsv8
5er'
Installing collected packages: cocotb
```

However now the simulation fails with:
```
asim   +access +w -interceptcoutput -O2 -dbg -pli /home/adrian/.local/lib/python3.9/site-packages/cocotb/libs/libcocotbvpi_aldec.so endian_swapper_mixed
# ELBREAD: Elaboration process.
# ELBREAD: Elaboration time 0.0 [s].
# KERNEL: Main thread initiated.
# KERNEL: Kernel process initialization phase.
# ELAB2: Elaboration final pass...
# KERNEL: PLI/VHPI kernel's engine initialization done.
# PLI: Loading library '/home/adrian/.local/lib/python3.9/site-packages/cocotb/libs/libcocotbvpi_aldec.so'
# COUT:      -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
# COUT:      -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
# COUT:      -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VHPI registered
# PLI: Loading library '/opt/aldec/rivierapro/bin/libsystf.so'
# ELAB2: Create instances ...
# KERNEL: Time resolution set to 1ps.
# ELAB2: Create instances complete.
# ELAB2: Elaboration final pass complete - time: 0.0 [s].
# KERNEL: Kernel process initialization done.
# Allocation: Simulator allocated 6642 kB (elbread=427 elab2=6064 kernel=150 sdf=0)
# KERNEL: ASDB file was created in location /afs/cern.ch/work/a/adbyszuk/praca/temp/cocotb/examples/mixed_language/tests/dataset.asdb
run -all
# COUT: ModuleNotFoundError: No module named 'pygpi'
```

abyszuk, 2022-09-28:
@imphil I've tried debugging this `ModuleNotFoundError` and it could be a problem in `cocotb-config`, but that's just my guess.

```
adrian@pcte247806:tests$ cocotb-config --libpython
/usr/lib64/libpython3.so
```
However, having multiple python versions installed, there are also multiple `libpython` binaries:
```
adrian@pcte247806:/$ ls -l /usr/lib64/libpython3.so 
-rwxr-xr-x 1 root root 6888 Sep 10  2021 /usr/lib64/libpython3.so

adrian@pcte247806:/$ ls -l /lib64/libpython3.9.so.1.0 
-rwxr-xr-x 1 root root 3918968 Aug 25  2021 /lib64/libpython3.9.so.1.0

adrian@pcte247806:/$ ls -l /lib64/libpython3.6m.so.1.0 
-rwxr-xr-x 1 root root 3289680 Sep 10  2021 /lib64/libpython3.6m.so.1.0
```
Any idea how can I debug this further?

imphil, 2022-09-28:
The module loading error that you resolved by setting LIBPYTHON_LOC seems unrelated to the segfault issue that we discussed earlier. I'd suggest to keep this issue to the segfault; I opened https://github.com/cocotb/cocotb/issues/3097 for the other issue.

On the segfault: I can reproduce the issue on my side, but I haven't yet found enough time to deep-dive into it. My first suspicion is that we're freeing a piece of memory that was allocated by another libstdc++ (we link one statically into cocotb, Aldec ships its own version dynamically linked, and of course the operating system provides another one). But again, that's a hypothesis which needs checking.

abyszuk, 2022-09-28:
Thank you, let me know if I can help!

imphil, 2022-10-13:
Let's keep this open for a backport into 1.7.

imphil, 2022-10-14:
@abyszuk can you help us confirm that the issue is indeed resolved in master?

First, download https://github.com/cocotb/cocotb/suites/8784257799/artifacts/398860678, which contains wheels from the lastest master release.

Then install cocotb from these wheels:

```
mkdir testdir
unzip cocotb-dist.zip
pip3 install --only-binary=cocotb --find-links . cocotb==1.8.0.dev0
```

And then re-run the test on your machine.

abyszuk, 2022-10-28:
@imphil sorry for late response.
Looks like it's working! Thank you very much for your help!
```
asim   +access +w -interceptcoutput -O2 -dbg -pli /home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libcocotbvpi_aldec.so endian_swapper_mixed 
# ELBREAD: Elaboration process.
# ELBREAD: Elaboration time 0.0 [s].
# KERNEL: Main thread initiated.
# KERNEL: Kernel process initialization phase.
# ELAB2: Elaboration final pass...
# KERNEL: PLI/VHPI kernel's engine initialization done.
# PLI: Loading library '/home/adrian/.local/lib/python3.6/site-packages/cocotb/libs/libcocotbvpi_aldec.so'
# COUT:      -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
# COUT:      -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
# COUT:      -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VHPI registered
# PLI: Loading library '/opt/aldec/rivierapro/bin/libsystf.so'
# ELAB2: Create instances ...
# KERNEL: Time resolution set to 1ps.
# ELAB2: Create instances complete.
# ELAB2: Elaboration final pass complete - time: 0.1 [s].
# KERNEL: Kernel process initialization done.
# Allocation: Simulator allocated 6642 kB (elbread=427 elab2=6064 kernel=150 sdf=0)
# KERNEL: ASDB file was created in location /afs/cern.ch/work/a/adbyszuk/praca/temp/cocotb/examples/mixed_language/tests/dataset.asdb
run -all
# COUT:      0.00ns INFO     cocotb                             Running on Riviera-PRO version 2021.04.107.8106
# COUT:      0.00ns INFO     cocotb                             Running tests with cocotb v1.8.0.dev0 from /home/adrian/.local/lib/python3.6/site-packages/cocotb
# COUT:      0.00ns INFO     cocotb                             Seeding Python random module with 1666952181
# COUT:      0.00ns INFO     cocotb.regression                  pytest not found, install it to enable better AssertionError messages
# COUT:      0.00ns INFO     cocotb.regression                  Found test test_mixed_language.mixed_language_accessing_test
# COUT:      0.00ns INFO     cocotb.regression                  Found test test_mixed_language.mixed_language_functional_test
# COUT:      0.00ns INFO     cocotb.regression                  running mixed_language_accessing_test (1/2)
# COUT:                                                           Try accessing handles and setting values in a mixed language environment.
# COUT:    100.00ns INFO     cocotb.endian_swapper_mixed        Got: 'i_swapper_sv'
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 39004: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# COUT:    100.00ns WARNING  gpi                                VPI: Simulator does not know this type (1029) via VPI
# COUT:    100.00ns INFO     cocotb.endian_swapper_mixed        Got: 'i_swapper_vhdl'
# COUT:    300.00ns INFO     cocotb.regression                  mixed_language_accessing_test passed
# COUT:    300.00ns INFO     cocotb.regression                  running mixed_language_functional_test (2/2)
[... cut ...]
# COUT:                                                         ************************************************************************************************************
# COUT:                                                         
# RUNTIME: Info: RUNTIME_0068 $finish called.
# KERNEL: Time: 2690002 ps,  Iteration: 0,  TOP instance.
# KERNEL: stopped at time: 2690002 ps
# VSIM: Simulation has finished. There are no more test vectors to simulate.
endsim
# VSIM: Simulation has finished.
```

imphil, 2022-11-15:
In addition to being fixed on master, the required fix has also been backported to the 1.7.2 release, which was just published. Please reopen this issue if the problem persists.

closed 2022-11-15 by imphil