cocotb/cocotb #3083: Default `libpythonX.X.so` not found with binary install

marlonjames, 2022-09-16:
cocotb 1.7.0

At least two users have reported problems loading the `libpython` shared library when using cocotb 1.7 binary wheel installation.

They did not set [LIBPYTHON_LOC](https://docs.cocotb.org/en/stable/building.html#envvar-LIBPYTHON_LOC) and were relying on the default `libpythonX.X.so` being found by `dlopen()`.

If the correct directory is added to `LD_LIBRARY_PATH` there is no issue.

Also, when building 1.7 from source locally, there is no issue finding the default `libpythonX.X.so`.

---

[Gitter discussion](https://gitter.im/cocotb/Lobby?at=632388257ccf6b6d45d857a7):

@jwprice100
Hmm with cocotb 1.7 I'm getting issues not being able to find the python shared library. This is probably a consequent of my setup but was not an issue in 1.6 and prior. Did anything relevant change that I can look at? Note I do make use of virtual environments

Adding the actual location of the so to the LD_LIBRARY_PATH works but I'm concerned something broke with some kind of path resolution

@marlonjames
`cocotb-config --libpython` should print out the path that it uses

@jwprice100
hmmm prints out the same path I added to my ld_library_path. However it uses a different libpython so.
```
cocotb-config --libpython
/opt/tools/python/3.9.6/lib/libpython3.9.so.1.0
```
```
COUT:      -.--ns ERROR    gpi                                ..s/cocotb_utils.cpp:68   in utils_dyn_open                  Unable to open lib libpython3.9.so: libpython3.9.so: cannot open shared object file: No such file or directory
```
Not sure if thats helpful in debugging
but if I just added /opt/tools/python/3.9.6/lib/ to my LD_LIBRARY_PATH it works fine

imphil, 2022-09-16:
@marlonjames is the following understanding right?
* Users have Python somewhere where `dlopen()` doesn't find it.
* When building from source, we're baking that non-standard library path into the cocotb libraries.
* With wheels, users need another mechanism to inform cocotb where to find libpython. LIBPYTHON_LOC and LD_LIBRARY_PATH serve that purpose.

Is it the case that users have the `python` binary from their non-standard installation in their PATH, but not the libpython.so in their library search path? [We might be able to use that as a discovery mechanism for libpython?]

marlonjames, 2022-09-16:
That is my understanding.

I'm not sure about this:
> Is it the case that users have the python binary from their non-standard installation in their PATH, but not the libpython.so in their library search path?

pinging @jwprice100

jwprice100, 2022-09-17:
Marlon is correct. We have many python versions installed in a central
location. From there we can pick a python version and we use that version
to make a local virtual environment. LIBPYTHON_LOC is not something I've
ever explicitly set. LD_LIBRARY_PATH does not have the libpython location
on its shared path. I am not aware of any other aspects of our setup that
may be exotic but please feel free to ask questions and I'll report
whatever I can learn.

imphil, 2022-09-24:
I just refreshed my mind on what we actually do behind the scenes, and we actually do more magic than I remembered to find the right libpython, and set LIBPYTHON_LOC accordingly.

@jwprice100 What do you get when you run `cocotb-config --libpython`? And if that's not correct, can you do a bit of digging in the find_libpython library, e..g by running `python3 -c 'from cocotb._vendor import find_libpython; find_libpython.main()' --list-all` (run ` --help`  for more options)

imphil, 2022-09-24:
And to confirm, @jwprice100, you're using the cocotb Makefiles? Or something else?

ktbarrett, 2022-11-25:
@jwprice100 Has this been resolved with cocotb v1.7.2?

The `libpythonX.X.so` should only be attempted to be opened by `dlopen` if `LIBPYTHON_LOC` were not set. The makefiles will error if `cocotb-config --libpython` fails, so the only rationale is that something is deleting that environment variable between the makefiles and the attempt to load libpython in the cocotb startup phase.

https://github.com/cocotb/cocotb/blob/196f26c17ea5de2868ccc09180a3f66e310458bf/cocotb/share/lib/embed/embed.cpp#L58-L62

closed 2023-01-02 by github-actions[bot]