cocotb/cocotb #3085: Python UVM setup issue

Skavarei, 2022-09-19:
Hello,

I'm trying to run a basic example using Cocotb, but I'm having compilation problems. Could you kindly assist me in fixing this?

Below three files I'm using for example:

`import cocotb
from cocotb.triggers import Timer
from uvm import *

class NewTest(UVMTest):
    
    async def run_phase(self, phase):
        phase.raise_objection(self)
        await Timer(100, "NS")
        phase.drop_objection(self)

uvm_component_utils(NewTest)

class NewTest2(NewTest):
    pass

@cocotb.test()
async def test_dut(dut):
    await run_test('NewTest')`

``timescale 1ns/1ns
module new_dut(input clk, input rst, output[7:0] byte_out);
    assign byte_out = 8'hAB;
endmodule: new_dut`

`# Minimal Makefile for uvm-python minimal example
TOPLEVEL_LANG ?= verilog
VERILOG_SOURCES ?= new_dut.sv
TOPLEVEL := new_dut
MODULE   ?= new_test

PYTHONPATH := ../../../src:$(PYTHONPATH)
export PYTHONPATH

include $(shell cocotb-config --makefiles)/Makefile.sim`

Before running the simulation, I am setting up the below environment variables:

[image]

After performing the "make" (run) command, I'm getting below error:

     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
Traceback (most recent call last):
  File "/tools/gnu/python/DV_eng/3.7.7/lib/python3.7/site-packages/cocotb/__init__.py", line 41, in <module>
    import cocotb.handle
  File "/tools/gnu/python/DV_eng/3.7.7/lib/python3.7/site-packages/cocotb/handle.py", line 35, in <module>
    if sys.version_info.major < 3:
AttributeError: 'tuple' object has no attribute 'major'
     -.--ns ERROR    cocotb.gpi                                  gpi_embed.c:224  in get_module_ref                  Failed to load Python module "cocotb"

cmarqu, 2022-09-19:
You seem to be getting some old Python 2.6 (or even earlier) in the mix.
Maybe `python` is this old Python 2 and `python3` is Python 3?

Skavarei, 2022-09-20:
> You seem to be getting some old Python 2.6 (or even earlier) in the mix. Maybe `python` is this old Python 2 and `python3` is Python 3?

Using Python 3

**Python 3.7.7** (default, Apr 23 2020, 06:14:54) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-17)] on linux
Type "help", "copyright", "credits" or "license" for more information.

marlonjames, 2022-09-20:
What version of cocotb are you using and how did you install it?

imphil, 2022-09-21:
@Skavarei `sys.version_info.major` was added in Python 3.1 (https://docs.python.org/3/library/sys.html#sys.version_info), hence somehow you're not using Python 3.7 to execute the test, as Colin and Marlon pointed out. So you'll need to do some digging how an older Python version got into your environment. It's a good idea to re-install cocotb and write down what you did, and then post that here. Or maybe the problem disappears (or you find the issue) along the way.

closed 2022-10-30 by github-actions[bot]