cocotb/cocotb #3094: Force function take no effect on internal signals

sordhlm, 2022-09-27:
Hi, 
I am using Force() function to force internal signals. But I found it has no effect. 
Force() does work for those input signals without HDL drive.
Since force function in systemverilog can force internal signals, so I am wondering if cocotb can support this?

imphil, 2022-09-27:
Can you show a small code snippet of what you're trying to do?

sordhlm, 2022-09-28:
@imphil 
[image]
Like the DUT code above, it's ok to force rx signal. But I can't force counter reg. However during Asic verification, we are often required to force some internal signal for some special test.

marlonjames, 2022-09-28:
What simulator are you using? You may need to provide special flags for accessing/modifying the internals of a design while the simulation is running.
cocotb's `dut.sig_name.value = Force(<value>)` uses `vpi_put_value()` with `vpiForceFlag` set.
That is supposed to be:
>same as the SystemVerilog procedural **force**

imphil, 2022-09-28:
@sordhlm Can you produce a small runnable example (your Verilog code + a testbench), so we can test it on various simulators?  Thanks so much for your help!

sordhlm, 2022-09-29:
@imphil Thanks for keep watching.
I'm afraid I have made a mistake. By creating a simple test code to try Force function, this works for internal and output signals. 
I am going to close this ticket. Thanks for the help!

closed 2022-09-29 by sordhlm