cocotb/cocotb #3095: Aldec Riviera: Signals not detected correctly when using busses

Botnic, 2022-09-27:
Simulator: Aldec Riviera 64bit
Python: 3.10.0

I would like to report an issue originally detected by [@jackbpdx](https://github.com/jackbpdx) and reported in [cocotbext-axi/issues/42](https://github.com/alexforencich/cocotbext-axi/issues/42):

When using the simulator Riviera-PRO (version 2022.04.117.8517 built for Linux64) and accessing a signal over the cocotb-bus interface the signal is not detected correctly. 
If the signal is written prior assigning it to the bus, it works correctly. The issue was detected with the cocotbext-axi but can also be reproduced by other bus classes.

I'm creating a minimal example for you to reproduce the issue.

Botnic, 2022-09-27:
Test based on axis:

[riviera_example.zip](https://github.com/cocotb/cocotb/files/9656755/riviera_example.zip)

ktbarrett, 2022-10-04:
Is this a problem with cocotb, or with [cocotb-bus](https://github.com/cocotb/cocotb-bus).

alexforencich, 2022-10-04:
To me, TBH, it smells like a PLI issue of some sort.  Not sure if cocotb is doing something incorrectly, or if the simulator is implementing some API incorrectly.

Botnic, 2022-10-04:
I can not help with debugging this issue further at the moment. My trial license of Riviera has expired :-(.

ktbarrett, 2022-10-05:
@Botnic FOSSi has Riviera licenses for cocotb. You can open a draft PR with a minimal test which reproduces the error, then we can go from there.

jackbpdx, 2022-10-21:
I rolled back cocotb-bus to 0.1.1 and the issue goes away for me.