cocotb/cocotb #3097: cocotb 1.7.1 wheel install fails with "ModuleNotFoundError: No module named 'pygpi'"

imphil, 2022-09-28:
I've uninstalled cocotb, changed Python to 3.9 and installed cocotb again
```
pip3 uninstall cocotb
dnf module install python39
alternatives --config python3
pip3 install cocotb --user --verbose
```
Wheel install:
```
Collecting cocotb
  Created temporary directory: /tmp/pip-unpack-otqwhxbk
  Looking up "https://files.pythonhosted.org/packages/77/90/5a819a2eaa92fbdd53e5031b86ef49d610cb8283cea2efa5e22864ce7b14/cocotb-1.7.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl" in the cache
  No cache entry available
  Starting new HTTPS connection (1): files.pythonhosted.org:443
  https://files.pythonhosted.org:443 "GET /packages/77/90/5a819a2eaa92fbdd53e5031b86ef49d610cb8283cea2efa5e22864ce7b14/cocotb-1.7.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl HTTP/1.1" 200 3695683
  Downloading cocotb-1.7.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (3.7 MB)
     |████████████████████████████████| 3.7 MB 6.4 MB/s eta 0:00:01  Ignoring unknown cache-control directive: immutable
  Updating cache with response from "https://files.pythonhosted.org/packages/77/90/5a819a2eaa92fbdd53e5031b86ef49d610cb8283cea2efa5e22864ce7b14/cocotb-1.7.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl"
  Caching due to etag
     |████████████████████████████████| 3.7 MB 6.4 MB/s
  Added cocotb from https://files.pythonhosted.org/packages/77/90/5a819a2eaa92fbdd53e5031b86ef49d610cb8283cea2efa5e22864ce7b14/cocotb-1.7.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl#sha256=6eded67546534bb51dd7b2f95ea0e2f7ac90579a7fad72a170fc796d05ac775a to build tracker '/tmp/pip-req-tracker-1rsv85er'
  Removed cocotb from https://files.pythonhosted.org/packages/77/90/5a819a2eaa92fbdd53e5031b86ef49d610cb8283cea2efa5e22864ce7b14/cocotb-1.7.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl#sha256=6eded67546534bb51dd7b2f95ea0e2f7ac90579a7fad72a170fc796d05ac775a from build tracker '/tmp/pip-req-tracker-1rsv8
5er'
Installing collected packages: cocotb
```

However now the simulation fails with:
```
asim   +access +w -interceptcoutput -O2 -dbg -pli /home/adrian/.local/lib/python3.9/site-packages/cocotb/libs/libcocotbvpi_aldec.so endian_swapper_mixed
# ELBREAD: Elaboration process.
# ELBREAD: Elaboration time 0.0 [s].
# KERNEL: Main thread initiated.
# KERNEL: Kernel process initialization phase.
# ELAB2: Elaboration final pass...
# KERNEL: PLI/VHPI kernel's engine initialization done.
# PLI: Loading library '/home/adrian/.local/lib/python3.9/site-packages/cocotb/libs/libcocotbvpi_aldec.so'
# COUT:      -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
# COUT:      -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
# COUT:      -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VHPI registered
# PLI: Loading library '/opt/aldec/rivierapro/bin/libsystf.so'
# ELAB2: Create instances ...
# KERNEL: Time resolution set to 1ps.
# ELAB2: Create instances complete.
# ELAB2: Elaboration final pass complete - time: 0.0 [s].
# KERNEL: Kernel process initialization done.
# Allocation: Simulator allocated 6642 kB (elbread=427 elab2=6064 kernel=150 sdf=0)
# KERNEL: ASDB file was created in location /afs/cern.ch/work/a/adbyszuk/praca/temp/cocotb/examples/mixed_language/tests/dataset.asdb
run -all
# COUT: ModuleNotFoundError: No module named 'pygpi'
```

_Originally posted by @abyszuk in https://github.com/cocotb/cocotb/issues/3078#issuecomment-1249218313_

abyszuk, 2022-09-28:
```
adrian@pcte247806:tests$ python3 -c 'from cocotb._vendor import find_libpython; find_libpython.main()' --list-all
/usr/lib64/libpython3.so
```
What I find a bit strange is that `/usr/lib64/libpython3.so` library is suspiciously small. Could it be a shim library that's just redirecting to the proper one?

abyszuk, 2022-09-28:
A bit more digging:
```
adrian@pcte247806:tests$ python3 -c 'from cocotb._vendor import find_libpython; find_libpython.main()' --candidate-paths
/usr/lib64/python3.9/config-3.9-x86_64-linux-gnu/libpython3.9.so
/usr/lib64/python3.9/config-3.9-x86_64-linux-gnu/libpython3.so
/usr/lib64/python3.9/config-3.9-x86_64-linux-gnu/libpython.so
/usr/lib64/libpython3.9.so
/usr/lib64/libpython3.so
/usr/lib64/libpython.so
/usr/lib/libpython3.9.so
/usr/lib/libpython3.so
/usr/lib/libpython.so
/usr/libpython3.9.so
/usr/libpython3.so
/usr/libpython.so
```
It doesn't list libraries available in `/lib64/`

marlonjames, 2022-09-28:
On your system, are `/lib64/` and `/usr/lib64/` separate directories? on many systems they have been merged:
```
$ ls -l /lib64
lrwxrwxrwx   1 root root      9 Aug  4  2020 lib64 -> usr/lib64/
```

imphil, 2022-09-28:
If this is an RPM-based system, can you check that the python3 binary you're calling, and the libpython library that find_libpython provides you are from the same package?

Some of the following commands might help:

```shell-session
$ command -v python3
$ # Get the package containing python3
$ rpm -qf /usr/bin/python3
$ # get the files in this package
$ rpm -q --filesbypkg python310-base-3.10.7-2.1.x86_64
```

abyszuk, 2022-09-28:
@marlonjames yes, you're right. `/lib64` points to `/usr/lib64`. I totally forgot about that, sorry for the confusion.
@imphil 
```
adrian@pcte247806:~/afs_work/praca/temp/cocotb/examples/mixed_language/tests$ rpm -qf /usr/bin/python3                                                        
python36-3.6.8-38.module_el8.5.0+895+a459eca8.x86_64
python39-3.9.6-2.module_el8.5.0+897+68c4c210.x86_64
adrian@pcte247806:tests$ rpm -q --filesbypkg python36-3.6.8-38.module_el8.5.0+895+a459eca8.x86_64
python36                  /usr/bin/easy_install-3
python36                  /usr/bin/pip-3
python36                  /usr/bin/pip3
python36                  /usr/bin/pydoc-3
python36                  /usr/bin/pydoc3
python36                  /usr/bin/python3
python36                  /usr/bin/python3.6
python36                  /usr/bin/python3.6m
python36                  /usr/bin/pyvenv-3
python36                  /usr/bin/unversioned-python
python36                  /usr/share/doc/python36
python36                  /usr/share/doc/python36/README
python36                  /usr/share/licenses/python36
python36                  /usr/share/licenses/python36/LICENSE
python36                  /usr/share/man/man1/python.1.gz
python36                  /usr/share/man/man1/python3.1.gz
adrian@pcte247806:tests$ rpm -q --filesbypkg python39-3.9.6-2.module_el8.5.0+897+68c4c210.x86_64
python39                  /usr/bin/pydoc-3
python39                  /usr/bin/pydoc3
python39                  /usr/bin/pydoc3.9
python39                  /usr/bin/python3
python39                  /usr/bin/python3.9
python39                  /usr/bin/unversioned-python
python39                  /usr/lib/.build-id
python39                  /usr/lib/.build-id/1e
python39                  /usr/lib/.build-id/1e/092647406fa830f0c9552b6bf771c982910e4a
python39                  /usr/share/doc/python39
python39                  /usr/share/doc/python39/README.rst
python39                  /usr/share/man/man1/python.1.gz
python39                  /usr/share/man/man1/python3.1.gz
python39                  /usr/share/man/man1/python3.9.1.gz
adrian@pcte247806:tests$ rpm -qf /usr/lib64/libpython3.so
python3-libs-3.6.8-41.el8.x86_64
```

marlonjames, 2022-09-28:
Is it possible that `pip3 install cocotb --user --verbose` is using the `pip3` from the 3.6 package?

If you have python 3.9 selected, does `python -m pip install cocotb --user --verbose` work?

imphil, 2022-09-28:
Looking at your original comment, "pip3 install cocotb --user --verbose" installed into '/home/adrian/.local/lib/python3.9/site-packages/cocotb", which suggests Python 3.9 was used. At the same time, your cocotb-config --libpython call returns `/usr/lib64/libpython3.so` , which belongs to Python 3.6. Now the interesting question becomes: why does find_libpython return the libpython3.so file, even though it belongs to Python 3.6, instead of Python 3.9?

Things to check for:
* Is cocotb-config indeed being executed by Python 3.9? (Not sure what the easiest way is to confirm that, I'd go with an strace, or add a debug print into config.py in your installation at /home/adrian/.local/lib/python3.9/site-packages/cocotb)
* If so: we should be returning a "normalized" version of the first candidate path. `python3 -c 'from cocotb._vendor import find_libpython; find_libpython.main()' --verbose` should give a bit more insight into what's going on.
* /usr/lib64/libpython3.so looks indeed rather small. Can you have a look at what symbols it exports using `ldd /usr/lib64/libpython3.so`?

abyszuk, 2022-09-29:
>     * Is cocotb-config indeed being executed by Python 3.9? (Not sure what the easiest way is to confirm that, I'd go with an strace, or add a debug print into config.py in your installation at /home/adrian/.local/lib/python3.9/site-packages/cocotb)

Looking at strace output it looks like it's using Python 3.9:
```
adrian@pcte247806:tests$ strace cocotb-config
(...)
stat("/usr/lib64/python3.9", {st_mode=S_IFDIR|0755, st_size=12288, ...}) = 0
openat(AT_FDCWD, "/usr/lib64/python3.9", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
(...)
```
No mention of Python 3.6 in this output. This would also agree with the output of 
`python3 -c 'from cocotb._vendor import find_libpython; find_libpython.main()' --candidate-paths`:
```
/usr/lib64/python3.9/config-3.9-x86_64-linux-gnu/libpython3.9.so
/usr/lib64/python3.9/config-3.9-x86_64-linux-gnu/libpython3.so
/usr/lib64/python3.9/config-3.9-x86_64-linux-gnu/libpython.so
/usr/lib64/libpython3.9.so
/usr/lib64/libpython3.so
/usr/lib64/libpython.so
/usr/lib/libpython3.9.so
/usr/lib/libpython3.so
/usr/lib/libpython.so
/usr/libpython3.9.so
/usr/libpython3.so
/usr/libpython.so
```
I'd say that for some reason `candidate_paths()` from `find_libpython` is adding e.g. `/usr/lib64/libpython3.so` to this list while it shouldn't.

>     * If so: we should be returning a "normalized" version of the first candidate path. `python3 -c 'from cocotb._vendor import find_libpython; find_libpython.main()' --verbose` should give a bit more insight into what's going on.

```
adrian@pcte247806:tests$ python3 -c 'from cocotb._vendor import find_libpython; find_libpython.main()' --verbose
DEBUG is_windows = False
DEBUG is_apple = False
DEBUG is_mingw = False
DEBUG is_msys = False
DEBUG Candidate: /usr/lib64/python3.9/config-3.9-x86_64-linux-gnu/libpython3.9.so
DEBUG Not found.
DEBUG Candidate: /usr/lib64/python3.9/config-3.9-x86_64-linux-gnu/libpython3.so
DEBUG Not found.
DEBUG Candidate: /usr/lib64/python3.9/config-3.9-x86_64-linux-gnu/libpython.so
DEBUG Not found.
DEBUG Candidate: /usr/lib64/libpython3.9.so
DEBUG Not found.
DEBUG Candidate: /usr/lib64/libpython3.so
DEBUG Found: /usr/lib64/libpython3.so
```
>     * /usr/lib64/libpython3.so looks indeed rather small. Can you have a look at what symbols it exports using `ldd /usr/lib64/libpython3.so`?

I believe you meant `objdump`, but I'm pasting the output of both:
```
adrian@pcte247806:tests$ ldd /usr/lib64/libpython3.so
        linux-vdso.so.1 (0x00007ffc0a7a6000)
        libcrypto.so.1.1 => /lib64/libcrypto.so.1.1 (0x00007f188ed6a000)
        libpython3.6m.so.1.0 => /lib64/libpython3.6m.so.1.0 (0x00007f188e827000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f188e607000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f188e242000)
        libz.so.1 => /lib64/libz.so.1 (0x00007f188e02b000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007f188de27000)
        libutil.so.1 => /lib64/libutil.so.1 (0x00007f188dc23000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f188d8a1000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f188f455000)
```
```
adrian@pcte247806:tests$ objdump -TC /usr/lib64/libpython3.so
/usr/lib64/libpython3.so:     file format elf64-x86-64

DYNAMIC SYMBOL TABLE:
0000000000000000  w   D  *UND*  0000000000000000              __gmon_start__
0000000000000000  w   D  *UND*  0000000000000000              _ITM_deregisterTMCloneTable
0000000000000000  w   D  *UND*  0000000000000000              _ITM_registerTMCloneTable
0000000000000000  w   DF *UND*  0000000000000000  GLIBC_2.2.5 __cxa_finalize
0000000000201008 g    D  .bss   0000000000000000  Base        _end
0000000000201000 g    D  .bss   0000000000000000  Base        _edata
0000000000201000 g    D  .bss   0000000000000000  Base        __bss_start
```

cmarqu, 2022-09-29:
`libpython3.so` is likely indeed a shim according to https://peps.python.org/pep-0384/#linkage.

abyszuk, 2022-09-29:
Out of curiosity - is there any *nix standard describing when the system must use symlinks? And why is the suffix always `.1.0` - can't remember ever seeing any other.

imphil, 2022-09-29:
The suffix is the soname (https://en.wikipedia.org/wiki/Soname), indicating the ABI level. glibc and libstdc++ are good examples which do ABI versioning "right":

```
$ ls -l /usr/lib64/libstdc++.so.6*
lrwxrwxrwx. 1 root root      19 28. Apr 16:08 /usr/lib64/libstdc++.so.6 -> libstdc++.so.6.0.25
-rwxr-xr-x. 1 root root 1661416 28. Apr 16:11 /usr/lib64/libstdc++.so.6.0.25
`

imphil, 2022-09-29:
Further comments while debugging: find_libpython as vendored into cocotb is slightly older than the most recent upstream version (https://github.com/ktbarrett/find_libpython). With that version I get output which does not include the shim library, but also does not find any other libpython -- i.e. the problem is different, but not solved :-)

imphil, 2022-09-29:
OK, so it looks like the heuristics in find_libpython need updating. I opened https://github.com/ktbarrett/find_libpython/issues/35 for that. Thankfully, additional checks that a more recent find_libpython does prevent the shim library "libpython3.so" from being returned as candidate, so we at least get a clear error instead of picking up the wrong Python version.

imphil, 2022-10-05:
@abyszuk We should have a fix for the libpython discovery now (https://github.com/ktbarrett/find_libpython/issues/35). Can you try:

```
# Make sure to use the Python version you're having trouble with!
python38 -m pip install git+https://github.com/ktbarrett/find_libpython.git@master#egg=find_libpython
python38 -m find_libpython
```

The command should give you the right libpython.so file. Can you post the output here?

If that works we'll integrate those changes into cocotb.

abyszuk, 2022-10-05:
It works!
```
adrian@pcte247806:tb$ python3.9 -m find_libpython
/usr/lib64/libpython3.9.so.1.0

adrian@pcte247806:tb$ ls -l /usr/lib64/libpython3.9.so.1.0 
-rwxr-xr-x 1 root root 3918968 Aug 25  2021 /usr/lib64/libpython3.9.so.1.0
```

imphil, 2022-11-15:
cocotb 1.7.2 is now released and should resolve the issue you were seeing. Please reopen this isuse if the problem persists.

closed 2022-11-15 by imphil