cocotb/cocotb #3101: handle changes between ModifiableObject and HierarchyArrayObject depending on if accessed first

bengiac, 2022-10-06:
cocotb=1.7.1
64Bit 
Questa Sim-64 2022.3 64bit
Python 3.10, system
Windows 10

I'm finding the handle I'm expecting to get from getattr changes depending if I access the signal directly first
A use case of this is passing a dictionary of signals to cocotb-bus class

minimal example
where dut is a vhdl file with a record, containing std_logic signals

```
class TB:
  def __init__(self, dut):
    self.dut = dut
    self.log = logging.getLogger("cocotb.tb")
    self.log.setLevel(logging.DEBUG)
    self.dut._log.setLevel(logging.DEBUG)

    h = self.dut.axi_sw_to_reg.awid

    handle1 = getattr(self.dut, "axi_sw_to_reg.awid")
    print("{}".format(type(handle1)))
```
output 
```
<class 'cocotb.handle.ModifiableObject'>
```

```
class TB:
  def __init__(self, dut):
    self.dut = dut
    self.log = logging.getLogger("cocotb.tb")
    self.log.setLevel(logging.DEBUG)
    self.dut._log.setLevel(logging.DEBUG)

    #h = self.dut.axi_sw_to_reg.awid

    handle1 = getattr(self.dut, "axi_sw_to_reg.awid")
    print("{}".format(type(handle1)))
```
output 
```
<class 'cocotb.handle.HierarchyArrayObject'>
```

I expected to get a <class 'cocotb.handle.ModifiableObject'> in both cases

marlonjames, 2022-10-06:
This appears to be a misconception of `getattr()`, combined with behavior of the GPI in handle lookup by name and the singleton handle store.
`getattr()` doesn't support "a.b.c" type attribute names, you have to do each one individually.

So you are looking for an attribute named "axi_sw_to_reg.awid". This will not be found on `dut`, so it will attempt to look up "axi_sw_to_reg.awid" below the `dut` handle.
This is succeeding, but the handle type is something that converts to `HierarchyArrayObject`.

The reason the first example succeeds, is because the singleton handle store is checked against the path. You have already correctly accessed the handle in ` h = self.dut.axi_sw_to_reg.awid`. The lookup sequence is: it will find the same handle as in the second example, but it will be discarded because the path matches an existing handle, so you get back a `ModifiableObject`.

You can check this by examining the `dut._sub_handles` dictionary before and after your `getattr()` call.

bengiac, 2022-10-06:
Thanks for the explanation

>This will not be found on dut, so it will attempt to look up "axi_sw_to_reg.awid" below the dut handle. This is succeeding

Are you able to elaborate more on why it succeeds at all? 
As I'm new to this and would like to understand

closed 2022-10-06 by bengiac

marlonjames, 2022-10-06:
> Are you able to elaborate more on why it succeeds at all?

Sure.
First, when the initial attribute lookup fails, it calls `get_handle_by_name("axi_sw_to_reg.awid")` on the handle for `dut`:
https://github.com/cocotb/cocotb/blob/1ff2bb26317d0a582a7dd6e210373bd9570c9795/cocotb/handle.py#L315
That goes through the simulator Python extension module and into GPI here:
https://github.com/cocotb/cocotb/blob/1ff2bb26317d0a582a7dd6e210373bd9570c9795/cocotb/share/lib/gpi/GpiCommon.cpp#L337
That makes its way into the FLI, VHPI, or VPI functions for handle lookup by name. Assuming you are using FLI, it ends up here:
https://github.com/cocotb/cocotb/blob/1ff2bb26317d0a582a7dd6e210373bd9570c9795/cocotb/share/lib/fli/FliImpl.cpp#L273
and this function concatenates the path of the parent with the passed in name and does a lookup with the full path, which apparently succeeds in this case.