cocotb/cocotb #3108: Failed to find a signal from an interface in Questa 2022.3

felixdube, 2022-10-13:
Cocotb 1.7.0
Simulator Questa 2022.3
Ubuntu 20.04 

Hi all,
In the design, I have an interface part of the ports of a module. There is a failure trying to access one of the signals in that interface.

`
#      0.00ns DEBUG    gpi                                Searching for out_if
#      0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiRefObj(608)
#      0.00ns DEBUG    gpi                                Checking tb.dut.a_inst.b_inst.c_inst.out_if exists
#      0.00ns DEBUG    cocotb.out_if                      Created
#      0.00ns DEBUG    cocotb.out_if                      Signal name data, idx None
#      0.00ns DEBUG    cocotb.out_if                      Discovering all on out_if
#      0.00ns WARNING  gpi                                VPI: Implementation does not know how to iterate over vpiRefObj(608)
#      0.00ns DEBUG    gpi                                Reached end of iterator
#      0.00ns DEBUG    cocotb.out_if                      Signal name sop, idx None
#      0.00ns DEBUG    cocotb.out_if                      Signal name eop, idx None
#      0.00ns DEBUG    cocotb.out_if                      Signal name error, idx None
#      0.00ns DEBUG    cocotb.out_if                      Signal name ready
#      0.00ns DEBUG    gpi                                Searching for ready
#      0.00ns DEBUG    gpi                                VPI: ready initialized with 1 elements
#      0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiReg(48)
#      0.00ns DEBUG    gpi                                Checking tb.dut.a_inst.b_int.c_inst.out_if.ready exists
#      0.00ns DEBUG    cocotb.ready                       Created
#      0.00ns DEBUG    cocotb.out_if                      Signal name ready, idx None
`

Why is it trying to iterate over the interface? Notice that the signal ready from the same interfaces successfully found. I have tried with Questa 2022.1 and there is no issue. Here is what the output looks like:

`
#      0.00ns DEBUG    gpi                                Searching for out_if
#      0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiRefObj(608)
#      0.00ns DEBUG    gpi                                Checking tb.dut.a_inst.b_inst.c_inst.out_if exists
#      0.00ns DEBUG    cocotb.out_if                      Created
#      0.00ns DEBUG    cocotb.out_if                      Signal name data, idx None
#      0.00ns DEBUG    gpi                                Searching for data
#      0.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
#      0.00ns DEBUG    gpi                                VPI: data initialized with 8 elements
#      0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiReg(48)
#      0.00ns DEBUG    gpi                                Checking tb.dut.a_inst.b_inst.c_inst.out_if.data exists
#      0.00ns DEBUG    cocotb.data                        Created
#      0.00ns DEBUG    cocotb.out_if                      Signal name sop, idx None
#      0.00ns DEBUG    gpi                                Searching for sop
#      0.00ns DEBUG    gpi                                VPI: sop initialized with 1 elements
#      0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiReg(48)
#      0.00ns DEBUG    gpi                                Checking tb.dut.a_inst.b_inst.c_inst.out_if.sop exists
#      0.00ns DEBUG    cocotb.sop                         Created
#      0.00ns DEBUG    cocotb.out_if                      Signal name eop, idx None
#      0.00ns DEBUG    gpi                                Searching for eop
#      0.00ns DEBUG    gpi                                VPI: eop initialized with 1 elements
#      0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiReg(48)
#      0.00ns DEBUG    gpi                                Checking tb.dut.a_inst.b_inst.c_inst.out_if.eop exists
#      0.00ns DEBUG    cocotb.eop                         Created
#      0.00ns DEBUG    cocotb.out_if                      Signal name error, idx None
#      0.00ns DEBUG    gpi                                Searching for error
#      0.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [0:0] and length >1<
#      0.00ns DEBUG    gpi                                VPI: error initialized with 1 elements
#      0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiReg(48)
#      0.00ns DEBUG    gpi                                Checking tb.dut.a_inst.b_inst.c_inst.out_if.error exists
#      0.00ns DEBUG    cocotb.error                       Created
#      0.00ns DEBUG    gpi                                Searching for ready
#      0.00ns DEBUG    gpi                                VPI: ready initialized with 1 elements
#      0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiReg(48)
#      0.00ns DEBUG    gpi                                Checking tb.dut.a_inst.b_inst.c_inst.out_if.ready exists
#      0.00ns DEBUG    cocotb.ready                       Created
#      0.00ns DEBUG    cocotb.out_if                      Signal name ready, idx None
`

Another interesting note is that, if the interface is not instantiated as part of the ports a module, I do not see this issue.
Let me know if there is anything I should try.

Thank you!

imphil, 2022-10-13:
Can you post a minimal example with your HDL code and a cocotb test that allows us to reproduce the problem?

felixdube, 2022-10-14:
Seems like it might have something to do with the busMonitor... I've enhance the test_sv_interface: https://github.com/felixdube/cocotb/tree/fdube-enhance-test_sv_interface/tests/test_cases/test_sv_interface 
Notice that:
1. The test fails with Questa 2022.3 and passes with Questa 2022.1
2. When accessing the interface instantiated inside the module there is no problem (ie. my_bfm1)