cocotb/cocotb #3109: ModuleNotFoundError: No module named 'encodings'

PiyushSaini07, 2022-10-14:
Hi All,

When I invoke `make`, I'm facing the following issue

```
#Make sure all libs in SOURCES_VHDL_* are mentioned in VHDL_LIB_ORDER and vice versa
set -o pipefail; \
MODULE=test_counter TESTCASE= TOPLEVEL=tb_counter GPI_EXTRA= TOPLEVEL_LANG=verilog \
 /data/CadenceInstall/xcelium2203/xcelium2203_s004/tools/bin/xrun -timescale 1ns/1ps \
"-f counter_xrun.f"    -licqueue -64 -xmlibdirpath ./ -plinowarn  -top tb_counter -loadvpi /usr/local/lib64/python3.6/site-packages/cocotb/libs/libcocotbvpi_ius.so:vlog_startup_routines_bootstrap  -access +rwc -createdebugdb     2>&1 | tee .//sim.log
TOOL:	xrun(64)	22.03-s004: Started on Oct 14, 2022 at 01:06:21 CDT
xrun(64): 22.03-s004: (c) Copyright 1995-2022 Cadence Design Systems, Inc.
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007fe8261b3f00 (most recent call first):
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007f224b7ce740 (most recent call first):
xmelab: *F,INTERR: INTERNAL EXCEPTION
```

I checked for `encodings` package in python installation and it was available. I'm not sure what could be the issue here.

Some details that might be useful:

    
    Operating System: Red Hat Enterprise Linux
    CPE OS Name: cpe:/o:redhat:enterprise_linux:7.9:GA:server
    Kernel: Linux 3.10.0-1160.71.1.el7.x86_64
    
    Python version : 3.6.8
   

I'm facing this issue on cocotb v1.7.0 and v1.7.1

I ran the exact same code on CentOS 7 and it worked fine there (I used iverilog here).
Cocotb works perfectly there (ran with python v3.6.8 and tried on cocotb v1.7.0 and v1.7.1).

       Operating System: CentOS Linux 7 (Core)
       CPE OS Name: cpe:/o:centos:centos:7
       Kernel: Linux 3.10.0-1160.76.1.el7.x86_64

I compared the paths where cocotb was installed on both machines and everything seems fine.
Some outputs:
```
which python3              -->/bin/python3
cocotb-config --libpython  -->/usr/lib64/libpython3.so
cocotb-config --python-bin  --> /bin/python3

```

I referred to issue #2474 as well. But could not get this working.
Please help me in debugging this problem.

marlonjames, 2022-10-14:
I would try something like `LD_DEBUG=all make > log.txt 2>&1` and look through the log to see if the correct libpython is being loaded.

themperek, 2022-10-18:
This is not a solution but using different python version that 3.6 may help (is this version shipped with Xcelium?).

Can try use [conda](https://docs.conda.io/) or [pyenv](https://github.com/pyenv/pyenv).

PiyushSaini07, 2022-10-18:
@marlonjames I'll need to get the "log.txt" out from work server. Will attach the output then.
@themperek The one we have, python version shipped with Xcelium is v3.6. System-wide we are using v3.6.8(not sure how different these two versions are). 
I'll try with conda or pyenv and see if it works.

PiyushSaini07, 2022-10-19:
I tried to set a python virtual environment and invoked via the python virtual environment. 
This did not resolve the issue.

I'm under the impression that conda does the same thing when installing any package. 
Will still try to install cocotb via conda.

themperek, 2022-10-19:
>Will still try to install cocotb via conda.
```cmd
curl -L https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh -o miniconda.sh
bash miniconda.sh -b -p $HOME/miniconda
export PATH=$HOME/miniconda/bin/:$PATH
conda update --yes conda
pip install cocotb
```

PiyushSaini07, 2022-10-20:
> I would try something like LD_DEBUG=all make > log.txt 2>&1 and look through the log to see if the correct libpython is being loaded.

I'm adding the `log.txt` here. I'm not sure how to debug this libpython problem. Can you please help me in checking this?
[log.txt.tar.gz](https://github.com/cocotb/cocotb/files/9826479/log.txt.tar.gz)

themperek, 2022-10-20:
@PiyushSaini07 
```
find library=libpython3.6m.so.1.0 [0]; searching
     10805:	 search path=/data/CadenceInstall/xcelium2203/xcelium2203_s004/tools.lnx86/inca/bin/64bit/../../../lib/64bit		(RPATH from file xmelab)
     10805:	  trying file=/data/CadenceInstall/xcelium2203/xcelium2203_s004/tools.lnx86/inca/bin/64bit/../../../lib/64bit/libpython3.6m.so.1.0
```

Xcelium 22 ships with python 3.6 itself and manipulates libraries it is loading making issues (One solution I think worked in the past was just to use different python version (3.7 or any other). I do not have access to Xcelium anymore so hard to debug/find solution.

Maybe installing cocotb in python shipped with Xcelium would do.

PiyushSaini07, 2022-11-24:
Our IT expert tried out a few things and was able to resolve the issue.
I'm not sure what he did but to take a wild guess, he changed some libpython path.

Thanks for the help!
I'm closing the issue.

closed 2022-11-24 by PiyushSaini07