cocotb/cocotb #3110: Signals not updated after rising edge

gruvw, 2022-10-14:
It seems that cocotb does not update my signal after awaiting a rising edge.

Simple example:
```python
# test.py
import cocotb
from cocotb.clock import Clock
from cocotb.runner import get_runner
from cocotb.triggers import RisingEdge, FallingEdge

@cocotb.test()
async def test(dut):

    cocotb.start_soon(Clock(dut.clk, 4).start(start_high=False))

    dut.e.value = 1
    await RisingEdge(dut.clk)
    assert dut.q.value == 1

def test_runner():
    runner = get_runner("ghdl")()
    runner.build(
        vhdl_sources=["test.vhd"],
        toplevel="test",
        build_dir="sim_build"
    )
    runner.test(
        toplevel="test",
        py_module="test",
        extra_args=["--vcd=waves.vcd"]
    )

if __name__ == "__main__":
    test_runner()
 ```
 ```vhdl
 -- test.vhd
 LIBRARY ieee;
USE ieee.std_logic_1164.ALL;

ENTITY test IS
    PORT (
        clk : IN STD_LOGIC;
        e   : IN STD_LOGIC;
        q   : OUT STD_LOGIC
    );
END test;

ARCHITECTURE arch OF test IS
BEGIN
    dff : PROCESS (clk)
    BEGIN
        IF rising_edge(clk) AND e = '1' THEN
            q <= e;
        END IF;
    END PROCESS dff;
END arch;
```

When I run this python file, I get the following error: `ValueError: Unresolvable bit in binary string: 'U'`  
It seems that even after the rising edge, the value of `q` is still `U` even though when I look at the wave output, I see that it is `1`.  
If I wait just a `fs` after the rising edge, I see that `q` has the right value.

### Infos

Cocotb version: 1.7.1  
OS: Ubuntu 22.04.1 LTS x86_64  
GHDL version: GHDL 2.0.0 (tarball) [Dunoon edition] Compiled with GNAT Version: 10.3.0
Python version: 3.10.6

felixdube, 2022-10-14:
I don't think this is an issue. The signals q is updated the next delta cycle after the rising edge of the clock. So it make sense that if you sample q on the same delta cycles as the rising edge of the clock, q is still 'U'.

You might want to read cocotb's timing model: https://github.com/cocotb/cocotb/wiki/Timing-Model

closed 2022-10-14 by ktbarrett

gruvw, 2022-10-15:
Ok thank you @felixdube for pointing me toward that part of the documentation, I did not read it before.  
So what I should be doing to check (if I understood correctly) is the following:
```python
await RisingEdge(dut.clk)
await ReadOnly()
assert dut.q.value == 1
```
Is this correct ?

felixdube, 2022-10-15:
Yes this would technically work, but honestly I try to avoid using the ReadOnly(). I am not sure what is the consensus on that in the community, but this construct is rarely useful to me.

First, you cannot drive a signal after ReadOnly() without advancing time. Doing the following you give you an error:
```python
await RisingEdge(dut.clk)
await ReadOnly()
assert dut.q.value == 1
dut.e.value = 0
# do something useful
```

Second, I like to have my code behave similar to how HDL works. I would probably write something like:
```python
while True:
    await RisingEdge(dut.clk)
    if(dut.q.value == 1):
        dut.e.value = 0
        # do something useful
```

The “something useful" would happen one clock cycle than if you used ReadOnly(), but from a hardware perspective that makes much more sense. As @ktbarret said, think about clocked process.

ktbarrett, 2022-10-16:
I avoid `ReadOnly` and think most code should avoid it (in fact I haven't come across a senario in "typical" code where I would use it). Most internal Drivers and Monitors use something like VHDL processes, so basically the second example @felixdube  posted.

gruvw, 2022-10-17:
What about `ReadWrite()` then ?

ktbarrett, 2022-10-17:
I would also avoid it.