cocotb/cocotb #3111: Simulation time increase when using newer version

quintiminator, 2022-10-14:
Hello, I am running with Python 3.7 on the 32-bit simulator version.

I was running a rather short simulation with cocotb version 1.5.1 and it fairly quick. However, when comparing the simulation times with 1.6.2 and 1.7.1 there is a significant slowdown.

The 1.5.1 version would be around 200s whereas the 1.6.2 and 1.7.1 result in a simulation time which is around 600 and 900 seconds, respectively.

In another simulation which takes longer, using cocotb 1.5.1. resulted in a simulation time less than 1000 seconds whereas the newer versions could not even finish within hours.

Besides the cocotb version, there were no other variables in this test as they were all run in a docker container. I assume this slowdown has probably something to do with the scheduler but I cannot find significant changes between 1.5.X and 1.6.X and 1.7.X when comparing sources codes.

What else could have such a great influence on the time? Is it connected to how the new coroutines are planned? the fork vs start_soon  for example? However, I do not see how this would be affect the simulation this greatly. The Python version remained the same.

felixdube, 2022-10-14:
You might want to run the profiler on both version and compare.
https://docs.cocotb.org/en/stable/building.html?highlight=profiling#envvar-COCOTB_ENABLE_PROFILING

quintiminator, 2022-10-17:
Hello, thank you for the reply!

I actually did the profiling but the graphs did not make much sense to me.

It seemed all relative and not absolute(?). The scheduler e..g. was in all three versions (tested 1.5, 1.6 and 1.7) 97% and the 'subtrees' were all similar. It did not show the exact time. And also the decorator and monitor class were in all three cases similar percentage. Where can I find how to read such graph? I used graph2dot.

ktbarrett, 2022-10-18:
Can you post the results here? They do require a lot of interpretation.

themperek, 2022-10-18:
BTW: One could use this to get line profiler (maybe easier to read) : https://github.com/vpelletier/pprofile

quintiminator, 2022-10-18:
@themperek  I will take a look at it, from the doc alone it seems intuitive!

@ktbarrett for example a snippet of 1.5:
[image]

Snippet of 1.7:
[image]

themperek, 2022-10-18:
Maybe here: https://github.com/cocotb/cocotb/blob/77e22580bb189bdaf4367d7fa6aecba6c2505a77/cocotb/scheduler.py#L331

```python3
    import pprofile
    prof = pprofile.Profile()
    with prof():

      # code

    prof.print_stats()
```

instead od `ctx`. Or similar,

quintiminator, 2022-10-24:
Hello, so I tried some profiling but it was really difficult with the asynchronous set-up and cocotb framework around it.

I tried finding the problem in a different manner and I might have found what causes the simulation time to be drastically greater. I updated my initial comment.

But basically, I think it could have something to do with the communication between the Python framework and the simulator such as Modelsim when checking for rising edges (or wait for X cycles). I checked this by removing all features and only kept a counter which increases every clock cycle. So I used a RisingEdge function (and even await Timer) to test how fast it is, and it is much slower in 1.7 than in 1.5.

ktbarrett, 2023-02-06:
@quintiminator What version of Modelsim are you using?

quintiminator, 2023-04-12:
@ktbarrett I did not expect response anymore because I posted it a long time ago. Anyway, I am using Modelsim version 2019.3. I will read the posts of #3229 first because that person has the same issue.

ktbarrett, 2023-04-12:
@quintiminator If the issue has been resolved with the development version of cocotb, consider closing this issue.

quintiminator, 2023-04-18:
@ktbarrett  I am currently checking, I will close it when it is solved!

quintiminator, 2023-04-18:
It is indeed fixed in the 1.8.0 development version!

closed 2023-04-18 by quintiminator