cocotb/cocotb #3114: (Bug?) Cannot access custom-defined array signals with GHDL

bellaz89, 2022-10-20:
Failing example zip:
https://desycloud.desy.de/index.php/s/swyorDfHqXB4es6

run `make` to execute the example

The important files are
* tb_cocotb_utils_iq.py cocotb test bench
* iq.py methods to read and write the iq type
* ent_math_cocotb_utils_iq.vhd VHDL dut
* pkg_math_iq.vhd t_iq definition (just see line 64)

When trying to access ports with a custom defined array (t_iq, see definition at pkg_math_iq.vhd), cocotb gives the following error:
```
  Failed to find a handle at index 0 via any registered implementation
```
e.g
```
  dut.po_const_iq1[0] # fails?
```

The `len()` method gives the correct values for the array (24), however, any attempt to access the value of the custom defined type, either by reading or writing it, results in a failure

* cocotb version: 1.7.1
* operating system: Ubuntu 20.04.5 LTS (64 bit)
* simulator: GHDL 3.0.0-dev (2.0.0.r859.g5cbfa82eb) [Dunoon edition]
                   Compiled with GNAT Version: 9.4.0
                   mcode code generator
python version: 3.8 (Anaconda)

```
rm -f results.xml
make -f Makefile results.xml
make[1]: Entering directory '/home/bellandi/Projects/failing_cocotb_array'
mkdir -p sim_build
/usr/local/bin/ghdl -i  --std=93  --workdir=sim_build --work=desy pkg_common_logic_utils.vhd pkg_math_utils.vhd pkg_math_signed.vhd pkg_math_iq.vhd &&  \
/usr/local/bin/ghdl -i  --std=93  --workdir=sim_build --work=work ent_math_cocotb_utils_iq.vhd && \
/usr/local/bin/ghdl -m  --std=93  --workdir=sim_build -Psim_build --work=work math_cocotb_utils_iq
../../src/ieee/math_real-body.vhdl:830:14:warning: declaration of "real_vector" hides type "real_vector" [-Whide]
        type REAL_VECTOR is array (NATURAL range <>) of REAL;
             ^
rm -f results.xml
MODULE=tb_cocotb_utils_iq TESTCASE= TOPLEVEL=math_cocotb_utils_iq TOPLEVEL_LANG=vhdl \
 /usr/local/bin/ghdl -r  --std=93 --workdir=sim_build -Psim_build --work=work math_cocotb_utils_iq --vpi=/home/bellandi/.local/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_ghdl.so --wave=dump.ghw --vcd=dump.vcd 
loading VPI module '/home/bellandi/.local/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_ghdl.so'
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
VPI module loaded!
     0.00ns INFO     cocotb                             Running on GHDL version 3.0.0-dev (2.0.0.r859.g5cbfa82eb) [Dunoon edition]
     0.00ns INFO     cocotb                             Running tests with cocotb v1.7.1 from /home/bellandi/.local/lib/python3.8/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1666259583
     0.00ns INFO     cocotb.regression                  Found test tb_cocotb_utils_iq.test_constant
     0.00ns INFO     cocotb.regression                  Found test tb_cocotb_utils_iq.test_read
     0.00ns INFO     cocotb.regression                  Found test tb_cocotb_utils_iq.test_write
     0.00ns INFO     cocotb.regression                  running test_constant (1/3)
                                                          Test constant output values
     2.00ns WARNING  gpi                                Failed to find a handle at index 0 via any registered implementation
     2.00ns INFO     cocotb.regression                  test_constant failed
                                                        Traceback (most recent call last):
                                                          File "/home/bellandi/Projects/failing_cocotb_array/tb_cocotb_utils_iq.py", line 16, in test_constant
                                                            assert iq.read(dut.po_const_iq1) == (0, 0)
                                                          File "/home/bellandi/Projects/failing_cocotb_array/iq.py", line 48, in read
                                                            i_binary[idx] = handle[idx][0]
                                                          File "/home/bellandi/.local/lib/python3.8/site-packages/cocotb/handle.py", line 652, in __getitem__
                                                            raise IndexError(
                                                        IndexError: po_const_iq1(GPI_ARRAY) contains no object at index 0
     2.00ns INFO     cocotb.regression                  running test_read (2/3)
                                                          Test iq.read
     6.00ns WARNING  gpi                                Failed to find a handle at index 0 via any registered implementation
     6.00ns INFO     cocotb.regression                  test_read failed
                                                        Traceback (most recent call last):
                                                          File "/home/bellandi/Projects/failing_cocotb_array/tb_cocotb_utils_iq.py", line 35, in test_read
                                                            assert iq.read(dut.po_iq) == (i - 3, q + 4)
                                                          File "/home/bellandi/Projects/failing_cocotb_array/iq.py", line 48, in read
                                                            i_binary[idx] = handle[idx][0]
                                                          File "/home/bellandi/.local/lib/python3.8/site-packages/cocotb/handle.py", line 652, in __getitem__
                                                            raise IndexError(
                                                        IndexError: po_iq(GPI_ARRAY) contains no object at index 0
     6.00ns INFO     cocotb.regression                  running test_write (3/3)
                                                          Test iq.write
     8.00ns WARNING  gpi                                Failed to find a handle at index 0 via any registered implementation
     8.00ns INFO     cocotb.regression                  test_write failed
                                                        Traceback (most recent call last):
                                                          File "/home/bellandi/Projects/failing_cocotb_array/tb_cocotb_utils_iq.py", line 45, in test_write
                                                            assert iq.write(dut.pi_iq, (i, q))
                                                          File "/home/bellandi/Projects/failing_cocotb_array/iq.py", line 25, in write
                                                            handle[idx][0] = i_binary[idx]
                                                          File "/home/bellandi/.local/lib/python3.8/site-packages/cocotb/handle.py", line 652, in __getitem__
                                                            raise IndexError(
                                                        IndexError: pi_iq(GPI_ARRAY) contains no object at index 0
     8.00ns INFO     cocotb.regression                  ******************************************************************************************
                                                        ** TEST                              STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        ******************************************************************************************
                                                        ** tb_cocotb_utils_iq.test_constant   FAIL           2.00           0.00       2644.58  **
                                                        ** tb_cocotb_utils_iq.test_read       FAIL           4.00           0.00       9248.74  **
                                                        ** tb_cocotb_utils_iq.test_write      FAIL           2.00           0.00       7079.00  **
                                                        ******************************************************************************************
                                                        ** TESTS=3 PASS=0 FAIL=3 SKIP=0                      8.00           0.28         28.73  **
                                                        ******************************************************************************************
                                                        
make[1]: Leaving directory '/home/bellandi/Projects/failing_cocotb_array'

```

bellaz89, 2022-10-21:
Most probably it is. However in my case only the outer index is unconstrained.

bellaz89, 2022-10-21:
I also tried to change the definition of `t_iq`

https://desycloud.desy.de/index.php/s/Fp9W5E5rk6qNXR4

using record as the inner type
```
type t_iq_bit is record                                                       
  i : std_logic;                                                              
  q : std_logic;                                                              
end record t_iq_bit; 

type t_iq is array (integer range <>) of t_iq_bit;
```

However I get the same error.

I am a bit confused. How standard ieee array types (std_logic_vector, signed, unsigned) work while it crashes for an array of -fixed-size- records? 

```
     2.00ns WARNING  gpi                                Failed to find a handle at index 0 via any registered implementation
     2.00ns INFO     cocotb.regression                  test_constant failed
                                                        Traceback (most recent call last):
                                                          File "/home/bellandi/Projects/failing_cocotb_array/tb_cocotb_utils_iq.py", line 16, in test_constant
                                                            assert iq.read(dut.po_const_iq1) == (0, 0)
                                                          File "/home/bellandi/Projects/failing_cocotb_array/iq.py", line 48, in read
                                                            i_binary[idx] = handle[idx].i
                                                          File "/home/bellandi/.local/lib/python3.8/site-packages/cocotb/handle.py", line 652, in __getitem__
                                                            raise IndexError(
                                                        IndexError: po_const_iq1(GPI_ARRAY) contains no object at index 0
```

ktbarrett, 2022-10-21:
`std_logic_vector` and all subtypes map to `vpiNet` in GHDL, so they aren't treated as array values.

imphil, 2022-11-10:
Since this is an upstream issue, would you mind filing an issue at the GHDL repository, @bellaz89? I don't think there's anything else we can do here on the cocotb side.

closed 2022-11-10 by imphil