cocotb/cocotb #3116: Seg fault with verilator 4.106

ollie-etl, 2022-10-27:
I experience seg faults when running both my testbenches, and also with the examples in this repo with verilator 4.106

These occur somewhere within `vlog_startup_routines_bootstrap()`, in `verilator.ccp`. I haven't yet more accurately pinpointed the problem.

Icarus works just fine for reference.

Has anyone else experienced this? I recall this working on older versions of cocotb.

```console
[nix-shell:~/repos/digitalrf]$ cd ../cocotb/examples/

[nix-shell:~/repos/cocotb/examples]$ SIM=verilator make
make[1]: Entering directory '~/repos/cocotb/examples/adder/tests'
rm -f results.xml
make -f Makefile results.xml
make[2]: Entering directory '~/repos/cocotb/examples/adder/tests'
rm -f results.xml
MODULE=test_adder TESTCASE= TOPLEVEL=adder TOPLEVEL_LANG=verilog \
         sim_build/Vtop 
make[2]: *** [/nix/store/wyp9gf8dlls1c3zv1l18n7h0af5lv5hy-python3.10-cocotb-1.7.1/lib/python3.10/site-packages/cocotb/share/makefiles/simulators/Makefile.verilator:61: results.xml] Segmentation fault (core dumped)
make[2]: Leaving directory '~/repos/cocotb/examples/adder/tests'
make[1]: *** [/nix/store/4pc8p2x8455ff76z1ljn70dpfw92saim-python3-3.10.4-env/lib/python3.10/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
make[1]: Leaving directory '~/repos/cocotb/examples/adder/tests'
make: *** [Makefile:45: adder/tests] Error 2
```

System Info:
```console
[nix-shell:~/repos/cocotb/examples]$ git rev-parse HEAD
fd351d0924dc2cd82cd6616ce66788d9dbad6c47

[nix-shell:~/repos/cocotb/examples]$ verilator --version
Verilator 4.106 2020-12-02 rev UNKNOWN_REV

[nix-shell:~/repos/my_repo]$ niv show verilator
verilator
  branch: master
  description: Verilator open-source SystemVerilog simulator and lint system
  homepage: https://verilator.org
  owner: verilator
  repo: verilator
  rev: v4.106
  sha256: 0yg13kzjwpp5if57d54lsaqvw8dkj0dnqm3xnx78kmnpyvjk702y
  type: tarball
  url: https://github.com/verilator/verilator/archive/v4.106.tar.gz
  url_template: https://github.com/<owner>/<repo>/archive/<rev>.tar.gz
  

[nix-shell:~/repos/my_repo]$ python3 --version
Python 3.10.4

[nix-shell:~/repos/my_repo]$ python3
Python 3.10.4 (main, Mar 23 2022, 20:25:24) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pkg_resources
>>> installed_packages = pkg_resources.working_set
>>> installed_packages_list = sorted(["%s==%s" % (i.key, i.version)
...     for i in installed_packages])
>>> print(*installed_packages_list, sep='\n')
apipkg==2.1.1
asttokens==2.0.5
attrs==21.4.0
backcall==0.2.0
certifi==2022.5.18.1
cffi==1.15.0
cocotb-bus==0.0.0
cocotb-test==0.2.2
cocotb==1.7.1
cocotbext-axi==0.1.18
cryptography==37.0.2
cycler==0.11.0
decorator==5.1.1
defusedxml==0.7.1
docutils==0.18.1
ecdsa==0.17.0
execnet==1.9.0
executing==0.8.2
fonttools==4.33.3
iniconfig==1.1.1
ipython-genutils==0.2.0
ipython==8.2.0
jedi==0.18.1
kiwisolver==1.4.2
matplotlib-inline==0.1.3
matplotlib==3.5.2
mock==4.0.3
numpy==1.21.6
olefile==0.46
packaging==21.3
parso==0.8.3
path==16.4.0
pexpect==4.8.0
pickleshare==0.7.5
pillow==9.1.1
pluggy==1.0.0
prompt-toolkit==3.0.28
psutil==5.9.1
ptyprocess==0.7.0
pure-eval==0.2.2
py==1.11.0
pycparser==2.21
pycrypto==3.14.1
pycryptodome==3.14.1
pygments==2.12.0
pyparsing==3.0.9
pytest-forked==1.4.0
pytest-xdist==2.5.0
pytest==7.1.1
python-dateutil==2.8.2
pytz==2022.1
scapy==2.4.5
setuptools==44.0.0.post0
six==1.16.0
stack-data==0.2.0
toml==0.10.2
tomli==2.0.1
tornado==6.1
traitlets==5.1.1
wcwidth==0.2.5
```

ollie-etl, 2022-10-27:
I can also recreate this on ubuntu 22.04 with a venv

```console
$ sudo apt-get install make gcc g++ python3 python3-dev python3-pip python3-venv
$ python3 -m venv venv
$ source venv/bin/activate
$ pip3 install -e .
$ cd examples
$ SIM=verilator make
```

marlonjames, 2022-10-28:
Have you tried the `debug` make target to get more info or a stack trace?

ollie-etl, 2022-10-31:
no, but i shall!

marlonjames, 2022-10-31:
There is one in the Verilator makefile

closed 2023-01-02 by github-actions[bot]