cocotb/cocotb #3121: Support graceful test termination when simulator terminates independently

mkorbel1, 2022-11-04:
If the SV simulator dies (e.g. through `$finish`) when cocotb is waiting on something (e.g. `await Timer`) it will just hang and wait forever.  There appears to be no way to gracefully detect that the simulator has completed and bring the test to completion.

Reproduced this with python v3.6.9, cocotb v1.6.0, and iverilog v11.0 on WSL2 Ubuntu on Windows 10.  Also reproduced in slightly different setup.

A common case which might cause this is if there is a UVM test being run in the SV simulator as well, which calls `$finish` when it has completed (e.g. dropped all objections, finished all phases).

It would be nice if there was at 

My current workaround for this issue is to have a real-time timeout (e.g. if the simulator hasn't "replied" in 5 seconds, then assume it is dead from a `$finish` or something else), but this solution is less than ideal.  Ideally, there would be at least some flag that gets set by cocotb that the simulator has completed and, if possible, the reason why it has completed.

Additionally, if the simulator kills the test for a "good" reason, cocotb will still detect this as a test "failure".  It would be nice if there was a way to mark this as a passing test, since it might be expected that some SV code is responsible for ending the test.

mkorbel1, 2022-11-04:
Reference to a chat in gitter about this which led to the issue filing: https://gitter.im/cocotb/Lobby?at=6362dc3bdcbf067fdcc34651

ktbarrett, 2022-11-25:
We don't support old version of cocotb. Does this happen with v1.7.2? If `$finish` isn't calling the shutdown callback to shutdown cocotb that's an iverilog bug.

mkorbel1, 2023-07-26:
I was still hitting this issue in newer versions of cocotb as well and on multiple simulators, and I still have these timeouts in my code that I'd like to get rid of.

ktbarrett, 2023-07-26:
Can you upload a test which reproduces the issue? As stated, this shouldn't be happening. We need to determine if this is an issue with your simulator not calling the shutdown callback or something that cocotb can actually handle. I'll remove the stale tag and give you some more time to provide that test.

mkorbel1, 2023-08-28:
I have a unit test that reproduces the issue and makes sure the timeout code is working properly, but it's a part of the rest of the [ROHD Cosim](https://github.com/intel/rohd-cosim) package:
https://github.com/intel/rohd-cosim/blob/main/test/finish_test.dart

Running the test requires having Dart installed, and the test currently passes because of some timeout checking code.  When I have a moment, if that's not convenient, I can try to extract just the cocotb-related portions to create a simpler reproduction.

ktbarrett, 2023-09-29:
Whoops. Forgot to remove the tag. We do have info now. Thank you.