cocotb/cocotb #3135: CI: Stale bot now also targets pull requests

imphil, 2022-11-17:
Looks like the most recent update of the stale GitHub action is now also targeting PRs like #1840, which were untouched by the bot before.

The GitHub actions log says for this particular PR:

```
[#1840] Pull request #1840
  [#1840] Found this pull request last updated at: 2020-08-14T14:07:25Z
  [#1840] The option only-labels (​[https://github.com/actions/stale#only-labels​)](https://github.com/actions/stale#only-labels%E2%80%8B)) was not specified
  [#1840] └── Continuing the process for this pull request
  [#1840] Days before pull request stale: 30
  [#1840] The pull request is not closed nor locked. Trying to remove the close label...
  [#1840] ├── The close-pr-label (​[https://github.com/actions/stale#close-pr-label​)](https://github.com/actions/stale#close-pr-label%E2%80%8B)) option was not set
  [#1840] └── Skipping the removal of the close label
  [#1840] This pull request hasn't a stale label
  [#1840] The option any-of-labels (​[https://github.com/actions/stale#any-of-labels​)](https://github.com/actions/stale#any-of-labels%E2%80%8B)) was specified to only process the issues and pull requests with one of those labels (2)
  [#1840] ├── One of the required labels is present on this pull request
  [#1840] └── Continuing the process for this pull request
  [#1840] This pull request has no milestone
  [#1840] └── Skip the milestones checks
  [#1840] This pull request has no assignee
  [#1840] └── Skip the assignees checks
  [#1840] This pull request is not stale
  [#1840] The option ignore-updates (​[https://github.com/actions/stale#ignore-updates​)](https://github.com/actions/stale#ignore-updates%E2%80%8B)) is disabled. The stale counter will take into account updates and comments on this pull request to avoid to stale when there is some update
  [#1840] This pull request should be stale based on the last update date the 14-08-2020 (2020-08-14T14:07:25Z)
  [#1840] This pull request should be marked as stale based on the option days-before-stale (​[https://github.com/actions/stale#days-before-stale​)](https://github.com/actions/stale#days-before-stale%E2%80%8B)) (30)
  [#1840] Marking this pull request as stale
  [#1840] This pull request is now stale
  [#1840] This pull request is already stale
  [#1840] Checking for label on this pull request
  [#1840] Pull request marked stale on: 2022-11-16T03:18:02Z
  [#1840] Checking for comments on pull request since: 2022-11-16T03:18:02Z
  [#1840] Comments that are not the stale comment or another bot: 0
  [#1840] Pull request has been commented on: false
  [#1840] Days before pull request close: 7
  [#1840] The option remove-stale-when-updated (​[https://github.com/actions/stale#remove-stale-when-updated​)](https://github.com/actions/stale#remove-stale-when-updated%E2%80%8B)) is: true
  [#1840] The stale label should not be removed
  [#1840] marked stale this run, so don't check for updates
  [#1840] Pull request has been updated since it was marked stale: false
  [#1840] Pull request has been updated in the last 7 days: true
  [#1840] Stale pull request is not old enough to close yet (hasComments? false, hasUpdate? true)
  [#1840] 3 operations consumed for this pull request
```

It's clearly a change in behavior, but maybe one we actually want? 

@ktbarrett WDYT?

ktbarrett, 2022-11-17:
I think I like it. I think we also need to get into the habit of deciding up front and making it obvious to the PR author whether the change would ever be merged.

ktbarrett, 2022-11-25:
Well since we let it happen, that means our minds are made up.

closed 2022-11-25 by ktbarrett