cocotb/cocotb #3138: Restarting Simulation from Riviera GUI Hangs with newer Python

jwprice100, 2022-11-18:
When restarting a simulation from within the Riviera GUI hangs Riviera in new versions of python the GUI hangs indefinitely.  I tested this in Python 3.9-11. However in Python 3.7.6. (the version I am upgrading from, also self installed) this issue does not occur. However in Python 3.7.6 I've always seen the following error:
`restart;
# KERNEL: stopped at time: 14880 ns
# COUT:  14880.00ns ERROR    cocotb.scheduler                   Failing test at simulator request before test run completion: Simulator shutdown prematurely
# COUT:  14880.00ns ERROR    gpi                                Passing event to upper layer failed
# COUT: Traceback (most recent call last):
# COUT:   File "/home/jprice/aps_env/lib/python3.7/site-packages/cocotb/__init__.py", line 342, in _sim_event
# COUT:     scheduler._finish_scheduler(SimFailure(msg))
# COUT:   File "/home/jprice/aps_env/lib/python3.7/site-packages/cocotb/scheduler.py", line 1048, in _finish_scheduler
# COUT:     self._abort_test(exc)
# COUT:   File "/home/jprice/aps_env/lib/python3.7/site-packages/cocotb/scheduler.py", line 1029, in _abort_test
# COUT:     self._unschedule(self._test)
# COUT:   File "/home/jprice/aps_env/lib/python3.7/site-packages/cocotb/scheduler.py", line 569, in _unschedule
# COUT:     self._cleanup()
# COUT:   File "/home/jprice/aps_env/lib/python3.7/site-packages/cocotb/scheduler.py", line 1080, in _cleanup
# COUT:     raise Exception("Cleanup() called outside of the main thread")
# COUT: Exception: Cleanup() called outside of the main thread
`
I am not sure if this is related or not. 

cocotb = 1.7.2
operating system = 64 bit Cent Os 7 or Ubuntu 18.04
simulator = Riviera Pro 2022.04 64 bit
python = 3.9-3.11, self installed.

ktbarrett, 2022-11-23:
Dupe of #1169. As discussed in another issue (can't find it rn, someone edit this later) restarting was never supported; if it ever worked it was by purely by accident.

closed 2022-11-23 by ktbarrett

jwprice100, 2022-11-23:
> Dupe of #1169. As discussed in another issue (can't find it rn, someone edit this later) restarting was never supported; if it ever worked it was by purely by accident.

Interesting. This seems like a useful use case. Is there any interest in making this a supported feature for interactive simulators like Active HDL/Riviera/Model Sim/Questa? I think I can work around this by simply forcing my test to die after a certain amount of time but it was very nice not needing to that, and the simulator hanging if I ever forget is rather...frustrating.

ktbarrett, 2022-11-23:
>Is there any interest in making this a supported feature for interactive simulators like Active HDL/Riviera/Model Sim/Questa?

Oh there absolutely is interest in it. The issue is that there needs to be a distinction between shutdown and restart. For whatever reason Riviera (and other simulators) instrumented restart as a shutdown followed by a startup. This is going to cause the Python interpreter to be finalized and not reinitialized. So either 1. Interpreter initialization needs to move to the startup callback or 2. Simulators need to instrument the simulation restart callback properly.

jwprice100, 2022-11-23:
Out of curiosity is this actually a duplicate issue? On Python 3.7.6 I see that error, which is a duplicate. However I'm not sure if that same path is actually exercised when the GUI hangs on a restart with new python. Any thoughts for a test/work around?

mikep996, 2024-02-07:
There is a patch (on request) already for Riviera-PRO 2023.10.
I don't have any issues with the restart procedure on Python 3.10 or higher.