cocotb/cocotb #3139: Not stop simulation at assertion

Louis-DR, 2022-11-20:
Hello,

Right now when a `TestFailure` or `assert` is triggered, it stops the current test and writes a rather ugly Traceback message (like an uncaught exception).

```
   280.00ns INFO     cocotb.regression                  test_routing_algorithm failed
                                                        Traceback (most recent call last):
                                                          File "/Users/louis/Workspace/AnyV-Fabric/gen/vrf/collector.py", line 78, in _collect
                                                            await self.scoreboard.register_collection(
                                                          File "/Users/louis/Workspace/AnyV-Fabric/gen/vrf/scoreboard.py", line 80, in register_collection
                                                            assert False, f"Received transaction that doesn't match any outstanding registered transaction"
                                                        AssertionError: Received transaction that doesn't match any outstanding registered transaction
   280.00ns INFO     cocotb.regression                  ********************************************************************************************
                                                        ** TEST                                STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        ********************************************************************************************
                                                        ** test_router.test_routing_algorithm   FAIL         280.00           0.02      11684.15  **
                                                        ********************************************************************************************
                                                        ** TESTS=1 PASS=0 FAIL=1 SKIP=0                      280.00           0.16       1781.10  **
                                                        ********************************************************************************************
```

When working with large DUT, I might want to run a complex test that can assert at multiple test points with useful messages. In such a case, I want the simulation to continue and show all of the assertion messages so I can solve them in whichever order I want, or assign someone to solve one bug while I work on another (and not have to disable tests to isolate each test point). Also, having information about all the assertions can show a bigger picture about what went wrong and make verification easier.

Therefore, I stopped using assert and rather use a custom function that emits a log error (prettier in the cocotb logs) that doesn't stop the simulation, and at the end of the test I check if there was an error and assert accordingly to show the failed test status.

Could cocotb handle assertion more gracefully and allow the simulation to keep going ? Perhaps with blocking and non-blocking in the test arguments. I might dig into cocotb's code and try some implementation.

ktbarrett, 2022-11-21:
`assert`s, since they are built into Python, cannot be instrumented in the way you want without some awful AST-level hacks. Instrumenting `assert`s are not the best solution. You are describing a Scoreboard which is a common idiom in verification frameworks. I suggest you look at cocotb-bus, which has a (pretty terrible) Scoreboard implementation in it. Alternatively make your own and upload for all cocotb users to take advantage of. Verification components have been deemed out-of-scope in the cocotb repo so I don't think this should be developed here.

Louis-DR, 2022-11-21:
Hi @ktbarrett , thanks for the answer. Yes the very nature of asserts might not be compatible with what I am suggesting. I suppose they were adopted for easier integration with pytest and to match the concept of SVA

I was working with cocotb-bus and opened the issue in the context of developping a Scoreboard. I also noticed some methods used by cocotb-bus are deprecated. So I'll probably keep working on my concept perhaps returning to some exception-based test status management and see if I can propose it to the cocotb-bus repo.

I have difficulties seeing the line that was drawn between the cocotb test features and the out-scoped verification features, I think the two need to go hand-in-hand.

Then we can close the ticket and I'll open a feature request / merge request in cocotb-bus sometimes in the future

closed 2022-11-21 by Louis-DR

ktbarrett, 2022-11-21:
> I have difficulties seeing the line that was drawn between the cocotb test features and the out-scoped verification features, I think the two need to go hand-in-hand.

cocotb is a way for Python to interact with the simulator. While there is some parts of cocotb that appear to only be useful in the context of testing, I consider that legacy cruft which will probably never go away. Consider cocotb directly comparable to languages like SystemVerilog or VHDL, but as a language only for simulation. SystemVerilog and UVM are not synonymous, neither is cocotb and any particular verification methodology. For example, there is PyUVM which implements UVM in Python on top of cocotb.