cocotb/cocotb #3149: Cocotb/Modelsim

omkars3, 2022-11-27:
Hi, I started using cocotb on windows through mingw.
I was able to run vhdl and verilog file separately and it worked fine with help of icarus and ghdl.
Now I need to simulate both files at a time i.e topmodule as vhdl and a verilog file as a sub module. So, downloaded modelsim starter edition (Intel 18.1).
Problem: Not able to locate modelsim inside mingw

C:\tdc_nfm>make SIM=modelsim
C:/msys64/mingw64/lib/python3.10/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:42: *** Unable to locate command >vsim<.  Stop.

* the cocotb version 0.2.2,
* the operating system is windows (using cocotb through mingw64)
* modelsim intel starter edition (18.1),
* the Python version : 3.10.8

ktbarrett, 2022-11-29:
This seems to be a problem with your path. Also mixed language simulation require special (EXPENSIVE) licenses, so I don't think this will work. Closing as off-topic.

closed 2022-11-29 by ktbarrett

Sujjan19, 2022-12-01:
Even i have the same error, there is a free version of modelsim which is supported by cocotb as mentioned in the documentation, 
please provide installation instructions for the any free mixed language simulators for mingW64

themperek, 2022-12-01:
>free mixed language simulators for mingW64

Does not exist to my knowledge.

>how to add PATH within mingw64 

`export PATH=/c/.../bin:$PATH` ?