cocotb/cocotb #3157: Coredump on assertion failure.

jahagirdar, 2022-12-08:
I have seen this issue on both local machine and on Github Actions.
https://github.com/dyumnin/cocotb_coredump/actions/runs/3646391815/jobs/6157420573#step:5:28

The assertion on Line 15 of https://github.com/dyumnin/cocotb_coredump/blob/delayed/tests/dut_test.py#L15 triggers a coredump.

On my local machine the version info is as follows

Python 3.10.6
OS:    Linux Mint 21
Modules in venv
pip3 freeze                                                                                                                     ─╯
attrs==22.1.0
cocotb==1.7.0
cocotb-bus==0.2.1
cocotb-coverage==1.1.0
exceptiongroup==1.0.4
iniconfig==1.1.1
packaging==22.0
pluggy==1.0.0
pytest==7.2.0
python-constraint==1.4.0
PyYAML==6.0
tomli==2.0.1

ktbarrett, 2022-12-08:
The repo link appears to be private.

jahagirdar, 2022-12-09:
Changed the visibility to public.

ktbarrett, 2022-12-13:
You can use the makefile variable [`SIM_CMD_PREFIX`](https://docs.cocotb.org/en/stable/building.html#var-SIM_CMD_PREFIX) to run your simulation with gdb to catch where it is segfaulting.

cmarqu, 2022-12-13:
Another way of attaching gdb is described in https://docs.cocotb.org/en/stable/troubleshooting.html?highlight=gdb#attaching-a-debugger

jahagirdar, 2022-12-15:
Looks like an issue with iverilog, The test works on the devel version.

closed 2022-12-15 by imphil