cocotb/cocotb #3168: parameter with large number does not get handled correctly

catkira, 2023-01-01:
I have a parameter with a rather long number (because iverilog does not support multidimensional parameters, so I sequeeze multiple numbers into one).
The parameter reaches the hdl code correctly, but something goes wrong on the python side.

The parameter is this number:
`
1043161809724755117294630463171082084379269244686810311029607490900248788420381392348463362674117467351380280716396061994807856922595934162415011370737945306097471762891582355257266129874607407022695562401368170834331191570587525271668692425035152628033539633035671272822809848481969293316809956131050719687308142627180187719859186358020489682827020566615291426464441948895237962125901086180674120729539754342713718442926328439298090536281793869892530559822533699520885460200043828645675500690787021624293704335548279396031745742539130949779391845360041558468656930060946865658437656234172368262891111818743752245284837962427971856480068009631783627803463593158425393489613888769529055843669738695165161575481544351020262523014509861557762884136920067057263401584677476858151676420644240180037273464619824125783108857328600765813075760694118749230749032679295972977002489474784749875908710432315440784621321055372136598051978044337017931726632721237632407266511479457338798037948587694505331102427565179203351786911664035714952736332056963449894161364778566454808486588047682353303007806935225584948894648374367526802728090180560360178179648464611823616681910331503501853221338831442171912737474561170439869566397269273475997939859449
`

However in my Python model the parameter becomes `65529`

You can see the details in the test run of my workflow at https://github.com/catkira/5G_SSB_sync/actions/runs/3816910503/jobs/6492835861
Just unfold "Verify with cocotb and icarus"

The parameter gets assembled here: https://github.com/catkira/5G_SSB_sync/blob/45874427129b440ed794299b9668608070e83452/tests/test_PSS_correlator.py#LL123-L125C128
And printed inside the model here:
https://github.com/catkira/5G_SSB_sync/blob/45874427129b440ed794299b9668608070e83452/model/PSS_correlator.py#LL5-L8C48

You can see all the versions that I am using the the CI run. The cocotb-test 0.2.4 is not yet on pypi.

catkira, 2023-01-01:
I checked a bit more where the problem lies.

`PSS_LOCAL `is of type `ModifiableObject`

The value function looks like this
```python
    @NonConstantObject.value.getter
    def value(self) -> BinaryValue:
        binstr = self._handle.get_signal_val_binstr()
        # Skip BinaryValue.assign() as we know we are using a binstr
        result = BinaryValue(n_bits=len(binstr))
        # Skip the permitted characters check as we trust the simulator
        result._set_trusted_binstr(binstr)
        return result
```

calling `_handle.get_signal_val_binstr()` from outside gives the correct parameter in full length like this
```
1111111110110011000000000010001111111111111101111111111110101101000000000011110000000000001101001111111111100000000000000100101111111111110000100000000001000010000000000101100100000000000010111111111111000111111111111100011111111111101100111111111111101010111111111010101000000000001100000000000000110111111111111011100111111111110001010000000000111101000000000010000011111111101110001111111110111100000000000010000111111111111011111111111110111101111111111011000000000000000000000000000000110110111111111111001000000000001100001111111110100010000000000011000000000000001010100000000000110000000000000101011000000000000101101111111111010001000000000110010111111111110011001111111111001110111111111101000100000000001101011111111110100100111111111100010000000000001011110000000000111001111111111111111000000000000001000000000000111010000000000001110100000000010000010000000001101000111111111100111000000000001001010000000000100101000000000111000111111111111111001111111111000000111111111101110011111111110110111111111111000101111111111111010011111111100111110000000000101000111111111100101100000000011010011111111111101100111111111101010000000000000100001111111111000101111111111100000100000000001110011111111110101110000000000100000100000000000110100000000000110101111111111110111100000000001101101111111110011101000000000011100000000000010110001111111111100001111111111101010100000000000011000000000000000011111111111110010000000000100000001111111111110001111111111101010111111111111110001111111110100000111111111100000011111111110100111111111111011111111111111011100111111111110111001111111111011001111111111011011000000000010010100000000001011000000000000100101011111111111111001111111111010111000000000001001111111111111010010000000000101010000000000111011111111111111010100000000000110100000000000010000000000000010110100000000001010011111111111010100011111111101101110000000000000001000000000000010100000000000000000000000001011001111111111101000000000000001011101111111110110101000000000010000111111111111110000000000000000000000000000111111011111111110111111111111111111000111111111101001011111111101101011111111110100111111111111101000011111111111110110000000000000000000000000100100100000000000000011111111110101101111111111010100011111111111000000000000001011010000000000001011000000000001101001111111111010110000000000111011111111111111011011111111111101001000000000000010011111111110101111111111110101000000000000100101000000000010010100000000001001010000000000010010011111111110110010000000000100001111111111011100100000000010000001111111111010011000000000000100011111111101000000000000000001111111111111101010100000000000111000000000010000000111111111111010000000000000000110000000000011111111111111101010111111111110010000000000001011000111111111100101011111111100111011111111111001011111111111110111111111111101111110000000000011010111111111100011111111111101011100000000000111011111111111100000100000000001011000000000000010000111111111001011111111111111011001111111111011000111111111100101100000000000011001111111110011111000000000010010111111111110001010000000001000000111111111101110011111111100011111111111111111100111111111101101100000000001001011111111110011000111111111100111011111111111000110000000001000001111111111111110000000000001110101111111111000111111111111111111000000000001111000000000000101111111111111100101111111111101001000000000000110010111111111101000111111111100110111111111111001100111111111110101011111111110100011111111111010000000000000101011011111111110100000000000000101010111111111101000011111111101000101111111111001010111111111111001000000000010100000000000000000000000000000001000111111111101111010000000001000100000000000010000111111111111000001111111110111000000000000011101100000000001111011111111111001001111111111011100100000000010101100000000000110000000000000100110111111111111010100000000000111001111111111100011111111111101001110000000000001011000000000011111000000000010000100000000000100000000000000100101111111111110001000000000000110100000000000000100111111111101011010000000001001101000000000010001100000000000000001111111111111001
```
so the problem is probably in these lines
```python
        result = BinaryValue(n_bits=len(binstr))
        # Skip the permitted characters check as we trust the simulator
        result._set_trusted_binstr(binstr)
```

catkira, 2023-01-01:
Hmm very strange, this prints the correct parameter
```python
        binstr = dut.PSS_LOCAL._handle.get_signal_val_binstr()
        result = cocotb.binary.BinaryValue(n_bits=len(binstr))
        result._set_trusted_binstr(binstr)
        print(result.value)
```
but this doesnt
```python
print(dut.PSS_LOCAL.value)
```

catkira, 2023-01-01:
ah I just found out the type of `PSS_LOCAL `is `ConstantObject `and not `ModifiableObject`

catkira, 2023-01-01:
I think the problem is that `dut.PSS_LOCAL._handle.get_signal_val_long()` only returns `65529`

catkira, 2023-01-01:
```
if handle_type in [simulator.INTEGER, simulator.ENUM]:
            self._value = self._handle.get_signal_val_long()
```
assumes that all variables of type `simulator.INTEGER` are at maximum as big as a `long` type. But the problem is in python the `int` type can have 'indefinite' many digits.

catkira, 2023-01-01:
A possible solution would be to get the value with `get_signal_val_str()` or `get_signal_val_binstr()` from the simulator and convert it to python `int`

catkira, 2023-01-01:
I have created [this](https://github.com/cocotb/cocotb/issues) PR to fix this issue.

alexforencich, 2023-01-18:
I also ran in to this same issue a couple of days ago (parameter values getting truncated to 32 bits); I was trying to figure out if it was an icarus verilog issue or a cocotb issue, but it looks like you isolated it to cocotb.  Nice work!

ktbarrett, 2024-02-23:
Should be resolved by #3691.

closed 2024-02-23 by ktbarrett