cocotb/cocotb #3170: Deprecation warnings about distutils

catkira, 2023-01-04:
I get these deprecation warnings:
```
../../../../home/benja/.local/lib/python3.10/site-packages/cocotb_test/simulator.py:16
  /home/benja/.local/lib/python3.10/site-packages/cocotb_test/simulator.py:16: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
    from distutils.spawn import find_executable

../../../../home/benja/.local/lib/python3.10/site-packages/cocotb_test/simulator.py:17
  /home/benja/.local/lib/python3.10/site-packages/cocotb_test/simulator.py:17: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
    from distutils.sysconfig import get_config_var

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
```
Is somebody already on it or should I do a PR to fix this?

imphil, 2023-01-04:
Feel free to do a PR. Thanks!

cmarqu, 2023-01-04:
Actually, this is an issue for https://github.com/themperek/cocotb-test only, I think.

catkira, 2023-01-04:
Ah yes, it will be fixed there in 0.2.4.

closed 2023-01-04 by catkira